openwrt/asu #344: ASU update issue

Tom-Neverwinter, 2022-04-08:
https://github.com/dangowrt/owrt-ubi-installer/commit/13e967448206ea1c56028a90d711027ad45440c8

Unsupported package(s): ip6tables, iptables

Please report the error message and request

{ "board_name": "linksys,e8450-ubi", "target": "mediatek/mt7622", "version": "SNAPSHOT", "packages": [ "auc", "base-files", "blockd", "busybox", "ca-bundle", "dnsmasq", "dropbear", "firewall", "fstools", "ip6tables", "iptables", "kernel", "kmod-fs-exfat", "kmod-fs-ext4", "kmod-fs-f2fs", "kmod-fs-vfat", "kmod-gpio-button-hotplug", "kmod-ipt-offload", "kmod-leds-gpio", "kmod-mt7615-firmware", "kmod-mt7615e", "kmod-mt7915e", "kmod-nls-base", "kmod-nls-cp437", "kmod-nls-iso8859-1", "kmod-nls-utf8", "kmod-usb-storage", "kmod-usb-storage-uas", "kmod-usb3", "libc", "libgcc", "libustream-openssl", "logd", "luci", "luci-app-attendedsysupgrade", "luci-ssl-openssl", "luci-theme-openwrt-2020", "mtd", "netifd", "odhcp6c", "odhcpd-ipv6only", "opkg", "ppp", "ppp-mod-pppoe", "procd", "procd-seccomp", "procd-ujail", "uboot-envtools", "uci", "uclient-fetch", "urandom-seed", "urngd", "wpad-openssl" ], ".anonymous": false, ".name": "request", ".type": "request", ".index": 0 }

jck112, 2022-06-02:
I ran into this problem too with "nftables". It looks like virtual packages (i.e. "Provides: nftables") also need to be added to the redis database in janitor.py, e.g.

```python
def update_target_packages(branch: dict, version: str, target: str):
    ...
    virtual_packages = { virtual_name for p in packages.values() if (virtual_name := p.get("provides")) }
    r.sadd(f"packages-{branch['name']}-{version}-{target}", *(virtual_packages | packages.keys()))
    ...
```

And something similar in the `update_arch_packages` function.

jck112, 2022-06-02:
Looks like it may be a bit more complicated since packages can provide multiple virtual packages with specific versions.

```
Package: erlang
Version: 24.2-1
Depends: libc, libncurses6, librt, zlib, libstdcpp6
Provides: erlang-erts=12.2, erlang-kernel=8.2, erlang-sasl=4.1.1, erlang-stdlib=3.17
Source: feeds/packages/lang/erlang
SourceName: erlang
License: Apache-2.0
LicenseFiles: LICENSE.txt
Section: lang
SourceDateEpoch: 1641764777
CPE-ID: cpe:/a:erlang:erlang
Maintainer: Arnaud Sautaux <arnaud.sautaux@infoteam.ch>
Architecture: aarch64_generic
Installed-Size: 8131004
Filename: erlang_24.2-1_aarch64_generic.ipk
Size: 8094755
SHA256sum: e053884f1dee62773476b70f1e706f44ea685455e91f74cdab9d226b21f72bfe
Description:  Erlang/OTP is a general-purpose programming language and runtime
 environment. Erlang has built-in support for concurrency, distribution
 and fault tolerance.
 .
 This package contains the runtime implementation and a minimal set of
 modules (erts, kernel, sasl & stdlib).
```

Tom-Neverwinter, 2022-06-02:
very interesting. above my head atm though. been spending my time on battletech 3.0 stuff for archival

aparcar, 2022-06-03:
@jck112 thanks for looking into this issue. I think a simple *split* could do the trick, the server shouldn't do any complex dependency tracking but just check if a request is valid before running the ImageBuilder.

What do you think about the following?

```python
virtual_packages = { virtual_name.split("=")[0] for p in packages.values() if (virtual_name := p.get("provides")) }
r.sadd(f"packages-{branch['name']}-{version}-{target}", *(virtual_packages | packages.keys()))
```

jck112, 2022-06-03:
Makes sense. I think that would only grab the first virtual package; to grab them all you could do something like:

```python
virtual_packages = {
    vpkg.split("=")[0]
    for pkg in packages.values() if (provides := pkg.get("provides"))
    for vpkg in provides.split(", ")
}
```

closed 2022-06-08 by aparcar