openwrt/asu #345: problem with firewall4

debaba, 2022-04-16:
Hi, first post here, don't know if i'm right here
Today I let built an image on sysupgrade.openwrt.org for an WRT3200ACM. The build was quick and complete. After installation no firewall code was shown in Luci. After some testing i noticed that firewall script in /etc/init.d was not starting. The problem is an incompatible ucode (new version of ucode gets installed but firewall4 depends on an older version).
how can i get a running system? Wait for a change on sysupgrade.openwrt.org? 
Thanks for information 
debaba

debaba, 2022-04-16:
Seems to be a space too much in dependency entry file: is '> = 20220322' should be '>= 20220322'

aparcar, 2022-04-19:
Can you please point me to that code?

debaba, 2022-04-19:
Hi,
Sorry, i needed this AP, so i reverted to stable 21.02
This version worked out of the box.

I tell you, what i remember: I tried to install firewall4. Either when 
installing this package or luci-app-firewall-de package via luci i received 
this install package tree, where the package ucode was signed as being of 
wrong version. After looking around i remarked, that the version compare 
string luci used was '> =202203...' (remark the space between > and =) and so 
luci thought package ucode should be exactly 202203... Not the installed 
version 202204...., which was right if that version compare string was 
'>=202203...'
After trying with luci, i also tried via CLI opk install, error remained the 
same. 

Then i gave up installing a SNAPSHOT and reverted to stable.

Sorry, i do not remembered more exactly, getting older is nothing for cowards

Hopefully you can use this,
Cheeers
Wolfgang

Am Dienstag, 19. April 2022, 11:21:06 EEST schrieb Paul Spooren:
> Can you please point me to that code?

-- 
Wolfgang Schindler
Schintech

8630 Praitori, Paphos
Cyprus

De: +49 611 9457497
Cy: +357 96708167
GPG: http://www.schintech.net/schindler.asc

自由

aparcar, 2022-06-07:
This was fixed upstream via https://github.com/openwrt/luci/commit/b84791bd029c2aaa67cbc06a15d23828f62f4608

closed 2022-06-07 by aparcar