openwrt/asu #346: asking for 22.03.0-rc1 results in "Invalid argument (22)"

hrw, 2022-04-25:
I have "linksys,e8450-ubi" (Belkin one) running SNAPSHOT which I update from time to time with simple "auc" call. But as there is work going on on 22.03.0 release I decided to move to it:

```
root@krolik:~# auc -b 22.03.0-rc1
auc (0.2.5-2)
Server:    https://asu.aparcar.org
Running:   SNAPSHOT r19504-c18c6e53b9 on mediatek/mt7622 (linksys,e8450-ubi)
Invalid argument (22)
root@krolik:~#
```

If I go to "22.03-SNAPSHOT" then it works:

```
root@krolik:~# auc -b 22.03.0-SNAPSHOT
auc (0.2.5-2)
Server:    https://asu.aparcar.org
Running:   SNAPSHOT r19504-c18c6e53b9 on mediatek/mt7622 (linksys,e8450-ubi)
Available: 22.03-SNAPSHOT r19309-0481a5a35a
Requesting package lists...
 luci-mod-system: git-22.102.44067-b6fb2b1 -> git-22.102.44406-0c0cf78
 openssl-util: 1.1.1n-2 -> 1.1.1n-1
 getrandom: 2022-01-06-b87a4fdc-2 -> 2021-08-03-205defb5-2
 ucode-mod-ubus: 2022-04-07-33f1e0b0-1 -> 2022-02-11-a29bad94-1
 luci-theme-bootstrap: git-22.084.38997-8e09f18 -> git-22.084.39030-f187425
 base-files: 1485-r19504-c18c6e53b9 -> 1476-r19309-0481a5a35a
 ucode-mod-uci: 2022-04-07-33f1e0b0-1 -> 2022-02-11-a29bad94-1
 kmod-mt7615e: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 firewall4: 2022-04-02-a3788839-1 -> 2022-02-12-53caa1a7-1
 kmod-mt7615-common: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 luci-mod-status: git-22.087.28325-8a46648 -> git-22.089.68955-66d9387
 kmod-mt76-connac: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 luci-app-firewall: git-22.089.67453-0eb3aeb -> git-22.089.67563-7e3c1b4
 ucode-mod-fs: 2022-04-07-33f1e0b0-1 -> 2022-02-11-a29bad94-1
 odhcpd-ipv6only: 2022-03-22-860ca900-1 -> 2021-08-11-01b4e604-1
 ucode: 2022-04-07-33f1e0b0-1 -> 2022-02-11-a29bad94-1
 kmod-mt7915e: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 libucode: 2022-04-07-33f1e0b0-1 -> 2022-02-11-a29bad94-1
 luci-base: git-22.089.43958-7110635 -> git-22.083.69105-af8e91c
 logd: 2022-01-06-b87a4fdc-2 -> 2021-08-03-205defb5-2
 libopenssl: 1.1.1n-2 -> 1.1.1n-1
 openwrt-keyring: 2022-03-25-62471e69-2 -> 2022-03-25-62471e69-3
 luci-app-attendedsysupgrade: git-22.081.81213-8936e3c -> git-22.068.37885-65266c4
 libopenssl-conf: 1.1.1n-2 -> 1.1.1n-1
 kmod-mt7615-firmware: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 zlib: 1.2.12-2 -> 1.2.11-4
 luci-lib-jsonc: git-22.079.54693-45f411b -> git-22.097.61921-7513345
 kmod-mt76-core: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 ubox: 2022-01-06-b87a4fdc-2 -> 2021-08-03-205defb5-2
 luci-mod-network: git-22.087.28325-8a46648 -> git-22.089.68955-66d9387
Are you sure you want to continue the upgrade process? [N/y] n
Interrupted system call (4)
```

Could auc give better message instead of "22" one?

aparcar, 2022-04-26:
@dangowrt ideas?

hrw, 2022-04-27:
Looks like 22 is 'no such branch'

```
root@krolik:~# auc -b does-not-exists
auc (0.2.5-2)
Server:    https://asu.aparcar.org
Running:   SNAPSHOT r19504-c18c6e53b9 on mediatek/mt7622 (linksys,e8450-ubi)
Invalid argument (22)```

hrw, 2022-04-30:
Looks like fixed on server:
```
root@krolik:~# auc -b 22.03.0-rc1
auc (0.2.5-2)
Server:    https://asu.aparcar.org
Running:   SNAPSHOT r19504-c18c6e53b9 on mediatek/mt7622 (linksys,e8450-ubi)
Available: 22.03.0-rc1 r19302-df622768da
Requesting package lists...
 iwinfo: 2022-03-01-90bfbb9a-1 -> 2022-04-26-dc6847eb-1
 luci-app-statistics: git-22.002.35118-8572ae0 -> git-22.115.68420-46794ea
 libiwinfo-lua: 2022-03-01-90bfbb9a-1 -> 2022-04-26-dc6847eb-1
 luci-mod-system: git-22.102.44067-b6fb2b1 -> git-22.102.44406-0c0cf78
 openssl-util: 1.1.1n-2 -> 1.1.1n-1
 getrandom: 2022-01-06-b87a4fdc-2 -> 2021-08-03-205defb5-2
 ucode-mod-ubus: 2022-04-07-33f1e0b0-1 -> 2022-04-13-e14b0993-1
 luci-theme-bootstrap: git-22.084.38997-8e09f18 -> git-22.084.39030-f187425
 base-files: 1485-r19504-c18c6e53b9 -> 1475-r19302-df622768da
 ucode-mod-uci: 2022-04-07-33f1e0b0-1 -> 2022-04-13-e14b0993-1
 kmod-mt7615e: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 firewall4: 2022-04-02-a3788839-1 -> 2022-04-21-fc83d462-1
 kmod-mt7615-common: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 luci-mod-status: git-22.087.28325-8a46648 -> git-22.089.68955-66d9387
 kmod-mt76-connac: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 luci-app-firewall: git-22.089.67453-0eb3aeb -> git-22.089.67563-7e3c1b4
 ucode-mod-fs: 2022-04-07-33f1e0b0-1 -> 2022-04-13-e14b0993-1
 odhcpd-ipv6only: 2022-03-22-860ca900-1 -> 2021-08-11-01b4e604-1
 libiwinfo-data: 2022-03-01-90bfbb9a-1 -> 2022-04-26-dc6847eb-1
 ucode: 2022-04-07-33f1e0b0-1 -> 2022-04-13-e14b0993-1
 kmod-mt7915e: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 libucode: 2022-04-07-33f1e0b0-1 -> 2022-04-13-e14b0993-1
 luci-base: git-22.089.43958-7110635 -> git-22.083.69105-af8e91c
 logd: 2022-01-06-b87a4fdc-2 -> 2021-08-03-205defb5-2
 libopenssl: 1.1.1n-2 -> 1.1.1n-1
 openwrt-keyring: 2022-03-25-62471e69-2 -> 2022-03-25-62471e69-3
 luci-app-attendedsysupgrade: git-22.081.81213-8936e3c -> git-22.068.37885-65266c4
 libopenssl-conf: 1.1.1n-2 -> 1.1.1n-1
 libiwinfo: 2022-03-01-90bfbb9a-1 -> 2022-04-26-dc6847eb-1
 kmod-mt7615-firmware: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 zlib: 1.2.12-2 -> 1.2.11-4
 luci-lib-jsonc: git-22.079.54693-45f411b -> git-22.097.61921-7513345
 kmod-mt76-core: 5.10.111+2022-04-07-506bb060-4 -> 5.10.111+2022-03-15-053668ac-4
 ubox: 2022-01-06-b87a4fdc-2 -> 2021-08-03-205defb5-2
 luci-mod-network: git-22.087.28325-8a46648 -> git-22.117.37207-4c263cf
Are you sure you want to continue the upgrade process? [N/y]
```

closed 2022-04-30 by hrw

trenchcoatjedi, 2022-09-08:
This issue is still happening. I have the same issue today:

root@OpenWrt:/etc/config# auc
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.1 r16325-88151b8303 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
root@OpenWrt:/etc/config# auc -B 22.03.0
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.1 r16325-88151b8303 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
root@OpenWrt:/etc/config# auc -B 21.02.3
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.1 r16325-88151b8303 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
root@OpenWrt:/etc/config#

hadifarnoud, 2025-02-17:
> This issue is still happening. I have the same issue today:
>
> root@OpenWrt:/etc/config# auc auc (0.2.4-8) Server: https://sysupgrade.openwrt.org Running: 21.02.1 r16325-88151b8303 on bcm27xx/bcm2711 [...]

I have the same issue. how did you fix this?

aparcar, 2025-02-17:
@hadifarnoud please update your AUC instance first. I did an upgrade yesterday from 21.07 all the way to 24.10 with `auc` and it worked great.