openwrt/asu #351: From 21.x to 22.x

StSaens, 2022-05-23:
As far as I see I can't use this package to do a major upgrade to a new release. 

Is this correct? And if so, is this in the pipeline?

aparcar, 2022-05-23:
It is possible but since 22.03 is not officially released yet (only release candidates) the tools doesn't notify you about the version

StSaens, 2022-05-23:
Right. Thanks.
Maybe a feature to allow upgrade to RC's. A flag under the advanced options or something, maybe?

All in all, great package.

StSaens, 2022-05-25:
> I've noticed that you can choose the 22.xx RC version, if you have enabled the "advanced mode" under configuration. However then you need to add the firewall4 packages and remove the previouse. If you leave it as is, you get [#347 (comment)](https://github.com/openwrt/asu/issues/347#issue-1215993338).
> I didn't tried messing with the packages. Good luck!

I’m assuming this is NOT from the Luci app?
I was meaning using the Luci app. Sorry I wasn’t clear.

Underworld6, 2022-05-25:
It is from Luci. 
[image]

StSaens, 2022-05-25:
Odd I don't see 22 rc when I enable these settings:

[image]

aparcar, 2022-05-26:
At this point you can't switch from a SNAPSHOT version to a RC or stable version since it's likely a downgrade.

aparcar, 2022-06-03:
For reference, if you want to use the CLI to upgrade to a pre-release, use `auc -b 21.02 -B 21.02-SNAPSHOT`

closed 2022-06-03 by aparcar