openwrt/asu #354: Server response: Error: Impossible package selection

aidvu, 2022-06-08:
Reporting per suggestion in luci ui.

Request Data:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19755-95adbc24e7",
    "advanced_mode": "0",
    "app_version": "git-22.081.81213-8936e3c",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "xiaomi,redmi-router-ax6s",
    "target": "mediatek/mt7622",
    "version": "SNAPSHOT",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "htop",
        "irqbalance",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7615-common",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-mt7915e",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-nat6",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-nat6",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-ssl",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

STDERR:
```
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/kmods/5.15.45-1-05aa59cfe636b265f9f2e53c5d544be3/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/mediatek/mt7622/kmods/5.15.45-1-05aa59cfe636b265f9f2e53c5d544be3/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * opkg_install_cmd: Cannot install package libwolfssl.
make[2]: *** [Makefile:169: package_install] Error 255
make[1]: *** [Makefile:134: _call_manifest] Error 2
make: *** [Makefile:253: manifest] Error 2
```

aidvu, 2022-06-08:
Could be related to recent `wolfssl` updates: https://git.openwrt.org/?p=openwrt/openwrt.git;a=shortlog;h=refs/heads/master

nbk15lk, 2022-06-08:
STDERROR:

Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/mediatek/mt7622/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/mediatek/mt7622/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/mediatek/mt7622/kmods/5.10.120-1-556ff95eb403ff85eb1fc79c3201b3c8/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/mediatek/mt7622/kmods/5.10.120-1-556ff95eb403ff85eb1fc79c3201b3c8/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/mediatek/mt7622/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20220515 for block-mount
 * pkg_hash_check_unresolved: cannot find dependency libblobmsg-json20220515 for block-mount
 * pkg_hash_fetch_best_installation_candidate: Packages for block-mount found, but incompatible with the architectures configured
 * pkg_hash_check_unresolved: cannot find dependency libblobmsg-json20220515 for blockd
 * pkg_hash_fetch_best_installation_candidate: Packages for blockd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package blockd.
 * pkg_hash_check_unresolved: cannot find dependency libubox20220515 for mtd
 * pkg_hash_fetch_best_installation_candidate: Packages for mtd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package mtd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mtd:
 * 	libubox20220515
 * opkg_install_cmd: Cannot install package mtd.
make[2]: *** [Makefile:169: package_install] Error 255
make[1]: *** [Makefile:134: _call_manifest] Error 2
make: *** [Makefile:253: manifest] Error 2

nbk15lk, 2022-06-08:
The linux SA in me is curious about using home directory space to create automation..  Updated list of available packages in /home/_aparcar_/asu/worker2

aidvu, 2022-06-08:
More or less the same thing on Belkin RT3200. Also mt7622.

AashishAS, 2022-06-09:
Same on Archer A6 and C6 v3. Both MT7621.

aparcar, 2022-06-09:
Should be fixed via https://github.com/openwrt/openwrt/commit/d55f12cc7925b7d84af80015cbeb0422dd7e0b36 could you please try again?

aidvu, 2022-06-09:
Looking good @aparcar. Guess I was correct about `wolfssl`. 😄 Closing.

```
Currently running: SNAPSHOT - r19782-ab1f3a8706
```

closed 2022-06-09 by aidvu