openwrt/asu #356: Error building the firmware image for Linksys mr8300

RaSto70, 2022-06-13:
Error building the firmware image
Server response: Error: Impossible package selection

Collected errors:
 * opkg_install_cmd: Cannot install package ipq-wifi-linksys_mr8300-v0.
make[2]: *** [Makefile:169: package_install] Error 255
make[1]: *** [Makefile:134: _call_manifest] Error 2
make: *** [Makefile:253: manifest] Error 2

It seems the package ipq-wifi-linksys_mr8300-v0 is not available for some reasons

RaSto70, 2022-06-14:
Hi Aparcar, I'm not sure if I understand properly. If I go to System - Software the package is not there.

aparcar, 2022-07-29:
It's working fine for 21.02.3, could you please elaborate where you experience the issue?

RaSto70, 2022-10-11:
Hi Paul,
I don't have any device with 21.02.3 version. Anyway, I was able to upgrade from openwrt-22.03.0-rc5 to openwrt-22.03.0-rc6 so it seems the problem is solved.
Best regards
Rastislav

Delawen, 2024-07-11:
I got the same issue right now.

This is my current setup:

```
Hostname	OpenWrt-mr8300
Model	Linksys MR8300 (Dallas)
Architecture	ARMv7 Processor rev 5 (v7l)
Target Platform	ipq40xx/generic
Firmware Version	OpenWrt 22.03.5 r20134-5f15225c1e / LuCI openwrt-22.03 branch git-24.067.02332-4c1ddfb
Kernel Version	5.10.176
```
And this is the error I get:

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20134-5f15225c1e",
    "advanced_mode": "0",
    "branch": "22.03",
    "efi": null,
    "request_hash": "96e5819d6ea152df7604a2cc7ef8c85a",
    "profile": "linksys,mr8300",
    "target": "ipq40xx/generic",
    "version": "22.03.6",
    "packages": [
        "ath10k-board-qca4019",
        "ath10k-board-qca9888",
        "ath10k-firmware-qca4019-ct",
        "ath10k-firmware-qca9888-ct",
        "base-files",
        "bind-client",
        "bind-libs",
        "bind-server",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "ipq-wifi-linksys_mr8300-v0",
        "isc-dhcp-dyndns-ipv4",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
[... 139 lines cut ...]
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.6/packages/arm_cortex-a7_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.6/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.6/packages/arm_cortex-a7_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.6/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.6/packages/arm_cortex-a7_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.6/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.6/packages/arm_cortex-a7_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /builder/build_dir/target-arm_cortex-a7+neon-vfpv4_musl_eabi/root-ipq40xx/../../../../builder/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check passed.
Collected errors:
 * opkg_install_cmd: Cannot install package ipq-wifi-linksys_mr8300-v0.
make[2]: *** [Makefile:172: package_install] Error 255
make[1]: *** [Makefile:137: _call_manifest] Error 2
make: *** [Makefile:257: manifest] Error 2
```

Delawen, 2024-07-11:
And a second attempt worked. Maybe it was a timeout error somewhere.