openwrt/asu #359: Unsupported profile: tplink,tl-wdr4900-v1

cosminsstoica, 2022-06-15:
Hi, 
I try to build a new image for Tp-Link WDR4900-v1. Seems to be no support for this router in 22.03 firmware

"Error building the firmware image
Server response: Unsupported profile: tplink,tl-wdr4900-v1
[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16554-1d4dea6d4f",
    "advanced_mode": "1",
    "app_version": "git-22.090.55699-bb6ef9f",
    "branch": "21.02",
    "efi": null,
    "profile": "tplink,tl-wdr4900-v1",
    "target": "mpc85xx/p1010",
    "version": "22.03.0-rc4",
"

aparcar, 2022-07-29:
The device was disabled due to Kernel size issues, see https://github.com/aparcar/openwrt/commit/d69bf6601e5bff897781361f2718513d4dd8efd8

closed 2022-07-29 by aparcar

cosminsstoica, 2022-07-29:
Thanks for the answer. 

Meantime I build my own image, just need to change the following option in menuconfig

Global build settings --->
Kernel build options --->
[*] Strip non-essential /proc functionality to reduce code size 

All good.

aparcar, 2022-07-30:
I wonder if there is a less invasive solution for that to make it available for everyone

tim-seoss, 2022-09-07:
The original TPLink uboot fork has some limitations which effectively limit the kernel size.  The solution is to add uboot support for this device (preferably upstream uboot).  The solutions would then be either:

1a. Overwrite the factory uboot with the fixed uboot, or:
1b. Load the fixed uboot from the factory uboot (i.e. leave the factory uboot intact) - this uses a little more flash space, but is usually preferred because it guarantees the vendor firmware can be restored (although probably not an issue for such an old device), and is the usual workaround for this issue on OpenWRT devices.
2. The new uboot loads the OpenWRT firmware (i.e. it no longer has the load size limit).

I haven't looked at the factory uboot, but it may be possible to make it load larger kernels (possibly even just by changing its configuration - and this config change may or may not be practical to carry out without a serial console).