openwrt/asu #360: ER Profile Still Unsupported.

papac0rn, 2022-06-15:
Tried again, and still getting this error for the Edgerouter.  Thanks again.:)

Error building the firmware image
Server response: Unsupported profile: er

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16495-bf0c965af0",
    "advanced_mode": "0",
    "app_version": "git-22.090.55699-bb6ef9f",
    "branch": "21.02",
    "efi": null,
    "profile": "er",
    "target": "octeon/generic",
    "version": "21.02.3",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "collectd",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall",
        "fstools",
        "ip6tables",
        "iptables",
        "kernel",
        "kmod-ipt-offload",
        "libc",
        "libustream-wolfssl",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-mwan3",
        "luci-app-nlbwmon",
        "mkf2fs",
        "mtd",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "ucert",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd"
    ],
    "diff_packages": true,
    "filesystem": "squashfs"

papac0rn, 2022-06-15:
login as: root
root@192.168.246.6's password:

BusyBox v1.33.2 (2022-03-20 00:32:36 UTC) built-in shell (ash)

  _______                     ________        __
 |       |.-----.-----.-----.|  |  |  |.----.|  |_
 |   -   ||  _  |  -__|     ||  |  |  ||   _||   _|
 |_______||   __|_____|__|__||________||__|  |____|
          |__| W I R E L E S S   F R E E D O M
 -----------------------------------------------------
 OpenWrt 21.02.2, r16495-bf0c965af0
 -----------------------------------------------------
root@OpenWrt:~# ubus call system board
{
        "kernel": "5.4.179",
        "hostname": "OpenWrt",
        "system": "UBNT_E200 (CN6120p1.1-1000-NSP)",
        "model": "UBNT_E200 (CN6120p1.1-1000-NSP)",
        "board_name": "er",
        "release": {
                "distribution": "OpenWrt",
                "version": "21.02.2",
                "revision": "r16495-bf0c965af0",
                "target": "octeon/generic",
                "description": "OpenWrt 21.02.2 r16495-bf0c965af0"
        }
}
root@OpenWrt:~#

In case you all need this again. . .  Thanks guys.

aparcar, 2022-07-29:
I just back ported this to 21.02 and 22.03, for the latter please be aware that it is currently broken due to an unrelated issue.