openwrt/asu #361: 自定义镜像05.24.2022无法升级

Jim-CN, 2022-06-17:
自定义镜像05.24.2022，选择定制系统升级，搜索后可识别组件清单，选择升级06.14.2022，进入排队，显示“构建固件中，进度：60%验证包选择”，约1分钟后报错，如下图：
 
[image]

Jim-CN, 2022-06-17:
补充信息
{
    "url": "https://op.supes.top",
    "revision": "05.24.2022",
    "advanced_mode": "1",
    "app_version": "git-22.147.57852-d7e4f23",
    "branch": "22.03",
    "efi": "efi",
    "rootfs_type": "squashfs",
    "profile": "default-string-default-string",
    "target": "x86/64",
    "version": "22.03.0-rc1",
    "packages": [
        "automount",
        "bandwidthd-sqlite",
        "base-files",
        "bash",
        "bnx2-firmware",
        "busybox",
        "ca-bundle",
        "chinadns-ng",
        "containerd",
        "coremark",
        "coreutils-base64",
        "curl",
        "dns2tcp",
        "dnsmasq-full",
        "docker",
        "dockerd",
        "dropbear",
        "e2fsprogs",
        "fdisk",
        "filebrowser",
        "firewall",
        "fstools",
        "getrandom",
        "haproxy",
        "htop",
        "ip-full",
        "ip6tables",
        "iptables",
        "kernel",
        "kmod-amazon-ena",
        "kmod-amd-xgbe",
        "kmod-bnx2",
        "kmod-button-hotplug",
        "kmod-e1000",
        "kmod-e1000e",
        "kmod-forcedeth",
        "kmod-fs-vfat",
        "kmod-igb",
        "kmod-igc",
        "kmod-ipt-offload",
        "kmod-ixgbe",
        "kmod-lib-zstd",
        "kmod-tcp-bbr",
        "kmod-tg3",
        "kmod-usb2",
        "kmod-usb3",
        "kmod-vmxnet3",
        "libc",
        "libcap",
        "libdbus",
        "libev",
        "libgd",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "liblucihttp",
        "libnl-tiny",
        "libopenssl",
        "libopenssl-conf",
        "libpcre",
        "libtasn1",
        "libubus-lua",
        "libuci",
        "libuci-lua",
        "libustream-wolfssl",
        "libuuid",
        "libwebsockets-full",
        "libxml2",
        "lm-sensors-detect",
        "logd",
        "luci-app-accesscontrol",
        "luci-app-advanced",
        "luci-app-arpbind",
        "luci-app-attendedsysupgrade",
        "luci-app-autoreboot",
        "luci-app-cifs-mount",
        "luci-app-control-timewol",
        "luci-app-diskman",
        "luci-app-dockerman",
        "luci-app-filebrowser",
        "luci-app-firewall",
        "luci-app-gpsysupgrade",
        "luci-app-opkg",
        "luci-app-passwall",
        "luci-app-samba4",
        "luci-app-ttyd",
        "luci-app-upnp",
        "luci-app-vlmcsd",
        "luci-app-wizard",
        "luci-app-wol",
        "luci-app-wrtbwmon",
        "luci-app-zerotier",
        "luci-base",
        "luci-compat",
        "luci-i18n-base-en",
        "luci-lib-ipkg",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-ssl-nginx",
        "luci-theme-edge",
        "miniupnpd",
        "mkf2fs",
        "mtd",
        "my-autocore-x86",
        "my-default-settings",
        "nano",
        "netifd",
        "nginx-mod-luci",
        "ntfs-3g",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssl-util",
        "opkg",
        "partx-utils",
        "pciutils",
        "perlbase-bytes",
        "perlbase-class",
        "perlbase-io",
        "perlbase-socket",
        "perlbase-symbol",
        "php8",
        "php8-cgi",
        "php8-mod-gd",
        "php8-mod-pdo",
        "php8-mod-pdo-sqlite",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "runc",
        "shadowsocksr-libev-ssr-local",
        "tailscale",
        "terminfo",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd",
        "v2ray-core",
        "v2ray-plugin",
        "wget-ssl",
        "xray-core",
        "zram-swap"
    ],
    "diff_packages": true,
    "partsize": "1004"
}

aparcar, 2022-07-29:
Hi, please write issues in English. I'm unaware of that server instance and neither know that OpenWrt revision `05.24.2022`. Could you please elaborate on your setup?