openwrt/asu #362: opkg_install_cmd: Cannot install package kmod-crypto-lib-blake2s.

StSaens, 2022-06-21:
Getting an impossible package selection error.

`{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19312-1b7cf4dd1c",
    "advanced_mode": "0",
    "app_version": "git-22.068.37885-65266c4",
    "branch": "22.03",
    "efi": null,
    "rootfs_type": "squashfs",
    "profile": "netgear,r7800",
    "target": "ipq806x/generic",
    "version": "22.03-SNAPSHOT",
    "packages": [
        "6in4",
        "6rd",
        "6to4",
        "ath10k-board-qca9984",
        "ath10k-firmware-qca9984-ct",
        "base-files",
        "block-mount",
        "busybox",
        "ca-bundle",
        "ca-certificates",
        "ccrypt",
        "cgi-io",
        "collectd",
        "collectd-mod-conntrack",
        "collectd-mod-cpu",
        "collectd-mod-cpufreq",
        "collectd-mod-dhcpleases",
        "collectd-mod-entropy",
        "collectd-mod-exec",
        "collectd-mod-interface",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-network",
        "collectd-mod-ping",
        "collectd-mod-rrdtool",
        "collectd-mod-thermal",
        "collectd-mod-uptime",
        "curl",
        "diffutils",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "gdbserver",
        "getrandom",
        "hostapd-common",
        "hostapd-utils",
        "htop",
        "ip6tables-nft",
        "iptables-mod-ipopt",
        "iptables-mod-ipsec",
        "iptables-nft",
        "irqbalance",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-ahci",
        "kmod-ata-ahci-platform",
        "kmod-ata-core",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cdrom",
[... 237 lines cut ...]
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/ipq806x/generic/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/ipq806x/generic/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a15_neon-vfpv4/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/ipq806x/generic/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/ipq806x/generic/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a15_neon-vfpv4/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/ipq806x/generic/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/ipq806x/generic/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a15_neon-vfpv4/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/ipq806x/generic/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/ipq806x/generic/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a15_neon-vfpv4/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/ipq806x/generic/build_dir/target-arm_cortex-a15+neon-vfpv4_musl_eabi/root-ipq806x/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03-SNAPSHOT/ipq806x/generic/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * opkg_install_cmd: Cannot install package kmod-crypto-lib-blake2s.
make[2]: *** [Makefile:169: package_install] Error 255
make[1]: *** [Makefile:134: _call_manifest] Error 2
make: *** [Makefile:253: manifest] Error 2 `

StSaens, 2022-06-23:
Found this: https://git.openwrt.org/?p=openwrt/openwrt.git;a=commit;h=cd634afe6cb6565eb6865931c8d73d97cab3600a
But I don't fully understand if upgrading the router without this lib breaks anything. Doesn't seem to be related to asu.

aparcar, 2022-07-29:
> as BLAKE2s is now built-in

Looks like you can just remove it.

closed 2022-07-29 by aparcar

mmokrejs, 2023-02-22:
```
Error building the firmware image

Server response: Unsupported package(s): kmod-crypto-lib-blake2s
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16554-1d4dea6d4f",
    "advanced_mode": "0",
    "branch": "21.02",
    "efi": null,
    "profile": "netgear,r7800",
    "target": "ipq806x/generic",
    "version": "21.02.5",
    "packages": [
        "acme-dnsapi",
        "ath10k-board-qca9984",
        "ath10k-firmware-qca9984-ct",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "cryptsetup",
        "cryptsetup-ssh",
        "diffutils",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip-bridge",
        "ip6tables",
        "ip6tables-mod-nat",
        "iptables",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-ahci",
        "kmod-ata-ahci-platform",
        "kmod-ata-core",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-arc4",
        "kmod-crypto-authenc",
        "kmod-crypto-cbc",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32",
        "kmod-crypto-crc32c",
        "kmod-crypto-cts",
        "kmod-crypto-deflate",
        "kmod-crypto-des",
        "kmod-crypto-ecb",
        "kmod-crypto-ecdh",
        "kmod-crypto-echainiv",
        "kmod-crypto-fcrypt",
        "kmod-crypto-gcm",
        "kmod-crypto-hw-geode",
        "kmod-crypto-hw-hifn-795x",
        "kmod-crypto-hw-padlock",
        "kmod-crypto-hw-talitos",
[... 154 lines cut ...]
        "strace",
        "swconfig",
        "syncthing",
        "tinc",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd",
        "usign",
        "vpnc",
        "vsftpd-tls",
        "wireless-regdb",
        "wpad-basic-wolfssl",
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.210.54423-3b0b325"
}
```