openwrt/asu #363: The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application

k2p2p, 2022-06-22:
Server response: The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.

Please report the error message and request
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19838-acbe1d4d36",
    "advanced_mode": "1",
    "app_version": "git-22.081.81213-8936e3c",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "phicomm,k3",
    "target": "bcm53xx/generic",
    "version": "SNAPSHOT",
    "packages": [
        "base-files",
        "bash",
        "block-mount",
        "busybox",
        "ca-bundle",
        "ca-certificates",
        "cgi-io",
        "coreutils",
        "coreutils-nohup",
        "curl",
        "dnsmasq-full",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "ip-full",
        "ip6tables-legacy",
        "ipset",

ccdunder, 2022-06-24:
Same issue for me.

IANAexpert but these server stats look unhealthy: https://sysupgrade.openwrt.org/stats

robhancock, 2022-06-25:
Same error. The stats page above seems to show the server has been broken for several days.

fra-iesus, 2022-06-25:
Indeed looks like a server-side problem of this particular server - changing the server address to https://chef.libremesh.org solved the problem for me.

tlaurion, 2022-06-28:
"collectd-mod-wireless", "dnsmasq", "dropbear", "firewall", "fstools", "iwinfo", "kernel", "kmod-btmrvl", "kmod-gpio-button-hotplug", "kmod-ipt-offload", "kmod-mwifiex-sdio", "kmod-mwlwifi", "libc", "libustream-wolfssl", "logd", "luci", "luci-app-advanced-reboot", "luci-app-attendedsysupgrade", "luci-app-diag-core", "luci-app-firewall", "luci-app-nlbwmon", "luci-app-opkg", "luci-app-qos", "luci-app-statistics", "mtd", "mwlwifi-firmware-88w8964", "netifd", "odhcp6c", "odhcpd-ipv6only", "opkg", "ppp", "ppp-mod-pppoe", "procd", "tor", "tor-geoip", "uboot-envtools", "ucert", "uci", "uclient-fetch", "urandom-seed", "urngd", "wpad-basic-wolfssl" ], "diff_packages": true }

rossiniscarface, 2022-06-30:
Could be the same problem as #350 
There it was the server run out of diskspace.

tlaurion, 2022-07-01:
@aparcar disk space issues?

tlaurion, 2022-07-01:
Changing servers is not possible really for users unless they want to run snapshot into production environments.

aparcar, 2022-07-01:
Space issues again. I'll write a cleanup daemon soonish

ccdunder, 2022-07-02:
It's working now. Thank you @aparcar !
I sent a donation to Open WRT earmarked for https://sysupgrade.openwrt.org/ in appreciation.

GeertVisser, 2022-07-07:
its broken again 😉

k2p2p, 2022-07-08:
its broken again

k2p2p, 2022-07-08:
> It's working now. Thank you @aparcar !
> I sent a donation to Open WRT earmarked for https://sysupgrade.openwrt.org/ in appreciation.

its broken again

k2p2p, 2022-07-09:
It's working now, Thanks

philwiffen, 2022-07-18:
Heads up that I'm getting this error message again

magneticflux-, 2022-07-27:
I'm getting the same error:
`The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.`
```json
{
    "url": "https://sysupgrade.openwrt.org/",
    "revision": "r19426-2b1941e47d",
    "advanced_mode": "1",
    "app_version": "git-22.068.37885-65266c4",
    "branch": "22.03",
    "efi": null,
    "rootfs_type": "squashfs",
    "profile": "linksys,ea8300",
    "target": "ipq40xx/generic",
    "version": "22.03.0-rc5",
    "packages": [
        "ath10k-board-qca4019",
        "ath10k-firmware-qca4019-ct",
        "ath10k-firmware-qca9888-ct",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables-legacy",
        "ipq-wifi-linksys_ea8300",
        "iptables-legacy",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb3",
        "libblobmsg-json",
        "libc",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
        "libwolfssl",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "swconfig",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true
}
```

robinsmith3, 2023-07-04:
Happening again

"The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."

pojlFDlxCOvZ4Kg8y1l4, 2023-07-04:
It is indeed happening again. I'll post logs anyways:  

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20123-38ccc47687",
    "advanced_mode": "0",
    "branch": "22.03",
    "efi": {
        "path": "/sys/firmware/efi",
        "type": "directory",
        "size": 0,
        "mode": 16877,
        "atime": 1688477729,
        "mtime": 1688477729,
        "ctime": 1688477729,
        "inode": 7404,
        "uid": 0,
        "gid": 0
    },
    "profile": "lenovo-10mqs9bw00",
    "target": "x86/64",
    "version": "22.03.5",
    "packages": [
        "adguardhome",
        "avahi-autoipd",
        "avahi-daemon-service-http",
        "avahi-daemon-service-ssh",
        "avahi-dnsconfd",
        "avahi-utils",
        "base-files",
        "busybox",
        "ca-bundle",
        "collectd-mod-cpu",
        "collectd-mod-cpufreq",
        "ddns-scripts-cloudflare",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "ethtool",
        "firewall4",
        "fstools",
        "grub2",
        "grub2-bios-setup",
        "grub2-efi",
        "htop",
        "igmpproxy",
        "intel-microcode",
        "iperf3",
        "iptables-zz-legacy",
        "kernel",
        "kmod-amazon-ena",
        "kmod-amd-xgbe",
        "kmod-bnx2",
        "kmod-button-hotplug",
        "kmod-e1000",
        "kmod-e1000e",
        "kmod-forcedeth",
        "kmod-fs-vfat",
        "kmod-igb",
        "kmod-igc",
        "kmod-ixgbe",
        "kmod-nft-offload",
        "kmod-r8169",
        "kmod-sched-mqprio",
        "kmod-tg3",
        "kmod-usb-net-rtl8152",
        "knot-host",
        "libavahi-client",
        "libc",
        "libjpeg-turbo",
        "libustream-wolfssl",
        "logd",
        "luci-app-advanced-reboot",
        "luci-app-attendedsysupgrade",
        "luci-app-ddns",
        "luci-app-nlbwmon",
        "luci-app-sqm",
        "luci-app-udpxy",
        "luci-app-upnp",
        "luci-app-vnstat",
        "luci-app-wireguard",
        "luci-i18n-udpxy-en",
        "luci-proto-autoip",
        "luci-proto-bonding",
        "luci-ssl",
        "mdns-utils",
        "miniupnpd-nftables",
        "mkf2fs",
        "mtd",
        "nano",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "partx-utils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "qrencode",
        "tc-tiny",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd",
        "wg-installer-server"
    ],
    "diff_packages": true,
    "filesystem": "ext4",
    "client": "luci/git-23.093.42303-58b861d"
}
```

wangchong666, 2023-08-24:
Happening again

Zahidpichen1, 2024-05-15:
it's still happening again