openwrt/asu #364: Issue with auc: No data available (61)

Duneathor, 2022-06-28:
See below

root@OpenWrt:~# auc
auc (0.2.5-2)
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r19787-e3e9eb31a2 on mediatek/mt7622 (linksys,e8450-ubi)
No data available (61)

Duneathor, 2022-06-28:
Works fine when changing
/etc/config/attendedsysupgrade 
to  https://asu.aparcar.org

rossiniscarface, 2022-07-06:
same here
```
root@rectangle:~#auc -c
auc (0.2.5-2)
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r19704-efff48529b on ramips/mt7621 (netgear,r6260)
No data available (61)
```

and in Luci i get "Server response: The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application."

But as "Selected firmware version" in the gui it states "SNAPSHOT -r20011-265f402fbd"

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19704-efff48529b",
    "advanced_mode": "1",
    "app_version": "git-22.081.81213-8936e3c",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "netgear,r6260",
    "target": "ramips/mt7621",
    "version": "SNAPSHOT",
    "packages": [
        "6in4",
        "acme",
        "acme-dnsapi",
        "adblock",
        "arp-scan",
        "arp-scan-database",
        "attendedsysupgrade-common",
        "auc",
        "base-files",
        "block-mount",
        "busybox",
        "ca-bundle",
        "ca-certificates",
        "cgi-io",
        "collectd",
        "collectd-mod-cpu",
        "collectd-mod-exec",
        "collectd-mod-interface",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-logfile",
        "collectd-mod-memory",
        "collectd-mod-netlink",
        "collectd-mod-network",
        "collectd-mod-ping",
        "collectd-mod-rrdtool",
        "collectd-mod-sqm",
        "collectd-mod-syslog",
        "collectd-mod-threshold",
        "collectd-mod-uptime",
        "coreutils",
        "coreutils-sort",
        "curl",
        "diffutils",
        "dnsmasq",
        "dropbear",
        "ethtool",
        "f2fs-tools",
        "firewall",
        "fstools",
        "fwtool",
        "gdisk",
        "getrandom",
        "hostapd-common",
        "htop",
        "https-dns-proxy",
        "ip6tables-legacy",
        "iperf3",
        "iptables-legacy",
        "iptables-mod-conntrack-extra",
        "iptables-mod-ipopt",
        "iputils-arping",
        "iputils-tracepath",
        "irqbalance",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-crc32",
        "kmod-fs-f2fs",
        "kmod-gpio-button-hotplug",
        "kmod-ipt-offload",
        "kmod-leds-gpio",
        "kmod-mt7603",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-scsi-core",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb-storage",
        "kmod-usb3",
        "libc",
        "libcurl",
        "libustream-wolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-acme",
        "luci-app-adblock",
        "luci-app-attendedsysupgrade",
        "luci-app-commands",
        "luci-app-firewall",
        "luci-app-https-dns-proxy",
        "luci-app-opkg",
        "luci-app-sqm",
        "luci-app-statistics",
        "luci-app-uhttpd",
        "luci-app-upnp",
        "luci-base",
        "luci-compat",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "luci-theme-openwrt-2020",
        "mesh11sd",
        "miniupnpd-iptables",
        "mtd",
        "nano",
        "netifd",
        "netperf",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssl-util",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rpcsys",
        "rpcd-mod-rrdns",
        "rrdtool1",
        "socat",
        "speedtest-netperf",
        "sqm-scripts",
        "sysfsutils",
        "tc-mod-iptables",
        "tc-tiny",
        "tcpdump",
        "terminfo",
        "ubi-utils",
        "ubox",
        "ubus",
        "ubusd",
        "ucert",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usbutils",
        "usign",
        "wget-ssl",
        "wireless-regdb",
        "wpad-wolfssl",
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "squashfs"
}
```

schuettecarsten, 2022-07-07:
Same here. When looking into the server stats, you can see that the server stopped to build images some hours (days) ago.
https://sysupgrade.openwrt.org/stats/?orgId=1&refresh=1m&from=now-7d&to=now

```
Server response: The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.
```

dangowrt, 2022-07-11:
The error is usually triggered in case the server replies empty when `branches.json` is requested.

schuettecarsten, 2022-07-16:
Today, same problem occurs again...

hwknsj, 2022-07-26:
> Today, same problem occurs again...
> 
> ```
>  -----------------------------------------------------
>  OpenWrt 21.02.3, r16554-1d4dea6d4f
>  -----------------------------------------------------
> auc (0.2.4-8)
> Server:    https://sysupgrade.openwrt.org
> Running:   21.02.3 r16554-1d4dea6d4f on ath79/generic (alfa-network,r36a)
> No data available (61)
> ```
> 
> ```
>  -----------------------------------------------------
>  OpenWrt 22.03.0-rc5, r19523-bfd070e7fa
>  -----------------------------------------------------
> auc (0.2.5-2)
> Server:    https://sysupgrade.openwrt.org
> Running:   22.03.0-rc5 r19523-bfd070e7fa on mediatek/mt7622 (linksys,e8450-ubi)
> No data available (61)
> ```

10+ days on, still seeing this error, both from the CLI or web UI

aparcar, 2022-07-26:
Odd, I'll investigate what's wrong. My first guess is that the server storage is full, yet again. I'll have to extend it...

Hurricos, 2022-07-26:
I currently get:

```sh
$ curl -H"Content-Type: application/json" -XPOST --upload-file request.json https://sysupgrade.openwrt.org/api/v1/build
{
  "detail": "The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.",
  "status": 500,
  "title": "Internal Server Error",
  "type": "about:blank"
}
```

... for `request.json` ...

```json
{
	"distro": "OpenWrt",
	"target": "ar71xx/generic",
	"profile": "unifiac-pro",
	"version": "19.07.5",
	"packages": [
		"terminfo",
		"iwinfo",
		"liblucihttp0",
		"cgi-io",
		"libc"
	]
}
```

aparcar, 2022-07-29:
Hey everyone, please try again. I changed the ImagebBuilder cache to 24h so it should run more stable now
https://github.com/openwrt/asu/commit/c7c61039bb601169061e41de3219725566f9158f

schuettecarsten, 2022-08-02:
Problem still occured today.

aparcar, 2022-08-02:
Thanks, again an issue storage. I extended the storage to 500GB and hope it's fine from now one... I'll keep you posted.

Panderine, 2022-08-23:
hi, I think it's acting up again :(

trenchcoatjedi, 2022-09-08:
Same issue here:

`root@OpenWrt:/etc/config# auc
auc (0.2.4-8)
Server:    https://asu.aparcar.org
Running:   21.02.1 r16325-88151b8303 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Connection error: Connection failed
No data available (61)
`

aparcar, 2022-09-10:
It's an upstream problem somewhat, I'm evaluating a fix here https://github.com/openwrt/openwrt/pull/10655

aparcar, 2023-02-04:
Fixed via e5965b2aa4ad6de841bc1e844aa2ef2198a93663

solsticedhiver, 2023-10-04:
I am getting this right now:
```
root@aegir:~# auc -n
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.0-rc3 r23389-5deed175a5 on ath79/generic (dlink,dap-2695-a1)
Available: 23.05.0-rc4 r23482-7fe85ce1f2
Requesting package lists...
No data available (61)
```
I don't know if it's "normal".

I had to do some manual nasty things because of https://github.com/openwrt/openwrt/issues/13487

aparcar, 2023-10-07:
Should work again. Please have a look and report back

solsticedhiver, 2023-10-09:
Seems ok; got a list of package that will be upgraded and the build is requested on server.
rebooting now ... 🤞

metaphys, 2023-11-29:
this issue is back for me...

solsticedhiver, 2023-11-29:
I can confirm it is back for me too.
```
root@xxxxxx:~# auc -n
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ath79/generic (dlink,dap-2695-a1)
No data available (61)
```

dw-loewe, 2023-11-29:
yeah, for me too. i think the best possible solution would be to host a server myself..

dataCobra, 2023-12-15:
Yeah I'm also encountering that issue.

brada4, 2023-12-15:
@dw-loewe @dataCobra would be nice to post device identifiers like from auc output so that @aparcar can look into them.

jannoke, 2023-12-16:
root@R1S-H5-J67:~# auc
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   22.03.5 r20134-5f15225c1e on sunxi/cortexa53 (friendlyarm,nanopi-r1s-h5)
Available: 22.03.6 r20265-f85a79bcb4
Requesting package lists...
No data available (61)

Getting same problem.  I can see openwrt.org wiki being really slow too for number of days. Do you guys need help with hw/bw?

marev3, 2024-03-27:
`root@Router_AP1:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ramips/mt7621 (zyxel,wsm20)
Available: 23.05.3 r23809-234f1a2efa
Requesting package lists...
No data available (61)`