openwrt/asu #369: Documentation

schuettecarsten, 2022-07-16:
Please provide a bit more "getting started"-documentation, e.g. how to setup a server.

aparcar, 2022-07-24:
Hi, what information would you like to see specifically?

magneticflux-, 2022-07-27:
I solved https://github.com/openwrt/asu/issues/372 on my own, but I'm stuck with a bunch of lines like
`WARNING in janitor: Couldn't download https://downloads.openwrt.org/snapshots/targets/mxs/generic/profiles.json`
and
`[2022-07-27 12:46:41,432] WARNING in janitor: Something weird about Package: opkg`
and
`1:M 27 Jul 2022 13:56:30.598 * 100 changes in 300 seconds. Saving...`
and
`12:05:47 Cleaning registries for queue: default`
and the request never completes.

magneticflux-, 2022-07-30:
I've still got errors like
```

openwrt-asu-janitor-1    | [2022-07-30 13:48:03,294] WARNING in janitor: Something weird about Package: umbim
openwrt-asu-janitor-1    | Version: 2015-07-04-29aaf43b-1
openwrt-asu-janitor-1    | Depends: libc, libubox, kmod-usb-net, kmod-usb-net-cdc-mbim, wwan
openwrt-asu-janitor-1    | Source: package/network/utils/umbim
openwrt-asu-janitor-1    | License: GPL-2.0
openwrt-asu-janitor-1    | Section: net
openwrt-asu-janitor-1    | Maintainer: John Crispin <john@phrozen.org>
openwrt-asu-janitor-1    | Architecture: mipsel_24kc
openwrt-asu-janitor-1    | Installed-Size: 12714
openwrt-asu-janitor-1    | Filename: umbim_2015-07-04-29aaf43b-1_mipsel_24kc.ipk
openwrt-asu-janitor-1    | Size: 13537
openwrt-asu-janitor-1    | SHA256sum: cab9d55a6e5984d7bc946bb3facc25f3d05d699a1e4e9fbb64e021de00941840
openwrt-asu-janitor-1    | Description: umbim is a command line tool for controlling mobile broadband modems using
openwrt-asu-janitor-1    |  the MBIM-protocol.
openwrt-asu-janitor-1    |
openwrt-asu-janitor-1    |
openwrt-asu-janitor-1    | [2022-07-30 13:48:03,295] WARNING in janitor: Something weird about Package: uqmi
openwrt-asu-janitor-1    | Version: 2016-12-19-8ceeab69-1
openwrt-asu-janitor-1    | Depends: libc, libubox, libblobmsg-json, kmod-usb-net, kmod-usb-net-qmi-wwan, wwan
openwrt-asu-janitor-1    | Source: package/network/utils/uqmi
openwrt-asu-janitor-1    | License: GPL-2.0
openwrt-asu-janitor-1    | Section: net
openwrt-asu-janitor-1    | Maintainer: Matti Laakso <malaakso@elisanet.fi>
openwrt-asu-janitor-1    | Architecture: mipsel_24kc
openwrt-asu-janitor-1    | Installed-Size: 21798
openwrt-asu-janitor-1    | Filename: uqmi_2016-12-19-8ceeab69-1_mipsel_24kc.ipk
openwrt-asu-janitor-1    | Size: 22599
openwrt-asu-janitor-1    | SHA256sum: b48fb80afedfc68923f0380f752420f7c74e7f9f9b255835a378f61584ef22ec
openwrt-asu-janitor-1    | Description: uqmi is a command line tool for controlling mobile broadband modems using
openwrt-asu-janitor-1    |  the QMI-protocol.
openwrt-asu-janitor-1    |
openwrt-asu-janitor-1    |
openwrt-asu-janitor-1    | [2022-07-30 13:48:03,296] WARNING in janitor: Something weird about Package: usbip-client
openwrt-asu-janitor-1    | Version: 2.0-6
openwrt-asu-janitor-1    | Depends: libc, usbip, kmod-usbip-client
openwrt-asu-janitor-1    | Source: feeds/packages/net/usbip
openwrt-asu-janitor-1    | License: GPL-2.0
openwrt-asu-janitor-1    | Section: net
openwrt-asu-janitor-1    | Maintainer: Nuno Goncalves <nunojpg@gmail.com>
openwrt-asu-janitor-1    | Architecture: mipsel_24kc
openwrt-asu-janitor-1    | Installed-Size: 10279
openwrt-asu-janitor-1    | Filename: usbip-client_2.0-6_mipsel_24kc.ipk
openwrt-asu-janitor-1    | Size: 11035
openwrt-asu-janitor-1    | SHA256sum: ae9af408056f59f277f579fe80db143f8b347f0bdbe43b6a9f0f8a16a6d5b6a2
openwrt-asu-janitor-1    | Description: USB-over-IP (client)
openwrt-asu-janitor-1    |
openwrt-asu-janitor-1    |
openwrt-asu-janitor-1    | [2022-07-30 13:48:03,296] WARNING in janitor: Something weird about Package: usbip-server
openwrt-asu-janitor-1    | Version: 2.0-6
openwrt-asu-janitor-1    | Depends: libc, usbip, kmod-usbip-server
openwrt-asu-janitor-1    | Source: feeds/packages/net/usbip
openwrt-asu-janitor-1    | License: GPL-2.0
openwrt-asu-janitor-1    | Section: net
openwrt-asu-janitor-1    | Maintainer: Nuno Goncalves <nunojpg@gmail.com>
openwrt-asu-janitor-1    | Architecture: mipsel_24kc
openwrt-asu-janitor-1    | Installed-Size: 8493
openwrt-asu-janitor-1    | Filename: usbip-server_2.0-6_mipsel_24kc.ipk
openwrt-asu-janitor-1    | Size: 9244
openwrt-asu-janitor-1    | SHA256sum: d9accdb20df0c6af5f6c229bf3ee75f6b33e28159a8531239aea3ac8e0aa3f22
openwrt-asu-janitor-1    | Description: USB-over-IP (server)
openwrt-asu-janitor-1    |
openwrt-asu-janitor-1    |
openwrt-asu-janitor-1    | [2022-07-30 13:48:03,297] WARNING in janitor: Something weird about Package: usbip
openwrt-asu-janitor-1    | Version: 2.0-6
openwrt-asu-janitor-1    | Depends: libc, libwrap, kmod-usbip, libudev-fbsd
openwrt-asu-janitor-1    | Source: feeds/packages/net/usbip
openwrt-asu-janitor-1    | License: GPL-2.0
openwrt-asu-janitor-1    | Section: net
openwrt-asu-janitor-1    | Maintainer: Nuno Goncalves <nunojpg@gmail.com>
openwrt-asu-janitor-1    | Architecture: mipsel_24kc
openwrt-asu-janitor-1    | Installed-Size: 211966
openwrt-asu-janitor-1    | Filename: usbip_2.0-6_mipsel_24kc.ipk
openwrt-asu-janitor-1    | Size: 212482
openwrt-asu-janitor-1    | SHA256sum: 914c87dcc749170a2bf401f58c5f4f0e4a6e8c6ec1a4066f33dda2e724fdf1f1
openwrt-asu-janitor-1    | Description: USB-over-IP (common)
openwrt-asu-janitor-1    |
openwrt-asu-janitor-1    |
```

@aparcar Could you share parts of your `config.py`? I tried updating the `BRANCHES` part myself, but I don't know what to set all the fields to.

schuettecarsten, 2022-08-11:
@aparcar

For example, how to add new OpenWRT sources to config.py, where to get the pubkey from.
When I follow your steps, I run a server 0.7.7, but git shows version 0.7.12.
...