openwrt/asu #371: Error with Unattended SysUpgrade

m1tchll, 2022-07-21:
[Option](http://192.168.2.1/)
[Status](http://192.168.2.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
[Overview](http://192.168.2.1/cgi-bin/luci/admin/status/overview)
[Routing](http://192.168.2.1/cgi-bin/luci/admin/status/routes)
[Firewall](http://192.168.2.1/cgi-bin/luci/admin/status/nftables)
[System Log](http://192.168.2.1/cgi-bin/luci/admin/status/logs)
[Processes](http://192.168.2.1/cgi-bin/luci/admin/status/processes)
[Channel Analysis](http://192.168.2.1/cgi-bin/luci/admin/status/channel_analysis)
[Realtime Graphs](http://192.168.2.1/cgi-bin/luci/admin/status/realtime)
[System](http://192.168.2.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
[System](http://192.168.2.1/cgi-bin/luci/admin/system/system)
[Administration](http://192.168.2.1/cgi-bin/luci/admin/system/admin)
[Software](http://192.168.2.1/cgi-bin/luci/admin/system/opkg)
[Startup](http://192.168.2.1/cgi-bin/luci/admin/system/startup)
[Scheduled Tasks](http://192.168.2.1/cgi-bin/luci/admin/system/crontab)
[Attended Sysupgrade](http://192.168.2.1/cgi-bin/luci/admin/system/attendedsysupgrade)
[LED Configuration](http://192.168.2.1/cgi-bin/luci/admin/system/leds)
[Backup / Flash Firmware](http://192.168.2.1/cgi-bin/luci/admin/system/flash)
[Custom Commands](http://192.168.2.1/cgi-bin/luci/admin/system/commands)
[Reboot](http://192.168.2.1/cgi-bin/luci/admin/system/reboot)
[Advanced Reboot](http://192.168.2.1/cgi-bin/luci/admin/system/advanced_reboot)
[Services](http://192.168.2.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
[Adblock](http://192.168.2.1/cgi-bin/luci/admin/services/adblock)
[Network](http://192.168.2.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
[Interfaces](http://192.168.2.1/cgi-bin/luci/admin/network/network)
[Wireless](http://192.168.2.1/cgi-bin/luci/admin/network/wireless)
[Routing](http://192.168.2.1/cgi-bin/luci/admin/network/routes)
[DHCP and DNS](http://192.168.2.1/cgi-bin/luci/admin/network/dhcp)
[Diagnostics](http://192.168.2.1/cgi-bin/luci/admin/network/diagnostics)
[Firewall](http://192.168.2.1/cgi-bin/luci/admin/network/firewall)
[VPN](http://192.168.2.1/cgi-bin/luci/admin/system/attendedsysupgrade#)
[OpenVPN](http://192.168.2.1/cgi-bin/luci/admin/vpn/openvpn)
[Logout](http://192.168.2.1/cgi-bin/luci/admin/logout)
PAUSED
[Overview](http://192.168.2.1/cgi-bin/luci/admin/system/attendedsysupgrade/overview)
[Configuration](http://192.168.2.1/cgi-bin/luci/admin/system/attendedsysupgrade/configuration)
Attended Sysupgrade
The attended sysupgrade service allows to easily upgrade vanilla and custom firmware images.

This is done by building a new firmware on demand via an online service.

Currently running: 22.03.0-rc4 - r19426-2b1941e47d

Search for firmware upgrade

[Powered by LuCI openwrt-22.03 branch (git-22.167.28394-8a4486a)](https://github.com/openwrt/luci) / OpenWrt 22.03.0-rc4 r19426-2b1941e47d
Error building the firmware image
Server response: Error: Impossible package selection

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19426-2b1941e47d",
    "advanced_mode": "1",
    "app_version": "git-22.068.37885-65266c4",
    "branch": "22.03",
    "efi": null,
    "rootfs_type": "squashfs",
    "profile": "linksys,wrt3200acm",
    "target": "mvebu/cortexa9",
    "version": "22.03.0-rc5",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
[... 151 lines cut ...]
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker2/cache/22.03.0-rc5/mvebu/cortexa9/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../../../../../../../../home/aparcar/asu/worker2/cache/22.03.0-rc5/mvebu/cortexa9/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for libustream-wolfssl20201210
 * pkg_hash_fetch_best_installation_candidate: Packages for libustream-wolfssl found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package libustream-wolfssl.
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for px5g-wolfssl
 * pkg_hash_fetch_best_installation_candidate: Packages for px5g-wolfssl found, but incompatible with the architectures configured
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for luci-ssl:
 * 	libwolfssl5.4.0.ee39414e
 * opkg_install_cmd: Cannot install package luci-ssl.
 * pkg_hash_check_unresolved: cannot find dependency libwolfssl5.4.0.ee39414e for wpad-basic-wolfssl
 * pkg_hash_fetch_best_installation_candidate: Packages for wpad-basic-wolfssl found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package wpad-basic-wolfssl.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for wpad-basic-wolfssl:
 * 	libwolfssl5.4.0.ee39414e
 * opkg_install_cmd: Cannot install package wpad-basic-wolfssl.
make[2]: *** [Makefile:169: package_install] Error 255
make[1]: *** [Makefile:134: _call_manifest] Error 2
make: *** [Makefile:253: manifest] Error 2
Close

hwknsj, 2022-07-26:
why did you paste the full page

https://cve.mitre.org/cgi-bin/cvename.cgi?name=CVE-2022-34293
https://www.wolfssl.com/docs/security-vulnerabilities/

libwolfssl5.4.0.ee39414e isn't on opkg repo anymore. but at least you got a response from `sysupgrade.openwrt.org`!

aparcar, 2022-07-29:
Please see #375

closed 2022-07-29 by aparcar

m1tchll, 2022-10-11:
I must have made a mistake