openwrt/asu #375: 22.03 requests currently broken

aparcar, 2022-07-29:
Due to upstream issues described [here](https://github.com/openwrt/openwrt/pull/10329) all images for 22.03 are currently broken. Please wait until the issue is fixed before reporting the error here.

aparcar, 2022-08-02:
Fixed via https://github.com/openwrt/openwrt/commit/8b6b73d0e895bbbfc448bea9fb5f3751593d7865

closed 2022-08-02 by aparcar

neheb, 2022-09-06:
you should unpin this.

sirweazel, 2022-09-12:
Comment says fixed and closed on Aug 2. I might not understand the SSL problem though. Do i need to update the luci-app-attendedsysupgrade package for the fix to apply?  Currently when checking for an update, i do not get any available. 

[image]
[image]
[image]