openwrt/asu #377: Error building Firmware Image (pi4)

jdebuhr, 2022-08-12:
Error building the firmware image
Server response: init

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16554-1d4dea6d4f",
    "advanced_mode": "1",
    "branch": "21.02",
    "efi": null,
    "profile": "raspberrypi,4-model-b",
    "target": "bcm27xx/bcm2711",
    "version": "22.03.0-rc6",
    "packages": [
        "adguardhome",
        "base-files",
        "bcm27xx-gpu-fw",
        "bcm27xx-userland",
        "block-mount",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "cypress-firmware-43455-sdio",
        "cypress-nvram-43455-sdio-rpi-4b",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "ip6tables",
        "iptables",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-brcmfmac",
        "kmod-fs-ext4",
        "kmod-fs-vfat",
        "kmod-hid",
        "kmod-hid-generic",
        "kmod-input-core",
        "kmod-input-evdev",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-lib-crc-ccitt",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-sound-arm-bcm2835",
        "kmod-sound-core",
        "kmod-usb-core",
        "kmod-usb-hid",
        "kmod-usb-net-rtl8152",
        "kmod-usb-storage",
        "kmod-usb2",
        "kmod-usb3",
        "libblkid",
        "libblobmsg-json",
        "libcomerr",
        "libext2fs",
        "libf2fs",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
        "libuuid",
        "libwolfssl",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-sqm",
        "luci-app-wireguard",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mkf2fs",
        "mtd",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "partx-utils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "usign",
        "wg-installer-server",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "ext4",
    "client": "luci/git-22.210.54423-3b0b325"
}

Bollex42, 2022-08-16:
+1

Error building the firmware image

Server response: init
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16495-bf0c965af0",
    "advanced_mode": "1",
    "branch": "21.02",
    "efi": null,
    "profile": "raspberrypi,4-model-b",
    "target": "bcm27xx/bcm2711",
    "version": "21.02.3",
    "packages": [
        "adblock",
        "banip",
        "base-files",
        "bcm27xx-gpu-fw",
        "bcp38",
        "bind-host",
        "busybox",
        "ca-bundle",
        "curl",
        "cypress-firmware-43455-sdio",
        "cypress-nvram-43455-sdio-rpi-4b",
        "ddns-scripts",
        "ddns-scripts-services",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "ethtool",
        "firewall",
        "fstools",
        "htop",
        "ip-full",
        "ip6tables",
        "iperf3",
        "iptables",
        "irqbalance",
        "iwinfo",
        "kernel",
        "kmod-brcmfmac",
        "kmod-fs-vfat",
        "kmod-ipt-offload",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-sound-arm-bcm2835",
        "kmod-sound-core",
        "kmod-usb-hid",
        "kmod-usb-net-rtl8152",
        "libc",
        "libuhttpd-wolfssl",
        "libustream-wolfssl",
        "logd",
        "luci",
        "luci-app-adblock",
        "luci-app-attendedsysupgrade",
        "luci-app-banip",
        "luci-app-bcp38",
        "luci-app-ddns",
        "luci-app-sqm",
        "luci-app-uhttpd",
        "luci-app-watchcat",
        "luci-ssl",
        "mkf2fs",
        "mtd",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "partx-utils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "python3-speedtest-cli",
        "sqm-scripts",
        "tcpdump",
        "ucert",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "watchcat",
        "wget-ssl",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "ext4",
    "client": "luci/git-22.210.54423-3b0b325"
}

aparcar, 2022-08-16:
storage full again, should work again...

Bollex42, 2022-08-16:
At the moment the problem persists.

closed 2022-08-16 by aparcar

Bollex42, 2022-08-16:
I have still issues:

```
root@rpi4:~# auc -x
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.2 r16495-bf0c965af0 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
root@rpi4:~# auc -c
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.2 r16495-bf0c965af0 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
root@rpi4:~# auc -r
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.2 r16495-bf0c965af0 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
root@rpi4:~# auc -b 21.02
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.2 r16495-bf0c965af0 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
root@rpi4:~# auc -b 22.03
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.2 r16495-bf0c965af0 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
```

aparcar, 2022-08-16:
Failed builds are cached for an hour. After that you should give it another try

PR #379 (2022-08-16): build: remove IB leftover on cleanup
The ImageBuilder folder is sometimes deleted without removing it's
stamp, this way the server thinks the folder is still there.

This commit removes stamp, sums ans sig files.

Fix: #377

Future versions should store things in Redis rather than files.
commit f2c1923d (2022-08-16): build: remove IB leftover on cleanup

The ImageBuilder folder is sometimes deleted without removing it's
stamp, this way the server thinks the folder is still there.

This commit removes stamp, sums ans sig files.

Fix: #377

Future versions should store things in Redis rather than files.