openwrt/asu #378: Error building the firmware image. Server response: Unsupported package(s): kmod-nft-nat6

koskem, 2022-08-15:
Starting from [5.15.60-1-1baface726c5a96994a4aedc66b694d7] kmod-nft-nat6.ipk is no longer going to. This causes an error when building the firmware.

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20212-beeb49740b",
    "advanced_mode": "1",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "xiaomi,redmi-router-ax6s",
    "target": "mediatek/mt7622",
    "version": "SNAPSHOT",
    "packages": [
        "6in4",
        "auc",
        "base-files",
        "bind-dig",
        "bind-host",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "curl",
        "dnsmasq-full",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "golang-gitlab-yawning-obfs4-dev",
        "grep",
        "hostapd-common",
        "https-dns-proxy",
        "i2pd",
        "ipset",
        "iptables-zz-legacy",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-ipt-nat",
        "kmod-ipt-nat-extra",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7615-common",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-mt7915e",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-nat6",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-nat6",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-openssl",
        "libwolfssl",
        "logd",
        "lua",
        "luabitop",
        "luasec",
        "luasocket",
        "luci",
        "luci-app-adblock",
        "luci-app-attendedsysupgrade",
        "luci-app-ddns",
        "luci-app-firewall",
        "luci-app-https-dns-proxy",
        "luci-app-opkg",
        "luci-app-upnp",
        "luci-app-wireguard",
        "luci-app-wol",
        "luci-base",
        "luci-i18n-adblock-ru",
        "luci-i18n-attendedsysupgrade-ru",
        "luci-i18n-base-ru",
        "luci-i18n-dashboard-ru",
        "luci-i18n-ddns-ru",
        "luci-i18n-firewall-ru",
        "luci-i18n-https-dns-proxy-ru",
        "luci-i18n-upnp-ru",
        "luci-i18n-wireguard-ru",
        "luci-i18n-wol-ru",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-theme-bootstrap",
        "mc",
        "mtd",
        "netifd",
        "nftables-json",
        "obfs4proxy",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "tor",
        "tor-geoip",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wget-ssl",
        "wireless-regdb",
        "wpad-mesh-openssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.210.53339-af23d0d"
}

aparcar, 2022-08-16:
Yes the error happens due to it's removal here:

```
commit b75425370d8de747457c137463bc4d15f6f44d00
Author: Hauke Mehrtens <hauke@hauke-m.de>
Date:   Sun Aug 7 14:06:14 2022 +0200

    kernel: kmod-nft-nat6: Remove package
    
    The nft NAT packages for IPv4 and IPv6 were merged into the common
    packages with kernel 5.1. The kmod-nft-nat6 package was empty in our
    build, remove it.
    
    Multiple kernel configuration options were also removed, remove them
    from our generic kernel configuration too.
    
    Signed-off-by: Hauke Mehrtens <hauke@hauke-m.de>
```

I'll look into a replacement logic.

StaRipper, 2022-08-17:
opkg remove kmod-nft-nat6

No packages removed.
Collected errors:
 * print_dependents_warning: Package kmod-nft-nat6 is depended upon by packages:
 * print_dependents_warning:    firewall4
 * print_dependents_warning: These might cease to work if package kmod-nft-nat6 is removed.

 * print_dependents_warning: Force removal of this package with --force-depends.
 * print_dependents_warning: Force removal of this package and its dependents
 * print_dependents_warning: with --force-removal-of-dependent-packages.

Anteus, 2022-08-19:
It works now.

Imwithstupid007, 2022-08-20:
@Anteus: Here it still doesn't work....

{
    "url": "https://sysupgrade.openwrt.org/",
    "revision": "r19592-596efe1356",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "linksys,e8450-ubi",
    "target": "mediatek/mt7622",
    "version": "22.03-SNAPSHOT",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd-mod-cpufreq",
        "collectd-mod-thermal",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "htop",
        "irqbalance",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7615-common",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-mt7915e",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-nat",
        "kmod-nf-nat6",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-nat6",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb3",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-https-dns-proxy",
        "luci-app-nlbwmon",
        "luci-app-opkg",
        "luci-app-statistics",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.210.54153-69231a7"
}

Anteus, 2022-08-20:
Check out this and the three following comments in the forum thread for rt3200 it might do it.

https://forum.openwrt.org/t/belkin-rt3200-linksys-e8450-wifi-ax-discussion/94302/2665?u=anteus