openwrt/asu #380: No "server" on clean docker based installation

schuettecarsten, 2022-08-16:
I've setup an asu server in docker on a clean ubuntu VM.
I does not work and I have no idea why.

I cannot see any servers in the swagger API docs page....?

[image]

muellerj, 2022-09-15:
Getting the same error on a Raspberry Pi:
```
$ auc
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.2 r16495-bf0c965af0 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
```

aparcar, 2022-09-16:
I just tested the docker container locally and it works fine... I'll setup a RaspberryPi to test that device specifically.

<details>

```
asu-janitor-1    | [2022-09-16 22:51:22,088] WARNING in janitor: Couldn't download https://downloads.openwrt.org/snapshots/targets/lantiq/xway_legacy/profiles.json
asu-janitor-1    | [2022-09-16 22:51:22,946] WARNING in janitor: Couldn't download https://downloads.openwrt.org/snapshots/targets/layerscape/armv8_32b/profiles.json
asu-redis-1      | 1:M 16 Sep 2022 22:51:32.044 * 100 changes in 300 seconds. Saving...
asu-redis-1      | 1:M 16 Sep 2022 22:51:32.048 * Background saving started by pid 48
asu-redis-1      | 48:C 16 Sep 2022 22:51:33.896 * DB saved on disk
asu-redis-1      | 48:C 16 Sep 2022 22:51:33.900 * Fork CoW for RDB: current 1 MB, peak 1 MB, average 1 MB
asu-redis-1      | 1:M 16 Sep 2022 22:51:33.988 * Background saving terminated with success
asu-redis-1      | 1:M 16 Sep 2022 22:56:34.018 * 100 changes in 300 seconds. Saving...
asu-redis-1      | 1:M 16 Sep 2022 22:56:34.031 * Background saving started by pid 49
asu-redis-1      | 49:C 16 Sep 2022 22:56:35.542 * DB saved on disk
asu-redis-1      | 49:C 16 Sep 2022 22:56:35.545 * Fork CoW for RDB: current 1 MB, peak 1 MB, average 0 MB
asu-redis-1      | 1:M 16 Sep 2022 22:56:35.574 * Background saving terminated with success
asu-worker-1     | 22:59:28 default: asu.build.build({'version': '22.03.0', 'version_code': 'r19685-512e76967f', 'target': 'medi...) (4b43e5bbc8633c7302bed21dbe0169e6)
asu-worker-1     | 22:59:29 Building {'version': '22.03.0', 'version_code': 'r19685-512e76967f', 'target': 'mediatek/mt7622', 'filesystem': 'squashfs', 'profile': 'linksys_e8450-ubi', 'diff_packages': True, 'packages_versions': {'iwinfo': '2022-08-19-0dad3e66-1', 'kmod-crypto-gf128': '5.10.138-1', 'opkg': '2022-02-24-d038e5b6-1', 'ubus': '2022-06-01-2bebf93c-1', 'iw': '5.16-1', 'kmod-crypto-manager': '5.10.138-1', 'libuci': '2021-10-22-f84f49f0-6', 'busybox': '1.35.0-3', 'kmod-crypto-ctr': '5.10.138-1', 'kmod-nft-fib': '5.10.138-1', 'kmod-nfnetlink': '5.10.138-1', 'kmod-crypto-hash': '5.10.138-1', 'kmod-nf-reject6': '5.10.138-1', 'kmod-crypto-aead': '5.10.138-1', 'strace': '5.16-1', 'kmod-nf-flow': '5.10.138-1', 'libnl-tiny': '2021-11-21-8e0555fb-1', 'libustream-wolfssl': '2022-01-16-868fd881-1', 'kmod-lib-crc-ccitt': '5.10.138-1', 'getrandom': '2021-08-03-205defb5-2', 'ucode-mod-ubus': '2022-08-29-344fa9e6-1', 'kmod-pppoe': '5.10.138-1', 'kmod-pppox': '5.10.138-1', 'wpad-basic-wolfssl': '2022-01-16-cff80b4f-12', 'kmod-nf-reject': '5.10.138-1', 'procd-ujail': '2022-06-01-7a009685-1', 'base-files': '1490-r19685-512e76967f', 'kmod-nf-nat': '5.10.138-1', 'kmod-crypto-crc32c': '5.10.138-1', 'ucode-mod-uci': '2022-08-29-344fa9e6-1', 'kmod-mt7615e': '5.10.138+2022-08-26-5ec78e1e-4', 'netifd': '2022-08-25-76d2d41b-1', 'libubus': '2022-06-01-2bebf93c-1', 'firewall4': '2022-09-01-f5fcdcf2-1', 'uboot-envtools': '2022.01-31', 'dnsmasq': '2.86-14', 'procd': '2022-06-01-7a009685-1', 'kmod-crypto-hmac': '5.10.138-1', 'kmod-crypto-sha256': '5.10.138-1', 'kmod-mt7615-common': '5.10.138+2022-08-26-5ec78e1e-4', 'ubusd': '2022-06-01-2bebf93c-1', 'kmod-nf-nat6': '5.10.138-1', 'kmod-crypto-acompress': '5.10.138-1', 'kmod-lib-crc32c': '5.10.138-1', 'libjson-script': '2022-05-15-d2223ef9-1', 'kmod-mt76-connac': '5.10.138+2022-08-26-5ec78e1e-4', 'kmod-crypto-seqiv': '5.10.138-1', 'kmod-nft-nat': '5.10.138-1', 'kmod-usb3': '5.10.138-1', 'libmnl': '1.0.5-1', 'jansson': '2.13.1-2', 'tcpdump': '4.9.3-4', 'ubi-utils': '2.1.4-1', 'odhcp6c': '2022-08-05-7d21e8d8-18', 'fstools': '2022-06-02-93369be0-2', 'uclient-fetch': '2021-05-14-6a6011df-1', 'kmod-crypto-ghash': '5.10.138-1', 'uci': '2021-10-22-f84f49f0-6', 'ucode-mod-fs': '2022-08-29-344fa9e6-1', 'dropbear': '2022.82-2', 'kmod-hwmon-core': '5.10.138-1', 'libnftnl': '1.2.1-1', 'mtd': '26', 'odhcpd-ipv6only': '2022-03-22-860ca900-1', 'procd-seccomp': '2022-06-01-7a009685-1', 'libiwinfo-data': '2022-08-19-0dad3e66-1', 'ucode': '2022-08-29-344fa9e6-1', 'kmod-mt7915e': '5.10.138+2022-08-26-5ec78e1e-4', 'kmod-crypto-ccm': '5.10.138-1', 'kmod-nf-log': '5.10.138-1', 'urandom-seed': '3', 'luci-mod-dashboard': 'git-21.246.68627-0ce25b0', 'ppp': '2.4.9.git-2021-01-04-3', 'kmod-leds-gpio': '5.10.138-1', 'kmod-gpio-button-hotplug': '5.10.138-3', 'logd': '2021-08-03-205defb5-2', 'kmod-nf-log6': '5.10.138-1', 'kmod-mac80211': '5.10.138+5.15.58-1-1', 'libblobmsg-json': '2022-05-15-d2223ef9-1', 'libwolfssl': '5.4.0-stable-5', 'openwrt-keyring': '2022-03-25-62471e69-3', 'luci-app-attendedsysupgrade': 'git-22.210.54153-69231a7', 'jshn': '2022-05-15-d2223ef9-1', 'nftables-json': '1.0.2-2.1', 'kmod-nft-offload': '5.10.138-1', 'kmod-ppp': '5.10.138-1', 'kmod-crypto-null': '5.10.138-1', 'libiwinfo': '2022-08-19-0dad3e66-1', 'kmod-mt7615-firmware': '5.10.138+2022-08-26-5ec78e1e-4', 'libjson-c': '0.15-2', 'auc': '0.2.5-2', 'kmod-nf-conntrack': '5.10.138-1', 'usign': '2020-05-23-f1f65026-1', 'kmod-crypto-gcm': '5.10.138-1', 'libubox': '2022-05-15-d2223ef9-1', 'ca-bundle': '20211016-1', 'libuclient': '2021-05-14-6a6011df-1', 'kmod-crypto-cmac': '5.10.138-1', 'kmod-crypto-rng': '5.10.138-1', 'luci': 'git-20.074.84698-ead5e81', 'kmod-nf-conntrack6': '5.10.138-1', 'usteer': '2022-08-18-7d2b17c9-1', 'libucode': '2022-08-29-344fa9e6-1', 'kmod-lib-lzo': '5.10.138-1', 'kmod-mt76-core': '5.10.138+2022-08-26-5ec78e1e-4', 'ubox': '2021-08-03-205defb5-2', 'kmod-nft-core': '5.10.138-1', 'fwtool': '2019-11-12-8f7fe925-1', 'j [... cut ...] ed80237d4a2bed8d2', 'type': 'preloader.bin'}], 'supported_devices': ['linksys,e8450-ubi'], 'titles': [{'model': 'E8450', 'variant': 'UBI', 'vendor': 'Linksys'}, {'model': 'RT3200', 'variant': 'UBI', 'vendor': 'Belkin'}], 'id': 'linksys_e8450-ubi', 'bin_dir': '4b43e5bbc8633c7302bed21dbe0169e6', 'build_at': '2022-09-03T02:55:34.000000Z', 'detail': 'done'}"
asu-worker-1     | 23:05:46 Result is kept for 604800 seconds
asu-worker-1     | 23:05:46 Cleaning registries for queue: default
```
</details>

schuettecarsten, 2022-09-17:
> I just tested the docker container locally and it works fine... I'll setup a RaspberryPi to test that device specifically.

BTW, you only provide amd64 docker images. Is it possible to create arm images to be able to run the server on a Raspberry Pi?

aparcar, 2022-09-17:
Not really since it would require arm64 imagebuilders

muellerj, 2022-11-25:
Is there anything I can help to analyse this? I'm still getting the same error on `0.3.1-1`:

```
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   21.02.2 r16495-bf0c965af0 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
```

DagNygren, 2024-04-24:
I have the same problem here on a Raspi-4. Also auc/0.3.1-1