openwrt/asu #382: Get an invalid argument when trying to use auc

RickvDi, 2022-08-24:
When I try to upgrade my router with auc. I get an invalid argument. It doesn't matter what argument I give with or which branch I use. It always gives invalid argument as output.
Can someone please help me?

aparcar, 2022-08-24:
Could you please give this another try? Also it helps if you tell me what release/device you're looking at.

RickvDi, 2022-08-25:
It was on x86_64 system and it works now

trenchcoatjedi, 2022-09-08:
I have the same issue currently:

root@OpenWrt:/etc/config# auc
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.1 r16325-88151b8303 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
root@OpenWrt:/etc/config# auc -B 22.03.0
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.1 r16325-88151b8303 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
root@OpenWrt:/etc/config# auc -B 21.02.3
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.1 r16325-88151b8303 on bcm27xx/bcm2711 (raspberrypi,4-model-b)
Invalid argument (22)
root@OpenWrt:/etc/config#

nmvega, 2023-10-02:
Same here:

```
user@friendlyelec$ sudo auc
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   22.03.5 r20134-5f15225c1e on rockchip/armv8 (friendlyelec,nanopi-r6s)
Invalid argument (22)
```