openwrt/asu #383: Bad message (74) when running auc

RickvDi, 2022-08-25:
I get a bad message error when running AUC on my x86_64 router build.
[image]

aparcar, 2022-08-26:
Yup, my bad. It's fixed

lpyparmentier, 2022-09-07:
Hello, similar issue with `mediatek/mt7622`

```sh
# auc
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.1 r16325-88151b8303 on mediatek/mt7622 (ubnt,unifi-6-lr)
Available: 21.02.3 r16554-1d4dea6d4f
Requesting package lists...
cannot request packagelist from rpcd
Are you sure you want to continue the upgrade process? [N/y] y
cannot request packagelist from rpcd
init
Bad message (74)
```

aparcar, 2022-09-10:
@lpyparmentier please run `/etc/init.d/rpcd restart` after installing `auc`.

ismail, 2022-09-13:
Getting a similar message for `mt7622` as well:

```
root@openwrt:~# auc
auc (0.2.5-2)
Server:    https://sysupgrade.openwrt.org/
Running:   SNAPSHOT r20614-c27279dc26 on mediatek/mt7622 (linksys,e8450-ubi)
Available: SNAPSHOT r20621-b21ddbfa18
Requesting package lists...
 base-files: 1497-r20614-c27279dc26 -> 1497-r20621-b21ddbfa18
Are you sure you want to continue the upgrade process? [N/y] y
init
Bad message (74)
```

lpyparmentier, 2022-09-14:
@aparcar thanks, it worked, I was able to go from `21.02.1` to `21.02.3`, however it seems unable to go to `22.03` ?

See below

```sh
root@lr6-node1:~# auc -b 22.03.0
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.3 r16554-1d4dea6d4f on mediatek/mt7622 (ubnt,unifi-6-lr)
Invalid argument (22)
```

I've check that the image exists for the architecture mediatek/mt7622 through the following command

```sh
curl -X 'GET' \
  'https://sysupgrade.openwrt.org/api/v1/overview' \
  -H 'accept: application/json'
```

andersk, 2022-11-05:
I’m getting the same error, and sysupgrade.openwrt.org seems to be responding to all requests with HTTP/2 502 (Bad Gateway).