openwrt/asu #387: kmod-nft-nat6 error upgrading from 22.03.0-rc6 to 22.03 using Linksys E8450 (UBI)

zxcvbn97, 2022-09-06:
Error building the firmware image
Server response: Unsupported package(s): kmod-nft-nat6

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19590-042d558536",
    "advanced_mode": "0",
    "branch": "22.03",
    "efi": null,
    "profile": "linksys,e8450-ubi",
    "target": "mediatek/mt7622",
    "version": "22.03.0",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iperf3",
        "iptables-nft",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7615-common",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-mt7915e",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-nat6",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-nat6",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb3",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-https-dns-proxy",
        "luci-app-opkg",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "tailscale",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "vim",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.210.54153-69231a7"
}

kitchung, 2022-09-07:
I received same error for same package, except for askey,rt4230w-rev6. LMK if this should be a different issue and paste of my log.

omijh, 2022-09-08:
I got the same issue as well