openwrt/asu #388: Docker image for ARM board

lpyparmentier, 2022-09-07:
Hello @aparcar,

Very cool project, I've tried to self-host it, but I only have ARM boards, and it seems that the image `aparcar/asu-server` is not available for this kind of architecture yet. Just wanted to let you know in case there is any chance to publish them, I will stay in touch :)

dangowrt, 2022-09-07:
While asu itself could quite easily run on a variety of platforms including 32-bit and 64-bit ARM it relies on using the OpenWrt ImageBuilder[1] to create firmware images, and that released only for 64-bit x86 (as that happens to be the architecture the buildbots are running on and hence bundled host tools are built for that). So unless you want to run static QEMU emulating x86_64 for those IB-bundled tools, running asu on any architecture other than x86_64 is not an option at this point.

[1]: available for each target platform on https://downloads.openwrt.org/

lpyparmentier, 2022-09-07:
Thanks for clarifications :)

closed 2022-09-09 by lpyparmentier