openwrt/asu #389: "Signature check failed." for packages/Packages -> "opkg_install_cmd: Cannot install package iptables."

weissel, 2022-09-08:
Trying attended sys upgrade for a 21.02 (target: ramips/mt7621) to 22.03.0.
Fails with error.  (Expected: no error, generating an update file, etc.)

It looks like the builder has no current package list, tries to download it but runs into a signature error with "packages/Packages".
See at bottom of the return.   (I took the freedom to **bolden** what I believe are the important parts.)

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16554-1d4dea6d4f",
    "advanced_mode": "1",
    "app_version": "git-22.090.55699-bb6ef9f",
    "branch": "21.02",
    "efi": null,
    "profile": "asus,rt-ac57u",
    "target": "ramips/mt7621",
    "version": "22.03.0",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd-mod-bind",
        "collectd-mod-conntrack",
        "collectd-mod-contextswitch",
        "collectd-mod-cpu",
        "collectd-mod-csv",
        "collectd-mod-curl",
        "collectd-mod-df",
        "collectd-mod-disk",
        "collectd-mod-dns",
        "collectd-mod-entropy",
        "collectd-mod-ethstat",
        "collectd-mod-filecount",
        "collectd-mod-fscache",
        "collectd-mod-interface",
        "collectd-mod-ipstatistics",
        "collectd-mod-iptables",
        "collectd-mod-irq",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-netlink",
        "collectd-mod-ntpd",
        "collectd-mod-ping",
        "collectd-mod-processes",
        "collectd-mod-protocols",
        "collectd-mod-smart",
        "collectd-mod-snmp",
        "collectd-mod-sqm",
        "collectd-mod-swap",
        "collectd-mod-tcpconns",
        "collectd-mod-ted",
        "collectd-mod-thermal",
        "collectd-mod-unixsock",
        "collectd-mod-uptime",
        "collectd-mod-wireless",
        "collectd-mod-write-graphite",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip6tables",
        "iptables",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-mt7603",
        "kmod-mt76x2",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb3",
        "less",
        "libblobmsg-json",
        "libc",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
        "libwolfssl",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-commands",
        "luci-app-dcwapd",
        "luci-app-ddns",
        "luci-app-diag-core",
        "luci-app-firewall",
        "luci-app-nlbwmon",
        "luci-app-ntpc",
        "luci-app-opkg",
        "luci-app-qos",
        "luci-app-shairplay",
        "luci-app-sqm",
        "luci-app-statistics",
        "luci-app-uhttpd",
        "luci-app-upnp",
        "luci-app-vnstat",
        "luci-app-watchcat",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "ubi-utils",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs"
}

STDERR:

**Package list missing or not up-to-date, generating it.**

Building package index...
Downloading https://downloads.openwrt.org/releases/22.03.0/targets/ramips/mt7621/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/22.03.0/targets/ramips/mt7621/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
**Downloading file:packages/Packages**
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/ramips/mt7621/dl/imagebuilder
**Downloading file:packages/Packages.sig**
**Signature check failed.**
Remove wrong Signature file.
Collected errors:
 * opkg_install_cmd: Cannot install package iptables.
make[2]: *** [Makefile:169: package_install] Error 255
make[1]: *** [Makefile:134: _call_manifest] Error 2
make: *** [Makefile:253: manifest] Error 2

aparcar, 2022-09-12:
@hauke do you have an idea here?

RoganDawes, 2022-09-14:
Related to move from iptables to nftables in 22.03?

aparcar, 2023-02-04:
Yea I think I-tables should become iptables-bpf or something, @dangowrt can you comment on this?