openwrt/asu #391: Server response: Unsupported package(s): luci-theme-argon-master, luci-app-banip

deaglenomics, 2022-09-10:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16554-1d4dea6d4f",
    "advanced_mode": "1",
    "branch": "21.02",
    "efi": null,
    "profile": "qemu-standard-pc-q35-ich9-2009",
    "target": "x86/64",
    "version": "22.03.0",
    "packages": [
        "atop",
        "base-files",
        "bash",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "coreutils-timeout",
        "curl",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "git",
        "git-http",
        "grub2",
        "grub2-efi",
        "htop",
        "ip6tables",
        "iperf3",
        "iptables",
        "irqbalance",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-bnx2",
        "kmod-button-hotplug",
        "kmod-e1000",
        "kmod-e1000e",
        "kmod-forcedeth",
        "kmod-fs-vfat",
        "kmod-igb",
        "kmod-input-core",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-ixgbe",
        "kmod-lib-crc-ccitt",
        "kmod-netatop",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-r8169",
        "kmod-slhc",
        "libblkid",
        "libblobmsg-json",
        "libcomerr",
        "libcurl",
        "libext2fs",
        "libf2fs",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libnl-tiny",
        "libopenssl",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
        "libuuid",
        "libwolfssl",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-banip",
        "luci-app-ddns",
        "luci-app-diag-core",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-sqm",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-argon-master",
        "luci-theme-bootstrap",
        "luci-theme-openwrt-2020",
        "mkf2fs",
        "mtd",
        "netdata",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "partx-utils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "px5g-wolfssl",
        "qemu-ga",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "sqm-scripts",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "ext4",
    "client": "luci/git-22.210.54423-3b0b325"
}

aparcar, 2022-09-10:
`luci-theme-argon-master` is not a package available in the official OpenWrt repositories, the server therefore can't install it.

deaglenomics, 2022-09-10:
yep , already worked that out but ban-ip was and is but was marked as unsupported

aparcar, 2022-09-10:
Feel free to suggest adding the theme as an official package.

closed 2023-05-25 by deaglenomics