openwrt/asu #394: 21.02.3 => 22.03.0, nanopi-r2s: Impossible package selection: firewall not in manifest

sdirkwinkel, 2022-09-12:
```
root@OpenWrt:~# auc -b 22.03
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.3 r16554-1d4dea6d4f on rockchip/armv8 (friendlyarm,nanopi-r2s)
Available: 22.03.0 r19685-512e76967f
Requesting package lists...
 openconnect: 8.20-1 -> 9.01-1
 terminfo: 6.2-3 -> 6.3-1
 luci-proto-wireguard: git-21.243.21928-71fe35c -> git-22.213.40001-18e6040
 liblucihttp: 2021-06-11-3dc89af4-1 -> 2022-07-08-6e68a106-1
 mkf2fs: 1.14.0-1 -> 1.14.0-3
 kmod-crypto-gf128: 5.4.188-1 -> 5.10.138-1
 opkg: 2021-06-13-1bf042dd-1 -> 2022-02-24-d038e5b6-1
 kmod-usb-core: 5.4.188-1 -> 5.10.138-1
 luci-app-opkg: git-22.154.41894-1cf976c -> git-22.154.41881-28e92e3
 ubus: 2021-06-30-4fc532c8-2 -> 2022-06-01-2bebf93c-1
 kmod-crypto-manager: 5.4.188-1 -> 5.10.138-1
installed package kmod-crypto-pcompress cannot be found in remote list!
 libuci: 2021-04-14-4b3db117-5 -> 2021-10-22-f84f49f0-6
 rpcd: 2022-02-19-8d26a1ba-1 -> 2022-08-24-82904bd4-1
 busybox: 1.33.2-2 -> 1.35.0-3
 kmod-crypto-ctr: 5.4.188-1 -> 5.10.138-1
 ddns-scripts-services: 2.8.2-12 -> 2.8.2-25
 libubus-lua: 2021-06-30-4fc532c8-2 -> 2022-06-01-2bebf93c-1
 kmod-crypto-hash: 5.4.188-1 -> 5.10.138-1
 kmod-nf-reject6: 5.4.188-1 -> 5.10.138-1
 kmod-tun: 5.4.188-1 -> 5.10.138-1
 kmod-crypto-aead: 5.4.188-1 -> 5.10.138-1
 libiwinfo-lua: 2022-04-26-dc6847eb-1 -> 2022-08-19-0dad3e66-1
 libtasn1: 4.16.0-1 -> 4.16.0-2
 luci-mod-system: git-22.130.00635-21f99bd -> git-22.140.66206-02913be
 kmod-nf-flow: 5.4.188-1 -> 5.10.138-1
 libnl-tiny: 2020-08-05-c291088f-2 -> 2021-11-21-8e0555fb-1
 kmod-lib-crc-ccitt: 5.4.188-1 -> 5.10.138-1
 getrandom: 2020-10-25-9ef88681-2 -> 2021-08-03-205defb5-2
 luci-theme-bootstrap: git-22.084.39047-f1d687e -> git-22.141.59265-d8ecf48
 kmod-pppoe: 5.4.188-1 -> 5.10.138-1
 libuuid: 2.36.1-2 -> 2.37.3-3
 kmod-pppox: 5.4.188-1 -> 5.10.138-1
 luci-app-wireguard: git-20.244.42172-21563a2 -> git-21.322.66896-8ae208d
 kmod-ipt-conntrack: 5.4.188-1 -> 5.10.138-1
 kmod-nf-reject: 5.4.188-1 -> 5.10.138-1
 base-files: 1442-r16554-1d4dea6d4f -> 1490-r19685-512e76967f
 kmod-nf-nat: 5.4.188-1 -> 5.10.138-1
 libubus: 2021-06-30-4fc532c8-2 -> 2022-06-01-2bebf93c-1
 libjson-script: 2021-05-16-b14c4688-2 -> 2022-05-15-d2223ef9-1
 partx-utils: 2.36.1-2 -> 2.37.3-3
 netifd: 2021-10-30-8f82742c-1 -> 2022-08-25-76d2d41b-1
 libf2fs: 1.14.0-1 -> 1.14.0-3
 ddns-scripts: 2.8.2-12 -> 2.8.2-25
 libsmartcols: 2.36.1-2 -> 2.37.3-3
 uboot-envtools: 2021.01-13 -> 2022.01-31
 tailscaled: 1.12.3-1 -> 1.24.2-1
 dnsmasq: 2.85-8 -> 2.86-14
 kmod-mii: 5.4.188-1 -> 5.10.138-1
 procd: 2021-03-08-2cfc26f8-1 -> 2022-06-01-7a009685-1
 kmod-crypto-hmac: 5.4.188-1 -> 5.10.138-1
 kmod-crypto-sha256: 5.4.188-1 -> 5.10.138-1
 ubusd: 2021-06-30-4fc532c8-2 -> 2022-06-01-2bebf93c-1
 libwolfssl: 5.2.0-stable-2 -> 5.4.0-stable-5
 px5g-wolfssl: 3 -> 4
 kmod-iptunnel: 5.4.188-1 -> 5.10.138-1
 luci-mod-status: git-22.089.70019-d4f0b06 -> git-22.189.48501-6731190
 libxtables: 1.8.7-1 -> 1.8.7-7
 kmod-crypto-seqiv: 5.4.188-1 -> 5.10.138-1
 luci-app-ddns: git-22.123.50005-9139ad4 -> git-22.193.49501-e0f0502
 kmod-crypto-lib-chacha20poly1305: 5.4.188-1 -> 5.10.138-1
 firewall: 2021-03-23-61db17ed-1.1 -> 2022-02-17-4cd7d4f3-3
 kmod-crypto-lib-poly1305: 5.4.188-1 -> 5.10.138-1
[... 306 lines cut ...]
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/aarch64_generic/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/rockchip/armv8/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/rockchip/armv8/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/aarch64_generic/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/aarch64_generic/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/rockchip/armv8/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/rockchip/armv8/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/aarch64_generic/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/rockchip/armv8/build_dir/target-aarch64_generic_musl/root-rockchip/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/rockchip/armv8/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://downloads.openwrt.org/releases/22.03.0/targets/rockchip/armv8/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://downloads.openwrt.org/releases/22.03.0/packages/aarch64_generic/base). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://downloads.openwrt.org/releases/22.03.0/packages/aarch64_generic/luci). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://downloads.openwrt.org/releases/22.03.0/packages/aarch64_generic/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://downloads.openwrt.org/releases/22.03.0/packages/aarch64_generic/routing). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_telephony https://downloads.openwrt.org/releases/22.03.0/packages/aarch64_generic/telephony). Skipping.
Error: Impossible package selection: firewall not in manifest
Bad message (74)

```

RoganDawes, 2022-09-14:
Getting similar/same errors with a Xiaomi 4A Gigabit router, configured as a dumb AP (i.e. no firewall running).

macherius, 2022-09-17:
Same symptoms for me on Linksys wrt1200ac. The initial switch from 21.02.3 to 22.03-SNAPSHORT worked find. After that multiple updates using "auc" were made and caused no problem. At some point errors started to show, currently updates are stuck.

`root@lede:/# auc -n
auc (0.2.5-2)
Server:    https://sysupgrade.openwrt.org
Running:   22.03-SNAPSHOT r19697-8ff1d27e48 on mvebu/cortexa9 (linksys,wrt1200ac)
Available: 22.03-SNAPSHOT r19712-81388f74b5
Requesting package lists...
 base-files: 1491-r19697-8ff1d27e48 -> 1491-r19712-81388f74b5
Are you sure you want to continue the upgrade process? [N/y] y
adblock - 4.1.4-2
attendedsysupgrade-common - 7
auc - 0.2.5-2
base-files - 1491-r19712-81388f74b5
busybox - 1.35.0-3
ca-bundle - 20211016-1
cgi-io - 2022-08-10-901b0f04-21
coreutils - 9.0-2
coreutils-sort - 9.0-2
dnsmasq - 2.86-14
dropbear - 2022.82-2
firewall4 - 2022-09-01-f5fcdcf2-1
fstools - 2022-06-02-93369be0-2
fwtool - 2019-11-12-8f7fe925-1
getrandom - 2021-08-03-205defb5-2
hostapd-common - 2022-01-16-cff80b4f-12
iw - 5.16-1
iwinfo - 2022-08-19-0dad3e66-1
jansson - 2.13.1-2
jshn - 2022-05-15-d2223ef9-1
jsonfilter - 2018-02-04-c7e938d6-1
kernel - 5.10.138-1-0de74e0cf2a0842ac532cfb36c232337
kmod-cfg80211 - 5.10.138+5.15.58-1-1
kmod-crypto-aead - 5.10.138-1
kmod-crypto-ccm - 5.10.138-1
kmod-crypto-cmac - 5.10.138-1
kmod-crypto-crc32c - 5.10.138-1
kmod-crypto-ctr - 5.10.138-1
kmod-crypto-gcm - 5.10.138-1
kmod-crypto-gf128 - 5.10.138-1
kmod-crypto-ghash - 5.10.138-1
kmod-crypto-hash - 5.10.138-1
kmod-crypto-hmac - 5.10.138-1
kmod-crypto-manager - 5.10.138-1
kmod-crypto-null - 5.10.138-1
kmod-crypto-rng - 5.10.138-1
kmod-crypto-seqiv - 5.10.138-1
kmod-crypto-sha256 - 5.10.138-1
kmod-gpio-button-hotplug - 5.10.138-3
kmod-ip6tables - 5.10.138-1
kmod-ipt-conntrack - 5.10.138-1
kmod-ipt-core - 5.10.138-1
kmod-ipt-nat - 5.10.138-1
kmod-ipt-offload - 5.10.138-1
kmod-lib-crc-ccitt - 5.10.138-1
kmod-lib-crc32c - 5.10.138-1
kmod-mac80211 - 5.10.138+5.15.58-1-1
kmod-mwlwifi - 5.10.138+2020-02-06-a2fd00bb-2
kmod-nf-conntrack - 5.10.138-1
kmod-nf-conntrack6 - 5.10.138-1
kmod-nf-flow - 5.10.138-1
kmod-nf-ipt - 5.10.138-1
kmod-nf-ipt6 - 5.10.138-1
kmod-nf-log - 5.10.138-1
kmod-nf-log6 - 5.10.138-1
kmod-nf-nat - 5.10.138-1
kmod-nf-reject - 5.10.138-1
kmod-nf-reject6 - 5.10.138-1
kmod-nfnetlink - 5.10.138-1
kmod-nft-core - 5.10.138-1
[... 121 lines cut ...]
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03-SNAPSHOT/mvebu/cortexa9/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03-SNAPSHOT/mvebu/cortexa9/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a9_vfpv3-d16/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03-SNAPSHOT/mvebu/cortexa9/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03-SNAPSHOT/mvebu/cortexa9/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a9_vfpv3-d16/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03-SNAPSHOT/mvebu/cortexa9/build_dir/target-arm_cortex-a9+vfpv3-d16_musl_eabi/root-mvebu/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03-SNAPSHOT/mvebu/cortexa9/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/mvebu/cortexa9/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a9_vfpv3-d16/base). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_kmods https://downloads.openwrt.org/releases/22.03-SNAPSHOT/targets/mvebu/cortexa9/kmods/5.10.138-1-0de74e0cf2a0842ac532cfb36c232337). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a9_vfpv3-d16/luci). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a9_vfpv3-d16/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a9_vfpv3-d16/routing). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_telephony https://downloads.openwrt.org/releases/22.03-SNAPSHOT/packages/arm_cortex-a9_vfpv3-d16/telephony). Skipping.
Error: Impossible package selection: firewall not in manifest
Bad message (74)
root@lede:/# 
`

macherius, 2022-09-17:
@sdirkwinkel @RoganDawes
There's an important difference between your update attempt and mine. Yet all three of us end up with the same finale error message. You try to update 21.02 to 22.03 while that already succeeded for me some time ago. What I try is merely an update from one version of the base files of 22.03.0 to another in the -SNAPSHOT branch.

My best guess that the problem was introduced a week ago when @aparcar  changed configuration on the update server machine. As such it strikes on bot intra-release and cross-release updates.

aparcar, 2022-09-17:
Please elaborate on your test case I I can check if the CI figuration change caused the issue

macherius, 2022-09-17:
I'll happily test anything and provide any additional information needed to help. Yet as for test case, what do you mean by that? Some days ago "auc" simply stopped working for me. It had successfully done the 21->22 upgrade and several upgrades within 22 before. I'm not aware of anything that changed on my side. No new packages or new config was involved.

lpyparmentier, 2022-09-18:
Similar issue on `ramips/mt7621`

jannoke, 2022-09-21:
Same here on TP-Link Archer C6 v2 (EU/RU/JP).

macherius, 2022-10-02:
Has this problem been triaged? I'm stuck with a broken auc command for some weeks and would apply any workaround possible. This includes a fresh install if need be. Normal updates via opkg work just fine, so I assume the OS is not broken. Is there any way to reset caches, manually force an update of sorts to a stable level or similar? Will a manual reset to the 22.03 release version followed by an auc upgrade to "22.03-SNAPSHOT" work and remove the problem?

aparcar, 2022-10-03:
@dangowrt could you please extend the auc to handle package_changes as we previously discussed? I'll add a v2 which later today to handle those requests

macherius, 2022-10-13:
`auc` has started to work again today on my box. Thank you!

```
root@lede:~# auc 
auc (0.2.5-2)
Server:    https://sysupgrade.openwrt.org
Running:   22.03-SNAPSHOT r19697-8ff1d27e48 on mvebu/cortexa9 (linksys,wrt1200ac)
Available: 22.03-SNAPSHOT r19792-b6487c3ccc
Requesting package lists...
 kmod-crypto-gf128: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-manager: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-ctr: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-hash: 5.10.138-1 -> 5.10.146-1
 kmod-nf-reject6: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-aead: 5.10.138-1 -> 5.10.146-1
 kmod-nf-flow: 5.10.138-1 -> 5.10.146-1
 kmod-lib-crc-ccitt: 5.10.138-1 -> 5.10.146-1
 kmod-pppoe: 5.10.138-1 -> 5.10.146-1
 kmod-pppox: 5.10.138-1 -> 5.10.146-1
 kmod-ipt-conntrack: 5.10.138-1 -> 5.10.146-1
 kmod-nf-reject: 5.10.138-1 -> 5.10.146-1
 base-files: 1491-r19697-8ff1d27e48 -> 1493-r19792-b6487c3ccc
 kmod-nf-nat: 5.10.138-1 -> 5.10.146-1
 uboot-envtools: 2022.01-31 -> 2022.01-32
 kmod-crypto-hmac: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-sha256: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-seqiv: 5.10.138-1 -> 5.10.146-1
 kmod-nf-ipt: 5.10.138-1 -> 5.10.146-1
 kmod-ip6tables: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-ghash: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-ccm: 5.10.138-1 -> 5.10.146-1
 kmod-nf-log: 5.10.138-1 -> 5.10.146-1
 kmod-gpio-button-hotplug: 5.10.138-3 -> 5.10.146-3
 kmod-nf-log6: 5.10.138-1 -> 5.10.146-1
 kmod-mac80211: 5.10.138+5.15.58-1-1 -> 5.10.146+5.15.58-1-1
 kmod-mwlwifi: 5.10.138+2020-02-06-a2fd00bb-2 -> 5.10.146+2020-02-06-a2fd00bb-2
 kmod-ipt-core: 5.10.138-1 -> 5.10.146-1
 kmod-ppp: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-null: 5.10.138-1 -> 5.10.146-1
 kmod-nf-conntrack: 5.10.138-1 -> 5.10.146-1
 kmod-nf-ipt6: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-gcm: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-cmac: 5.10.138-1 -> 5.10.146-1
 kmod-crypto-rng: 5.10.138-1 -> 5.10.146-1
 kmod-nf-conntrack6: 5.10.138-1 -> 5.10.146-1
 kmod-ipt-offload: 5.10.138-1 -> 5.10.146-1
 kmod-slhc: 5.10.138-1 -> 5.10.146-1
 kmod-cfg80211: 5.10.138+5.15.58-1-1 -> 5.10.146+5.15.58-1-1
 kmod-ipt-nat: 5.10.138-1 -> 5.10.146-1
Are you sure you want to continue the upgrade process? [N/y] y

Downloading image from https://sysupgrade.openwrt.org/store/13f33cc82b365a8d9daf34f0a14d5d81/openwrt-22.03-snapshot-r19792-b6487c3ccc-47bdabb66c56-mvebu-cortexa9-linksys_wrt1200ac-squashfs-sysupgrade.bin
Writing to 'openwrt-22.03-snapshot-r19792-b6487c3ccc-47bdabb66c56-mvebu-cortexa9-linksys_wrt1200ac-squashfs-sysupgrade.bin'
image verification succeeded
invoking sysupgrade
```

sdirkwinkel, 2022-11-07:
I'll close this since I can't test anymore (upgraded using webui) and @macherius reported it works now

closed 2022-11-07 by sdirkwinkel