openwrt/asu #396: 22.03.0 firmware not found on various routers

sirweazel, 2022-09-13:
I posted to the Pinned Comment that was later unpinned and closed.  So i'm guessing my issue is different?  It says that the issue was fixed.  Do i need to manually update the package on my various routers?

Comment says fixed and closed on Aug 2. I might not understand the SSL problem though. Do i need to update the luci-app-attendedsysupgrade package for the fix to apply?  Currently when checking for an update, i do not get any available. 

The Screenshots are from 1 router, but I have a few.  The others are Netgear WNDR3800 routers, but they also say no firmware updates available. 

[image]
[image]
[image]

_Originally posted by @sirweazel in https://github.com/openwrt/asu/issues/375#issuecomment-1243096376_

geminis3, 2022-09-21:
auc -b 22.03

sirweazel, 2022-09-22:
Thank you for the reply.  I can try that and report back, but does that mean there is a problem with the GUI components?  Should i just run "auc" without arguments, and report back what it finds to see if matches what the GUI is reporting instead of specifying a version?  I'm not sure if I'm encountering a bug or lack of education on my side.  I can force the update via various means, but if i found a bug, I'd like to provide as much information as possible or a solution in the post for others that encounter similar issue.

sirweazel, 2022-10-21:
I found the answer i was looking for. I had to enable advanced mode in configuration for the option to change releases to be available. Thank you.

closed 2022-10-21 by sirweazel