openwrt/asu #397: Signature check fails

muellerj, 2022-09-15:
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16495-bf0c965af0",
    "advanced_mode": "0",
    "branch": "21.02",
    "efi": null,
    "request_hash": "dec705b3c176ae20dc5d83299c87194e",
    "profile": "raspberrypi,4-model-b",
    "target": "bcm27xx/bcm2711",
    "version": "21.02.3",
    "packages": [
        "adguardhome",
        "base-files",
        "bash",
        "bcm27xx-gpu-fw",
        "bind-dig",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "curl",
        "cypress-firmware-43455-sdio",
        "cypress-nvram-43455-sdio-rpi-4b",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall",
        "fish",
        "fstools",
        "git",
        "git-http",
        "ip6tables",
        "iptables",
        "irqbalance",
        "iwinfo",
        "kernel",
        "kmod-brcmfmac",
        "kmod-fs-vfat",
        "kmod-ipt-offload",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-sound-arm-bcm2835",
        "kmod-sound-core",
        "kmod-usb-hid",
        "kmod-usb-net-rtl8152",
        "libc",
        "libcurl",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libopenssl",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lsblk",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-opkg",
        "luci-app-wireguard",
        "luci-base",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "mkf2fs",
        "mtd",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssh-sftp-server",
        "opkg",
[... 685 lines cut ...]
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../../../../../../../../home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../../../../../../../../home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../../../../../../../../home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../../../../../../../../home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
error: ext4_allocate_best_fit_partial: failed to allocate 2 blocks, out of space?
make[3]: *** [/home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/include/image.mk:334: /home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/root.ext4] Error 1
make[2]: *** [Makefile:186: build_image] Error 2
make[1]: *** [Makefile:124: _call_image] Error 2
make: *** [Makefile:240: image] Error 2
```

Imwithstupid007, 2022-09-22:
I don't know if this is the same error but Attended Sysupgrade on my RT3200 (OpenWrt SNAPSHOT r20638-cf2c9498be) is failing for a week. Always on "LibwolfSSL'.

{
    "url": "https://sysupgrade.openwrt.org/",
    "revision": "r20638-cf2c9498be",
    "advanced_mode": "1",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "linksys,e8450-ubi",
    "target": "mediatek/mt7622",
    "version": "SNAPSHOT",
    "packages": [
        "attendedsysupgrade-common",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "collectd",
        "collectd-mod-cpu",
        "collectd-mod-cpufreq",
        "collectd-mod-interface",
        "collectd-mod-iwinfo",
        "collectd-mod-load",
        "collectd-mod-memory",
        "collectd-mod-network",
        "collectd-mod-rrdtool",
        "collectd-mod-thermal",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "htop",
        "https-dns-proxy",
        "irqbalance",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7615-common",
        "kmod-mt7615-firmware",
[... 141 lines cut ...]
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/build_dir/target-aarch64_cortex-a53_musl/root-mediatek/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/mediatek/mt7622/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * opkg_install_cmd: Cannot install package libwolfssl.
make[2]: *** [Makefile:169: package_install] Error 255
make[1]: *** [Makefile:134: _call_manifest] Error 2
make: *** [Makefile:253: manifest] Error 2`

muellerj, 2022-09-30:
Tried again today, got a different error:

~~~
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16495-bf0c965af0",
    "advanced_mode": "0",
    "branch": "21.02",
    "efi": null,
    "profile": "raspberrypi,4-model-b",
    "target": "bcm27xx/bcm2711",
    "version": "21.02.3",
    "packages": [
        "adguardhome",
        "auc",
        "base-files",
        "bash",
        "bcm27xx-gpu-fw",
        "bind-dig",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "curl",
        "cypress-firmware-43455-sdio",
        "cypress-nvram-43455-sdio-rpi-4b",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall",
        "fish",
        "fstools",
        "git",
        "git-http",
        "ip6tables",
        "iptables",
        "irqbalance",
        "iwinfo",
        "kernel",
        "kmod-brcmfmac",
        "kmod-fs-vfat",
        "kmod-ipt-offload",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-sound-arm-bcm2835",
        "kmod-sound-core",
        "kmod-usb-hid",
        "kmod-usb-net-rtl8152",
        "less",
        "libc",
        "libcurl",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libopenssl",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lsblk",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-opkg",
        "luci-app-wireguard",
        "luci-base",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "mkf2fs",
        "mtd",
        "netifd",
        "odhcp6c",
[... 694 lines cut ...]
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../../../../../../../../home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../../../../../../../../home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../../../../../../../../home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/21.02.3/packages/aarch64_cortex-a72/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/build_dir/target-aarch64_cortex-a72_musl/root-bcm27xx/../../../../../../../../../../../home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
error: ext4_allocate_best_fit_partial: failed to allocate 1 blocks, out of space?
make[3]: *** [/home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/include/image.mk:334: /home/aparcar/asu/worker1/cache/21.02.3/bcm27xx/bcm2711/build_dir/target-aarch64_cortex-a72_musl/linux-bcm27xx_bcm2711/root.ext4] Error 1
make[2]: *** [Makefile:186: build_image] Error 2
make[1]: *** [Makefile:124: _call_image] Error 2
make: *** [Makefile:240: image] Error 2
~~~

aparcar, 2023-02-04:
Is this still a thing? I think there were issues with libwolfssl some time in the past.

Imwithstupid007, 2023-02-05:
Seems to work fine for the past few months (at least on snapshot).

closed 2023-02-05 by aparcar