openwrt/asu #399: Failure to update on 19.07.7 with luci and cli command

trichwan, 2022-09-29:
Luci error:
>Error connecting to upgrade server
Could not reach API at "https://sysupgrade.openwrt.org". Please try again later.
SyntaxError: JSON.parse: unexpected end of data at line 1 column 1 of the JSON data

auc error:

>auc (0.1.4)
requesting:
{
	"distro": "OpenWrt",
	"target": "ath79/generic",
	"version": "19.07.7",
	"revision": "r11306-c4a6851c72",
	"installed": {
		"terminfo": "6.1-5",
		"luci-proto-wireguard": "git-21.069.16534-310f532-1",
		"iwinfo": "2019-10-16-07315b6f-1",
		"liblucihttp0": "2019-07-05-a34a17d5-1",
		"cgi-io": "19",
		"libc": "1.1.24-2",
		"opkg": "2021-01-31-c5dccea9-1",
		"kmod-usb-core": "4.14.221-1",
		"luci-app-opkg": "git-21.044.30835-34e0d65-1",
		"wpad-basic": "2019-08-08-ca8c2bd2-5",
		"libpthread": "1.1.24-2",
		"ubus": "2019-12-27-041c9d1c-1",
		"iw": "5.0.1-1",
		"libuci20130104": "2019-09-01-415f9e48-4",
		"rpcd": "2020-05-26-67c8a3fd-1",
		"busybox": "1.30.1-6",
		"luci-lib-ip": "git-21.044.30835-34e0d65-1",
		"libubus-lua": "2019-12-27-041c9d1c-1",
		"kmod-nf-reject6": "4.14.221-1",
		"libiwinfo-lua": "2019-10-16-07315b6f-1",
		"swconfig": "12",
		"libubus20210603": "2022-02-21-b32a0e17-1",
		"luci-mod-system": "git-21.044.30835-34e0d65-1",
		"kmod-nf-flow": "4.14.221-1",
		"kmod-lib-crc-ccitt": "4.14.221-1",
		"getrandom": "2019-06-16-4df34a4d-3",
		"luci-theme-bootstrap": "git-21.044.30835-34e0d65-1",
		"kmod-pppoe": "4.14.221-1",
		"libcap": "2.27-1",
		"kmod-pppox": "4.14.221-1",
		"luci-app-wireguard": "git-21.069.16534-310f532-1",
		"kmod-ipt-conntrack": "4.14.221-1",
		"kmod-nf-reject": "4.14.221-1",
		"base-files": "204.2-r11306-c4a6851c72",
		"kmod-nf-nat": "4.14.221-1",
		"libpcre": "8.43-1",
		"netifd": "2021-01-09-753c351b-1",
		"kmod-ipt-raw": "4.14.221-1",
		"ddns-scripts": "2.7.8-13",
		"uboot-envtools": "2018.03-3.1",
		"dnsmasq": "2.80-16.3",
		"procd": "2020-03-07-09b9bd82-1",
		"kmod-phy-ath79-usb": "4.14.221-1",
		"libuclient20160123": "2020-06-17-51e16ebf-1",
		"ubusd": "2019-12-27-041c9d1c-1",
		"qrencode": "4.0.2-2",
		"libelf1": "0.177-1",
		"screen": "4.8.0-2",
		"kmod-usb-ehci": "4.14.221-1",
		"luci-mod-status": "git-21.044.30835-34e0d65-1",
		"libxtables12": "1.8.3-1",
		"libiwinfo20181126": "2019-10-16-07315b6f-1",
		"libmbedtls12": "2.16.12-1",
		"luci-app-ddns": "2.4.9-7",
		"kmod-usb2": "4.14.221-1",
		"firewall": "2019-11-22-8174814a-3",
		"kmod-ath10k-ct-smallbuffers": "4.14.221+2019-09-09-5e8cd86f-1",
		"luci-app-firewall": "git-21.044.30835-34e0d65-1",
		"kmod-ipt-conntrack-extra": "4.14.221-1",
		"kmod-nf-ipt": "4.14.221-1",
		"luci-lib-ipkg": "git-21.069.16534-310f532-1",
		"libmnl0": "1.0.4-2",
		"tc": "5.0.0-2.1",
		"tcpdump": "4.9.3-2",
		"kmod-ip6tables": "4.14.221-1",
		"odhcp6c": "2021-01-09-64e1b4e7-16",
		"fstools": "2020-05-12-84269037-1",
		"kmod-udptunnel4": "4.14.221-1",
		"uclient-fetch": "2020-06-17-51e16ebf-1",
		"kmod-udptunnel6": "4.14.221-1",
		"kmod-ath9k": "4.14.221+4.19.161-1-1",
		"iptables-mod-ipopt": "1.8.3-1",
		"uci": "2019-09-01-415f9e48-4",
		"lua": "5.1.5-3",
		"nmap": "7.70-1",
		"libip4tc2": "1.8.3-1",
		"libubox20191228": "2020-05-25-66195aee-1",
		"dropbear": "2019.78-2",
		"kmod-hwmon-core": "4.14.221-1",
		"luci-compat": "git-21.069.16534-310f532-1",
		"rpcd-mod-file": "2020-05-26-67c8a3fd-1",
		"mtd": "24",
		"odhcpd-ipv6only": "2020-05-03-49e4949c-3",
		"mtr": "0.93-1",
		"ath10k-firmware-qca988x-ct": "2020-07-02-1",
		"rpcd-mod-luci": "20201107",
		"luci-app-sqm": "1.4.0-2",
		"urandom-seed": "1.0-1",
		"luci-proto-ppp": "git-21.044.30835-34e0d65-1",
		"luci-mod-admin-full": "git-21.044.30835-34e0d65-1",
		"ppp": "2.4.7.git-2019-05-25-3",
		"luci-base": "git-21.044.30835-34e0d65-1",
		"kmod-gpio-button-hotplug": "4.14.221-3",
		"logd": "2019-06-16-4df34a4d-3",
		"kmod-wireguard": "4.14.221+1.0.20200611-2",
		"kmod-mac80211": "4.14.221+4.19.161-1-1",
		"wireguard-tools": "1.0.20191226-1",
		"libjson-script": "2020-05-25-66195aee-1",
		"luci-proto-ipv6": "git-21.044.30835-34e0d65-1",
		"libblobmsg-json": "2020-05-25-66195aee-1",
		"libip6tc2": "1.8.3-1",
		"iptables": "1.8.3-1",
		"openwrt-keyring": "2019-07-25-8080ef34-1",
		"luci-app-attendedsysupgrade": "git-22.115.68448-712bc8e-1",
		"sqm-scripts": "1.4.0-2",
		"jshn": "2020-05-25-66195aee-1",
		"kmod-ipt-core": "4.14.221-1",
		"attendedsysupgrade-common": "0.2-1",
		"ca-certificates": "20200601-1",
		"kmod-ppp": "4.14.221-1",
		"libncurses6": "6.1-5",
		"ip-tiny": "5.0.0-2.1",
		"libpcap1": "1.9.1-2.1",
		"libjson-c2": "0.12.1-3.1",
		"uhttpd": "2020-10-01-3abcc891-1",
		"auc": "0.1.4-1",
		"kmod-nf-conntrack": "4.14.221-1",
		"iptables-mod-conntrack-extra": "1.8.3-1",
		"usign": "2020-05-23-f1f65026-1",
		"ip6tables": "1.8.3-1",
		"liblua5.1.5": "5.1.5-3",
		"zlib": "1.2.11-3",
		"kmod-nf-ipt6": "4.14.221-1",
		"luci-lib-nixio": "git-21.044.30835-34e0d65-1",
		"rpcd-mod-rpcsys": "2020-05-26-67c8a3fd-1",
		"iptraf-ng": "1.1.4-1",
		"liblucihttp-lua": "2019-07-05-a34a17d5-1",
		"luci-lib-jsonc": "git-21.044.30835-34e0d65-1",
		"kmod-ipt-ipopt": "4.14.221-1",
		"luci": "git-21.044.30835-34e0d65-1",
		"kmod-nf-conntrack6": "4.14.221-1",
		"libgcc1": "7.5.0-2",
		"libstdcpp6": "7.5.0-2",
		"libqrencode": "4.0.2-2",
		"kmod-ath": "4.14.221+4.19.161-1-1",
		"ubox": "2019-06-16-4df34a4d-3",
		"kmod-sched-cake": "4.14.221+2019-03-12-057c7388-1",
		"kernel": "4.14.221-1-b84a5a29b1d5ae1dc33ccf9ba292ca1d",
		"libnl-tiny": "0.1-5",
		"libubus20191227": "2019-12-27-041c9d1c-1",
		"rpcd-mod-iwinfo": "2020-05-26-67c8a3fd-1",
		"luci-mod-network": "git-21.044.30835-34e0d65-1",
		"kmod-ifb": "4.14.221-1",
		"kmod-nls-base": "4.14.221-1",
		"uhttpd-mod-ubus": "2020-10-01-3abcc891-1",
		"fwtool": "2",
		"jsonfilter": "2018-02-04-c7e938d6-1",
		"ucert-full": "2020-05-24-00b921d8-1",
		"hostapd-common": "2019-08-08-ca8c2bd2-5",
		"iperf3": "3.7-1",
		"kmod-ipt-offload": "4.14.221-1",
		"kmod-ath9k-common": "4.14.221+4.19.161-1-1",
		"wireless-regdb": "2020.11.20-1",
		"urngd": "2020-01-21-c7f7b6b6-1",
		"kmod-slhc": "4.14.221-1",
		"kmod-cfg80211": "4.14.221+4.19.161-1-1",
		"libustream-mbedtls20150806": "2020-03-13-40b563b1-1",
		"rpcd-mod-rrdns": "20170710",
		"ppp-mod-pppoe": "2.4.7.git-2019-05-25-3",
		"kmod-ipt-nat": "4.14.221-1",
		"kmod-sched-core": "4.14.221-1",
		"arp-scan": "1.9.5-1"
	},
	"upgrade_packages": 1
}
headers:
{
	"access-control-allow-headers": "*",
	"access-control-allow-methods": "POST, GET, OPTIONS",
	"access-control-allow-origin": "*",
	"content-length": "205",
	"content-type": "application/problem+json",
	"date": "Thu, 29 Sep 2022 01:37:03 GMT",
	"server": "Caddy",
	"server": "gunicorn"
}
HTTP error 404
reply:
{
>	
>}
server reply invalid.
running OpenWrt 19.07.7 on ath79/generic (dlink,dir-859-a1)
checking https://sysupgrade.openwrt.org for release upgrade or updated packages

boretom, 2022-11-08:
If you check http://sysupgrade.openwrt.org there's a list of supported releases. They are 22.03, 21.02 and SNAPSHOT. 19.07 is not served there anymore.

trichwan, 2022-11-08:
I read that as what images were available on the servers to upgrade to.
I didn't read it as auc's requirements.

boretom, 2022-11-09:
I think I understand what you mean. According to the 22.03 [release page](https://openwrt.org/releases/22.03/notes-22.03.2) there's no upgrade path from 19.07 to 22.03.

Can you select 21.02 as your target in asu?

axelsimon, 2023-01-27:
@boretom It doesn't look like that's possible, the only options `auc` (0.1.4) offers are:
```
 -c	only check if system is up-to-date
 -F	ignore result of signature verification
 -d	enable debugging output
```
And the luci app has different options (show advanced, search on opening the app) but nothing to select a target.

And like @trichwan i'd expect the list of branches on the server to be what the server offers, not what the server expects you to come from. That's how a Debian repo mirror, for instance.