openwrt/asu #405: 21.02 -> 22.03 update fails due to missing swconfig

tohojo, 2022-10-11:
Can't seem to update from 21.02 to 22.03 because of a missing `swconfig` package?

```
root@uap1:~# auc -b 22.03 -y
auc (0.2.4-8)
Server:    https://sysupgrade.openwrt.org
Running:   21.02.3 r16554-1d4dea6d4f on ath79/generic (ubnt,unifiac-lite)
Available: 22.03.0 r19685-512e76967f
Requesting package lists...
 iwinfo: 2022-04-26-dc6847eb-1 -> 2022-08-19-0dad3e66-1
 liblucihttp: 2021-06-11-3dc89af4-1 -> 2022-07-08-6e68a106-1
 opkg: 2021-06-13-1bf042dd-1 -> 2022-02-24-d038e5b6-1
 luci-app-opkg: git-22.273.29015-e01e38c -> git-22.273.29004-9f6876b
 ubus: 2021-06-30-4fc532c8-2 -> 2022-06-01-2bebf93c-1
 iw: 5.9-8fab0c9e-3 -> 5.16-1
 libuci: 2021-04-14-4b3db117-5 -> 2021-10-22-f84f49f0-6
 rpcd: 2022-02-19-8d26a1ba-1 -> 2022-08-24-82904bd4-1
 busybox: 1.33.2-2 -> 1.35.0-3
 libubus-lua: 2021-06-30-4fc532c8-2 -> 2022-06-01-2bebf93c-1
 kmod-nf-reject6: 5.4.188-1 -> 5.10.138-1
 libiwinfo-lua: 2022-04-26-dc6847eb-1 -> 2022-08-19-0dad3e66-1
 luci-mod-system: git-22.264.46189-30ba277 -> git-22.264.46172-b6b7da4
 kmod-nf-flow: 5.4.188-1 -> 5.10.138-1
 libnl-tiny: 2020-08-05-c291088f-2 -> 2021-11-21-8e0555fb-1
 kmod-lib-crc-ccitt: 5.4.188-1 -> 5.10.138-1
 getrandom: 2020-10-25-9ef88681-2 -> 2021-08-03-205defb5-2
 luci-theme-bootstrap: git-22.084.39047-f1d687e -> git-22.141.59265-d8ecf48
 kmod-pppoe: 5.4.188-1 -> 5.10.138-1
 kmod-pppox: 5.4.188-1 -> 5.10.138-1
 kmod-ipt-conntrack: 5.4.188-1 -> 5.10.138-1
 kmod-nf-reject: 5.4.188-1 -> 5.10.138-1
 base-files: 1442-r16554-1d4dea6d4f -> 1490-r19685-512e76967f
 kmod-nf-nat: 5.4.188-1 -> 5.10.138-1
 libubus: 2021-06-30-4fc532c8-2 -> 2022-06-01-2bebf93c-1
 libjson-script: 2021-05-16-b14c4688-2 -> 2022-05-15-d2223ef9-1
 netifd: 2021-10-30-8f82742c-1 -> 2022-08-25-76d2d41b-1
 uboot-envtools: 2021.01-13 -> 2022.01-31
 dnsmasq: 2.85-8 -> 2.86-14
 procd: 2021-03-08-2cfc26f8-1 -> 2022-06-01-7a009685-1
 ubusd: 2021-06-30-4fc532c8-2 -> 2022-06-01-2bebf93c-1
 px5g-wolfssl: 4.1 -> 5.1
 luci-mod-status: git-22.089.70019-d4f0b06 -> git-22.189.48501-6731190
 libxtables: 1.8.7-1 -> 1.8.7-7
 firewall: 2021-03-23-61db17ed-1.1 -> 2022-02-17-4cd7d4f3-3
 luci-app-firewall: git-22.089.67741-3856d50 -> git-22.089.67563-7e3c1b4
 kmod-nf-ipt: 5.4.188-1 -> 5.10.138-1
 kmod-ip6tables: 5.4.188-1 -> 5.10.138-1
 odhcp6c: 2021-01-09-53f07e90-16 -> 2022-08-05-7d21e8d8-18
 fstools: 2021-01-04-c53b1882-1 -> 2022-06-02-93369be0-2
 kmod-ath9k: 5.4.188+5.10.110-1-1 -> 5.10.138+5.15.58-1-1
 uci: 2021-04-14-4b3db117-5 -> 2021-10-22-f84f49f0-6
 lua: 5.1.5-9 -> 5.1.5-10
 libip4tc: 1.8.7-1 -> 1.8.7-7
 dropbear: 2020.81-2 -> 2022.82-2
 kmod-hwmon-core: 5.4.188-1 -> 5.10.138-1
 rpcd-mod-file: 2022-02-19-8d26a1ba-1 -> 2022-08-24-82904bd4-1
 odhcpd-ipv6only: 2021-07-18-bc9d317f-3 -> 2022-03-22-860ca900-1
 wpad: 2020-06-08-5a8b3662-41 -> 2022-01-16-cff80b4f-13.1
 libblobmsg-json: 2021-05-16-b14c4688-2 -> 2022-05-15-d2223ef9-1
 libiwinfo-data: 2022-04-26-dc6847eb-1 -> 2022-08-19-0dad3e66-1
 luci-proto-ppp: git-21.163.64918-6c6559a -> git-21.158.38888-88b9d84
 ppp: 2.4.8.git-2020-10-03-3 -> 2.4.9.git-2021-01-04-3
 luci-base: git-22.245.77575-63bfee6 -> git-22.245.77528-487e58a
 kmod-gpio-button-hotplug: 5.4.188-3 -> 5.10.138-3
 snmpd: 5.8-2 -> 5.9.1-3
 logd: 2020-10-25-9ef88681-2 -> 2021-08-03-205defb5-2
 kmod-mac80211: 5.4.188+5.10.110-1-1 -> 5.10.138+5.15.58-1-1
 libwolfssl: 5.5.1-stable-2 -> 5.5.1-stable-3
 luci-proto-ipv6: git-21.148.49484-14511e5 -> git-21.148.48881-79947af
 libip6tc: 1.8.7-1 -> 1.8.7-7
[... 208 lines cut ...]
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/ath79/generic/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/ath79/generic/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.0/ath79/generic/build_dir/target-mips_24kc_musl/root-ath79/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.0/ath79/generic/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_core https://downloads.openwrt.org/releases/22.03.0/targets/ath79/generic/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_base https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/base). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_luci https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/luci). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_packages https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/packages). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_routing https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/routing). Skipping.
 * opkg_conf_parse_file: Duplicate src declaration (openwrt_telephony https://downloads.openwrt.org/releases/22.03.0/packages/mips_24kc/telephony). Skipping.
Error: Impossible package selection: swconfig not in manifest
Bad message (77)
```

aparcar, 2022-10-13:
Could you please update the auc package and try again?

tohojo, 2022-10-14:
Hmm, 0.2.4-8 seems to be the latest version?

```
root@uap1:~# opkg upgrade auc
root@uap1:~# opkg install auc
Package auc (0.2.4-8) installed in root is up to date.
```

Even updated to the just-released 21.02.4 version of openwrt, but that also seems to only have 0.2.4. I see there's a commit on the 21.02 branch of packages to update to 0.3.1, but that was only commited last night, so I guess it hasn't propagated to the builders yet?

sirweazel, 2022-10-19:
Do you think your issue is related to my issue?
[https://github.com/openwrt/asu/issues/396](url)
I was about to head over to the forums and then i saw your issue here.  I need to update my issue, as i'm now seeing the new 21.02.5 but not 22.03 versions. You think related, or different issue?

tohojo, 2022-11-16:
With auc 0.3.1, updating from 21.02->22.03 works - thanks! :)

closed 2022-11-16 by tohojo