openwrt/asu #415: [generic][x86] Allow setting of CONFIG_TARGET_ROOTFS_PARTSIZE

rany2, 2022-10-29:
It would be great if the client could request a custom CONFIG_TARGET_ROOTFS_PARTSIZE so that there would be no need to resize after flashing (or so in case 100MB is somehow not enough for the packages requested).

I added `[generic]` in the title because I think that this would be useful for generic x86 images.

aparcar, 2022-10-29:
Agree. This should be part of the upstream IB first

rany2, 2022-10-30:
Could asu modify `.config` with a regex instead? I'm not sure how the asu works exactly so I'm guessing that's not feasible.

aparcar, 2022-10-30:
Could you try to create a PR for this? It just needs an addition like [this](https://github.com/openwrt/asu/blob/master/asu/build.py#L275)

rany2, 2022-10-30:
I'll give it a try. From there I guess the new feature just needs to get used by auc which should be simple enough.

joshenders, 2023-03-11:
I came here to make this patch after installing a few x86 images and struggling with partition resizing. Have you had a chance to review this PR yet @aparcar?

joshenders, 2023-03-12:
Happy to see progress here as this has been a thorn in my side for years but has there been any discussion on whether it would make sense to package the x86 release as a small initrd with an installer?

commit 7a5d399c (2023-03-31): api: add rootfs_size_mb parameter

The parameter allows the client to specify a custom value for
the CONFIG_TARGET_ROOTFS_PARTSIZE. This is especially useful
for the generic X86 images.

Fixes #415
closed 2023-03-31 by aparcar