openwrt/asu #419: Error building the firmware image : Server response: Unsupported profile: 0x04CF:3500

kris969, 2022-11-05:
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19803-9a599fee93",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "0x04CF:3500",
    "target": "bcm47xx/mips74k",
    "version": "22.03.2",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-b43",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-ohci",
        "kmod-usb2",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-wireguard",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "nano",
        "netifd",
        "nftables-json",
        "nvram",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "otrx",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "swconfig",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireguard-tools",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.210.54153-69231a7"
}

aparcar, 2022-11-06:
What device are you running?

kris969, 2022-11-06:
Thank's to consider my request ;)
This issue appears a Netgear WNR3500L V1

Remarque: I'm also running Openwrt on Netgear WNDR3700 v4 devices, with no issue at all.

aparcar, 2022-11-06:
Looks rather odd to me, the value comes from [this Kernel file](https://github.com/aparcar/openwrt/commit/0c9d07a3b4964d15f10d719fa81e0d47145b5950#diff-0a0f8e4add0f520541906e1162983a87304493815067b385826cbffee0c3011cR37), however I don't know why it's taken as board_name.

@dangowrt do you have an idea about that?

kris969, 2022-11-25:
An attempt today gives the same result.
I understand this issue is still pending, isn't it?

aguslr, 2023-01-02:
I also have a Netgear WNR3500L v1 with the same issue. However, it works fine if I manually enter the correct profile name (`netgear,wnr3500l-v1`) when using *luci-app-attendedsysupgrade*.