openwrt/asu #425: Can't get extra repos working

greenbreakfast, 2022-11-25:
Hi there, I can't seem to get ASU to use my extra repos. Any help would be appreciated.

I'm using Docker compose according to the readme instructions (https://github.com/openwrt/asu#docker) on an AWS instance. 

Builds without packages from my custom repo work fine. Any builds with packages from my repo return 422 status and a `Unsupported package(s)` message.

Here is my modified config file that's in `asu-service/`:

```
from pathlib import Path

# disable test and debug features
TESTING = False
#DEBUG = False
DEBUG = True

# where to find the ImageBuildes
UPSTREAM_URL = "https://downloads.openwrt.org"

# where to store created images
STORE_PATH = Path.cwd() / "public/store/"

# where to store ImageBuilders. Do not set when multiple workers run
CACHE_PATH = None

# where to store JSON files
JSON_PATH = Path.cwd() / "public/json/v1/"

MAPPING_ABI = {"libubus20191227": "libubus"}

BRANCHES = {
  "22.03": {
    "name": "22.03",
    "enabled": True,
    "extra_keys": [
      "RWRWaPnVy9z4kIfO6MKRhEHB1KK7i8IoBA0JH9DWvjL93uId2vdUUbAp"
    ],
    "extra_repos": {
      "onion": "http://repo.onioniot.com/omega2/packages/openwrt-21.02.1/onion"
    },
    "git_branch": "openwrt-22.03",
    "branch_off_rev": 19160,
    "path": "releases/{version}",
    "path_packages": "releases/packages-{branch}",
    "pubkey": "RWRNAX5vHtXWFmt+n5di7XX8rTu0w+c8X7Ihv4oCyD6tzsUwmH0A6kO0",
    "release_date": "2022-10-17T00:00:00.000Z",
    "repos": [
      "base",
      "packages",
      "luci",
      "routing",
      "telephony",
      "onion",
    ],
    "snapshot": False,
    "updates": "features",
    "versions": [
      "22.03.2"
    ],
    "package_changes": [
      {
        "source": "kmod-nft-nat6",
        "revision": 19160,
        "mandatory": True
      },
      {
        "source": "firewall",
        "target": "firewall4",
        "revision": 18611
      }
    ]
  }
}
```

(I originally tried editing `asu/branches.yml` to add my repo but that didn't work. I also realize the package repo is for version openwrt-21.02.1 but it should still be able to detect the packages, right?)

When I build an image (without any custom packages), I see the request does have my extra_repos:

```
worker_1     | 21:46:08 Building {'profile': 'onion_omega2p',
...
'extra_repos': {'onion': 'http://repo.onioniot.com/omega2/packages/openwrt-21.02.1/onion'}, 'git_branch': 'openwrt-22.03', 
...
'repos': ['base', 'packages', 'luci', 'routing', 'telephony', 'onion'],
...
```

But then all builds that include packages from my repo return 422 status and a `Unsupported package(s)` message.
Any guidance would be appreciated!

greenbreakfast, 2022-12-14:
Bump. Would appreciate any guidance on what to try out next

aparcar, 2022-12-25:
Could you please verify that the janitor is picking up the extra package feed? It should appear in the log with a number of how many packages were found.

aparcar, 2022-12-28:
Please check the modified docs as of 15f1bfa and try again.

greenbreakfast, 2023-02-01:
Thanks @aparcar, I'm able to build firmware with packages from my custom repo now!

I had 2 follow up questions:
1. In the janitor logs there was a ton of output so I wasn't able to see if it was picking up my extra packages feed and how many it found. Is there a way to limit which targets the ASU will use/can build?
2. Do the `extra_repos` apply to all targets? Does ASU know which targets the packages in extra_repos are for?

greenbreakfast, 2023-05-10:
@aparcar any feedback on these two questions?

1. In the janitor logs there was a ton of output so I wasn't able to see if it was picking up my extra packages feed and how many it found. Is there a way to limit which targets the ASU will use/can build?
2. Do the `extra_repos` apply to all targets? Does ASU know which targets the packages in extra_repos are for?