openwrt/asu #427: ASU own Server (Docker): Build working with Swagger but not with AUC

onemarcfifty, 2022-12-27:
Followed the Readme in order to run my own Sysupgrade server in Docker. Testing with Swagger (i.e. the built-in API test, in my case `http://10.39.39.246/ui/#/default/asu.api.api_v1_build_post` with the following test data 

```
{
  "packages": [
    "auc",
    "luci-app-attendedsysupgrade"
  ],
  "profile": "8dev_carambola2",
  "target": "ath79/generic",
  "version": "22.03.2"
}

```
works well - the image gets built and I can see all the files in the ...public/store/hash/ subdirectory.

However, trying **auc** from an OpenWrt Router, auc throws a blunt `No data available (61)` - I've set the Server URL to 

```
uci set attendedsysupgrade.server.url='http://10.39.39.246'
uci commit
auc
```

(the Docker stack is answering on port 80). Looking at the screen output of  the docker stack, nothing happens - it looks like auc is not even requesting anything. I can however do a wget/curl from the router to the Docker Server (e.g. `wget http://http://10.39.39.246/` will grab the index.html)

Is there anything else that I need to tell auc in order to make it happy ?

uci shows
```
attendedsysupgrade.server=server
attendedsysupgrade.server.url='http://10.39.39.246'
attendedsysupgrade.client=client
attendedsysupgrade.client.upgrade_packages='1'
attendedsysupgrade.client.auto_search='0'
attendedsysupgrade.client.advanced_mode='1'
```
auc works well with `https://sysupgrade.openwrt.org`

Any help is appreciated - Many thanks in advance!

onemarcfifty, 2022-12-27:
Browsing through the source code of [the auc tool](https://github.com/openwrt/packages/blob/master/utils/auc/src/auc.c) it lloks like auc is trying to download a `branches.json` file which does not exist. (function `request_branches` , line 1330).

"Stealing" the json file from [the OpenWrt Sysupgrade Server](https://sysupgrade.openwrt.org/json/v1/branches.json) and putting it into .../public/json/v1 solved the issue!!!

Maybe worth adding this to the documentation - also - where does the OpenWrt Server generate that json file from ?

aparcar, 2022-12-28:
Please check the modified docs as of 15f1bfa and try again.

onemarcfifty, 2022-12-29:
Hi @aparcar , thanks for the reply. I've added `- FLASK_DEBUG=1` to the docker-compose and removed everything under asu-service/public (which creates the start condition you describe as I had done the chmod -R before). Still, the branches.json does not get created. The janitor downloads all the release and branch json files and then exits with an error:

```
janitor_1    | [2022-12-29 08:52:05,991] DEBUG in janitor: SNAPSHOT/mips64_octeonplus/luci: Found 2589 packages
janitor_1    | [2022-12-29 08:52:06,129] DEBUG in janitor: SNAPSHOT/mips64_octeonplus/routing: Found 95 packages
janitor_1    | [2022-12-29 08:52:06,424] DEBUG in janitor: SNAPSHOT/mips64_octeonplus/telephony: Found 872 packages
janitor_1    | [2022-12-29 08:52:06,519] INFO in janitor: mips64_octeonplus: found 8492 packages
janitor_1    | Traceback (most recent call last):
janitor_1    |   File "/home/build/.local/bin/flask", line 8, in <module>
janitor_1    |     sys.exit(main())
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/flask/cli.py", line 1047, in main
janitor_1    | Loading /home/build/asu/config.py
janitor_1    |     cli.main()
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/click/core.py", line 1055, in main
janitor_1    |     rv = self.invoke(ctx)
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
janitor_1    |     return _process_result(sub_ctx.command.invoke(sub_ctx))
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/click/core.py", line 1657, in invoke
janitor_1    |     return _process_result(sub_ctx.command.invoke(sub_ctx))
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/click/core.py", line 1404, in invoke
janitor_1    |     return ctx.invoke(self.callback, **ctx.params)
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/click/core.py", line 760, in invoke
janitor_1    |     return __callback(*args, **kwargs)
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/click/decorators.py", line 26, in new_func
janitor_1    |     return f(get_current_context(), *args, **kwargs)
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/flask/cli.py", line 357, in decorator
janitor_1    |     return __ctx.invoke(f, *args, **kwargs)
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/click/core.py", line 760, in invoke
janitor_1    |     return __callback(*args, **kwargs)
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/asu/janitor.py", line 457, in update
janitor_1    |     update_meta_json()
janitor_1    |   File "/home/build/.local/lib/python3.10/site-packages/asu/janitor.py", line 419, in update_meta_json
janitor_1    |     json.dumps(
janitor_1    |   File "/usr/local/lib/python3.10/json/__init__.py", line 238, in dumps
janitor_1    |     **kw).encode(obj)
janitor_1    |   File "/usr/local/lib/python3.10/json/encoder.py", line 201, in encode
janitor_1    |     chunks = list(chunks)
janitor_1    |   File "/usr/local/lib/python3.10/json/encoder.py", line 431, in _iterencode
janitor_1    |     yield from _iterencode_dict(o, _current_indent_level)
janitor_1    |   File "/usr/local/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
janitor_1    |     yield from chunks
janitor_1    |   File "/usr/local/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
janitor_1    |     yield from chunks
janitor_1    |   File "/usr/local/lib/python3.10/json/encoder.py", line 405, in _iterencode_dict
janitor_1    |     yield from chunks
janitor_1    |   File "/usr/local/lib/python3.10/json/encoder.py", line 438, in _iterencode
janitor_1    |     o = _default(o)
janitor_1    |   File "/usr/local/lib/python3.10/json/encoder.py", line 179, in default
janitor_1    |     raise TypeError(f'Object of type {o.__class__.__name__} '
janitor_1    | TypeError: Object of type date is not JSON serializable
asu_janitor_1 exited with code 1
redis_1      | 1:M 29 Dec 2022 08:52:56.070 * 10000 changes in 60 seconds. Saving...
redis_1      | 1:M 29 Dec 2022 08:52:56.072 * Background saving started by pid 21

```

Maybe this is where it wanted to create the branches.json ?

onemarcfifty, 2022-12-29:
Browsing through the asu code of janitor.py (lines 408 - 433), I assume this is where the code fails. The "branches" object contains a node 
`'release_date': datetime.date(2022, 10, 17),`
whereas it should/could be shown as 
`"release_date": "2022-10-17"`

This seems to be where lines 425... are failing when you do

```
    (current_app.config["JSON_PATH"] / "branches.json").write_text(
        json.dumps(list(branches.values()))
    )

```

aparcar, 2022-12-29:
Good catch, thanks for checking! Maybe different YANL versions work differently with handling the date, converting it to a datetime object or not.

I'll see to handle that in a more robust way, until then you could just fill it with any arbitrary string since it's not used by auc for now.

onemarcfifty, 2022-12-29:
Thanks mate. Seems to happen in `update_branch` - maybe adding something in line 100 like 

```
if isinstance(branch["release_date"],datetime.date):
   # convert to yyyy-mm-dd

```
but no rush ;-)

aparcar, 2022-12-29:
What do you think about [this](https://stackoverflow.com/a/36142844) approach? Would you be up to create a PR?

onemarcfifty, 2022-12-29:
That looks even easier, agreed. I'll give it a try and issue a PR if successful

onemarcfifty, 2022-12-29:
Now that was really easy ;-) Works like a charm ;-)

PR #428 (2022-12-30): fixed release_date in janitor.py
The janitor would fail if a branch node contained a date value. As a result, the branches.json, latest.json and overview.json files are never created and therefore acu and/or luci-app-attendedsysupgrade would not be able to update from the asu instance.

Converting json node values to string by default fixes the issue.

This fixes #427
closed 2022-12-30 by aparcar