openwrt/asu #430: Unsupported profile: linksys,wrt1900ac-v2

octe, 2023-01-10:
Hello upgrading failed with following message:

**Error building the firmware image**
Server response: Unsupported profile: linksys,wrt1900ac-v2

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:
```json
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19803-9a599fee93",
    "advanced_mode": "0",
    "branch": "22.03",
    "efi": null,
    "profile": "linksys,wrt1900ac-v2",
    "target": "mvebu/cortexa9",
    "version": "22.03.3",
    "packages": [
        "auc",
        "avahi-nodbus-daemon",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "coreutils-base64",
        "coreutils-sha1sum",
        "dnsmasq-full",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-utils",
        "hostapd-wolfssl",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-snmpd",
        "luci-app-sqm",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mosquitto-ssl",
        "mtd",
        "mwlwifi-firmware-88w8864",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssl-util",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "stubby",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.210.54153-69231a7"
}
```

makama80, 2023-01-11:
Also the WRT1200AC and WRT1900ACS (similar routers) report the same issue (have them both).

makama80, 2023-01-11:
Likely due to broken switch: [https://openwrt.org/releases/22.03/notes-22.03.3#broken_mv88e6176_switch](url)

dangowrt, 2023-01-11:
Duplicate of https://github.com/openwrt/asu/issues/429

closed 2023-01-11 by dangowrt