openwrt/asu #437: Error building the firmware image - Server response: Unsupported profile: linksys,wrt3200acm

Saijin-Naib, 2023-01-17:
Request Data:  

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19685-512e76967f",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "linksys,wrt3200acm",
    "target": "mvebu/cortexa9",
    "version": "22.03.3",
    "packages": [
        "atop",
        "auc",
        "avahi-autoipd",
        "avahi-daemon-service-http",
        "avahi-daemon-service-ssh",
        "avahi-dbus-daemon",
        "avahi-dnsconfd",
        "avahi-utils",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "irqbalance",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-btmrvl",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-cryptodev",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mwifiex-sdio",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libavahi-compat-libdnssd",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-advanced-reboot",
        "luci-app-attendedsysupgrade",
        "luci-app-commands",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-autoip",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "mwlwifi-firmware-88w8964",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssh-sftp-avahi-service",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl",
        "wsdd2"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.210.54153-69231a7"
}
```

HD7544, 2023-01-20:
This is by design until the next release of OpenWRT. See https://forum.openwrt.org/t/22-03-x-no-release-builds-for-turris-omnia-linksys-wrt-clearfog/147955 for explanation and alternatives.

Saijin-Naib, 2023-01-20:
Thanks for that.

I was unaware of the security issues and have been running 23.x since it launched 😬

Going snapshot this moment.

HD7544, 2023-01-20:
It was in the release notes, which I suppose we both should have read, but who really ever does that? ;-)