openwrt/asu #439: Error building the firmware image Server response: Error: Impossible package selection

naltabra, 2023-01-21:
Error building the firmware image
Server response: Error: Impossible package selection

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r21856-afcf1a4de4",
    "advanced_mode": "0",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "xiaomi,ax3600",
    "target": "ipq807x/generic",
    "version": "SNAPSHOT",
    "packages": [
        "ath10k-firmware-qca9887-ct",
        "ath11k-firmware-ipq8074",
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ipq-wifi-xiaomi_ax3600",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct-smallbuffers",
        "kmod-ath11k",
        "kmod-ath11k-ahb",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-michael-mic",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-qca-nss-dp",
        "kmod-qca-ssdk",
        "kmod-qrtr",
        "kmod-qrtr-smd",
        "kmod-slhc",
        "kmod-thermal",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-xhci-hcd",
        "kmod-usb3",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfsslcpu-crypto",
        "logd",
        "luci",
        "luci-proto-wireguard",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/undefined"
}
STDERR:
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/kmods/5.15.88-1-03ad7937b05d057fe30d4ef41e77ee0c/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/kmods/5.15.88-1-03ad7937b05d057fe30d4ef41e77ee0c/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
[libwolfsslcpu-crypto] Checking for Arm v8-A Cryptographic Extension support: ...[offline]... OK
[libwolfsslcpu-crypto] Checking for Arm v8-A Cryptographic Extension support: ...[offline]... OK
Collected errors:
 * check_data_file_clashes: Package libwolfsslcpu-crypto5.5.4.e624513f wants to install file /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/libwolfssl.so.5.5.4.e624513f
	But that file is already provided by package  * libwolfssl5.5.4.e624513f
 * opkg_install_cmd: Cannot install package libwolfsslcpu-crypto.
 * check_data_file_clashes: Package libwolfsslcpu-crypto5.5.4.e624513f wants to install file /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/usr/lib/libwolfssl.so.5.5.4.e624513f
	But that file is already provided by package  * libwolfssl5.5.4.e624513f
 * opkg_install_cmd: Cannot install package libwolfsslcpu-crypto.
make[2]: *** [Makefile:185: package_install] Error 255
make[1]: *** [Makefile:150: _call_manifest] Error 2
make: *** [Makefile:269: manifest] Error 2

dimfishr, 2023-01-23:
Problem is caused by package order: libustream-wolfssl is called before  libwolfsslcpu-crypto and install libwolfssl instead of  libwolfsslcpu-crypto.
Maybe revert commit https://github.com/openwrt/asu/commit/990c7bbd7882851196c51b5b287e12ff577bd46b
to allow manual package ordering?

aparcar, 2023-02-04:
I wonder what would happen if you remove libustream-wolfssl and only have libwolfsslcpu-crypto selected? Remove the determinism patch doesn't seem like a nice solution to me

dimfishr, 2023-02-05:
@aparcar yep without libustream-wolfssl build is successfully created. However this library is really necessary - all programs can't download https traffic like curl, wget etc. but the main headache is that opkg can't install anything.
OK so if you don't like manual ordering maybe we can add special character before package name like '+' ок '_' and after sort  just call map function to replace special character.
For example  libustream-wolfssl and _libwolfsslcpu-crypto will be sorted as [_libwolfsslcpu-crypto,libustream-wolfssl] 
and then just call map:
`map(lambda x: x.removeprefix('_'),sorted(req.get('packages', [])))`

aparcar, 2023-02-18:
I implemented your idea here, what do you think? https://github.com/openwrt/asu/pull/460

dimfishr, 2023-02-18:
@aparcar yeah perfect!