openwrt/asu #447: Error building the firmware image

domoclick, 2023-01-29:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r21775-c83dbcdf8f",
    "advanced_mode": "0",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "qemu-standard-pc-q35-ich9-2009",
    "target": "x86/64",
    "version": "SNAPSHOT",
    "packages": [
        "adblock",
        "base-files",
        "busybox",
        "ca-bundle",
        "ddns-scripts-cloudflare",
        "dnsmasq",
        "docker",
        "docker-compose",
        "dockerd",
        "dropbear",
        "e2fsprogs",
        "fdisk",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "grub2",
        "grub2-bios-setup",
        "grub2-efi",
        "iperf3",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-amazon-ena",
        "kmod-amd-xgbe",
        "kmod-bnx2",
        "kmod-button-hotplug",
        "kmod-crypto-acompress",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-e1000",
        "kmod-e1000e",
        "kmod-forcedeth",
        "kmod-fs-vfat",
        "kmod-igb",
        "kmod-igc",
        "kmod-input-core",
        "kmod-ixgbe",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-macvlan",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
[... 1072 lines cut ...]
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/dl/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/x86_64/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
/home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/root-x86/etc/init.d/nginx: line 12: /usr/bin/nginx-util: No such file or directory
./etc/init.d/nginx: line 12: /usr/bin/nginx-util: No such file or directory
error: ext4_allocate_best_fit_partial: failed to allocate 2184 blocks, out of space?
make[3]: *** [/home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/include/image.mk:328: /home/aparcar/asu/worker1/cache/SNAPSHOT/x86/64/build_dir/target-x86_64_musl/linux-x86_64/root.ext4] Error 1
make[2]: *** [Makefile:204: build_image] Error 2
make[1]: *** [Makefile:142: _call_image] Error 2
make: *** [Makefile:258: image] Error 2

mjnck, 2023-02-01:
Merging this PR should fix your issue https://github.com/openwrt/asu/pull/416

reukiodo, 2025-06-23:
However, this problem still seems to persist:

- https://github.com/openwrt/asu/issues/464
- https://github.com/openwrt/asu/issues/474
- https://github.com/openwrt/asu/issues/493
- https://github.com/openwrt/asu/issues/506
- https://github.com/openwrt/asu/issues/523
- https://github.com/openwrt/asu/issues/586
- https://github.com/openwrt/asu/issues/706
- https://github.com/openwrt/asu/issues/784
- https://github.com/openwrt/asu/issues/790
- https://github.com/openwrt/asu/issues/823
- https://github.com/openwrt/asu/issues/842
- https://github.com/openwrt/asu/issues/940
- https://github.com/openwrt/asu/issues/948
- https://github.com/openwrt/asu/issues/949
- https://github.com/openwrt/asu/issues/1028
- https://github.com/openwrt/asu/issues/1045
- https://github.com/openwrt/asu/issues/1197
- https://github.com/openwrt/asu/issues/1255
- https://github.com/openwrt/asu/issues/1279
- https://github.com/openwrt/asu/issues/1295
- https://github.com/openwrt/asu/issues/1298
- https://github.com/openwrt/asu/issues/1342
- https://github.com/openwrt/asu/issues/1382
- https://github.com/openwrt/asu/issues/1394

reukiodo, 2025-12-02:
unfortunately currently there is no way to specify the rootfs size in firmware-selector.openwrt.org nor the luci web UI

reukiodo, 2025-12-02:
according to https://openwrt.org/docs/guide-user/installation/sysupgrade.owut#servers the current max size is 1024 for https://sysupgrade.openwrt.org, so just switch to https://sysupgrade.guerra24.net/ to reduce headaches:

```
uci set attendedsysupgrade.server.url='https://sysupgrade.guerra24.net/'
uci set attendedsysupgrade.owut.rootfs_size=4096
uci commit
```

then use `owut upgrade` every time until the main sysupgrade and/or luci web ui gets fixed to increase the default rootfs size