openwrt/asu #462: Cannot determing currently running branch

mikeg-de, 2023-02-23:
Using both the GUI and auc via terminal get the following errors:

**Attended Sysupgrade GUI**
```
Error connecting to upgrade server

Could not reach API at "https://sysupgrade.openwrt.org/api/v1/revision/21.02-SNAPSHOT/mediatek/mt7986?1677169142601". Please try again later.

{
  "detail": "The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.",
  "status": 500,
  "title": "Internal Server Error",
  "type": "about:blank"
}
```

**auc Terminal**

```
root@Nest-Router:~# auc -c
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   21.02-SNAPSHOT unknown on mediatek/mt7986 (bananapi,bpi-r3)
WARNING: cannot determing currently running branch.
Invalid argument (22)
root@Nest-Router:~# auc -r
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   21.02-SNAPSHOT unknown on mediatek/mt7986 (bananapi,bpi-r3)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```

dangowrt, 2023-02-24:
You are not running OpenWrt from OpenWrt.org but rather SinoVoip's private fork which isn't supported. Please use the [SD card image](https://downloads.openwrt.org/snapshots/targets/mediatek/filogic/openwrt-mediatek-filogic-bananapi_bpi-r3-sdcard.img.gz) to start using official OpenWrt. You cannot upgrade from SinoVoip's fork, because the sysupgrade mechanism they use isn't compatible with official OpenWrt.

mikeg-de, 2023-02-24:
I thought, since being stored on the openwrt server, the images are accessible. Thanks anyways!

dangowrt, 2023-02-24:
The images are accessible, just `auc` can only work on official OpenWrt and cannot work on SinoVoip's proprietary OS (which is based on OpenWrt, but it's not official OpenWrt). That's because `auc` and the sysupgrade server don't know anything about the currently running image then, because it's not actually OpenWrt but rather *a proprietary OS based on OpenWrt*.

Furthermore, using official OpenWrt requires an updated bootloader, and `auc` only supports updating the OpenWrt firmware itself, it cannot take care of replacing the bootloader.