openwrt/asu #470: Error during attempted upgrade

lewinson, 2023-03-16:
The node was running a build from March 10, and I asked it for a new build.   I got:

Error building the firmware imageServer response: Unsupported package(s): libwolfssl, avahi-dbus-daemon, nftables-json, iperf3, libnftnl, avahi-daemon-service-http, snmpd, avahi-daemon-service-ssh, libmnl, libustream-wolfssl
Please report the error message and request (https://github.com/openwrt/asu/issues)Request Data:
{ "url": ["https://sysupgrade.openwrt.org"](https://sysupgrade.openwrt.org/), "revision": "r22257-166ab6f90e", "advanced_mode": "0", "branch": "SNAPSHOT", "efi": null, "profile": "netgear,wac104", "target": "ramips/mt7621", "version": "SNAPSHOT", "packages": [ "auc", "avahi-daemon-service-http", "avahi-daemon-service-ssh", "avahi-dbus-daemon", "base-files", "batctl-full", "busybox", "ca-bundle", "diffutils", "dnsmasq", "dropbear", "firewall4", "fstools", "fwtool", "getrandom", "hostapd-common", "iperf3", "iwinfo", "jansson", "jshn", "jsonfilter", "kernel", "kmod-batman-adv", "kmod-crypto-acompress", "kmod-crypto-crc32c", "kmod-crypto-hash", "kmod-gpio-button-hotplug", "kmod-leds-gpio", "kmod-lib-crc-ccitt", "kmod-lib-crc32c", "kmod-lib-lzo", "kmod-mt7603", "kmod-mt76x2", "kmod-nf-conntrack", "kmod-nf-conntrack6", "kmod-nf-flow", "kmod-nf-log", "kmod-nf-log6", "kmod-nf-nat", "kmod-nf-reject", "kmod-nf-reject6", "kmod-nfnetlink", "kmod-nft-core", "kmod-nft-fib", "kmod-nft-nat", "kmod-nft-offload", "kmod-ppp", "kmod-pppoe", "kmod-pppox", "kmod-slhc", "kmod-usb-ledtrig-usbport", "kmod-usb3", "libblobmsg-json", "libc", "libiwinfo", "libiwinfo-data", "libjson-c", "libjson-script", "libmnl", "libnftnl", "libnl-tiny", "libopenssl", "libubox", "libubus", "libuci", "libuclient", "libucode", "libustream-wolfssl", "libwolfssl", "logd", "luci", "luci-app-attendedsysupgrade", "mtd", "netifd", "nftables-json", "odhcp6c", "odhcpd-ipv6only", "openwrt-keyring", "opkg", "ppp", "ppp-mod-pppoe", "procd", "procd-seccomp", "procd-ujail", "snmpd", "ubi-utils", "ubox", "ubus", "ubusd", "uci", "uclient-fetch", "ucode", "ucode-mod-fs", "ucode-mod-ubus", "ucode-mod-uci", "urandom-seed", "urngd", "usign", "wpad-mesh-openssl" ], "diff_packages": true, "filesystem": "squashfs", "client": "luci/git-22.285.67526-18bfcca"}

CharlieYu4994, 2023-03-16:
I have the same issue, it seems that some packages are missing

illumiN8i, 2023-03-16:
Same issue here on two different devices:
Server response: Unsupported package(s): htop, mkf2fs, nftables-json
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r22260-a7a3de5edb",
    "advanced_mode": "0",
    "branch": "SNAPSHOT",
    "efi": {
        "path": "/sys/firmware/efi",
        "type": "directory",
        "size": 0,
        "mode": 16877,
        "atime": 1678676720,
        "mtime": 1678676720,
        "ctime": 1678676720,
        "inode": 6204,
        "uid": 0,
        "gid": 0
    },
    "profile": "default-string-default-string",
    "target": "x86/64",
    "version": "SNAPSHOT",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "grub2-bios-setup",
        "htop",
        "irqbalance",
        "kernel",
        "kmod-amazon-ena",
        "kmod-amd-xgbe",
        "kmod-bnx2",
        "kmod-button-hotplug",
        "kmod-e1000",
        "kmod-e1000e",
        "kmod-forcedeth",
        "kmod-fs-vfat",
        "kmod-igb",
        "kmod-igc",
        "kmod-ixgbe",
        "kmod-nft-offload",
        "kmod-r8169",
        "kmod-tg3",
        "libc",
        "libustream-mbedtls",
        "logd",
        "lsblk",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-bcp38",
        "luci-app-sqm",
        "luci-app-upnp",
        "luci-ssl",
        "mkf2fs",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "partx-utils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "stubby",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.285.67526-18bfcca"
}

```
Server response: Unsupported package(s): wpad-basic-wolfssl, htop, libwolfssl, nftables-json, libnftnl, libmnl, libustream-wolfssl

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r22248-bf055fcdca",
    "advanced_mode": "0",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "netgear,r7800",
    "target": "ipq806x/generic",
    "version": "SNAPSHOT",
    "packages": [
        "ath10k-firmware-qca9984-ct",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "htop",
        "irqbalance",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-ahci",
        "kmod-ata-ahci-platform",
        "kmod-ata-core",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-phy-qcom-ipq806x-usb",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-ramoops",
        "kmod-scsi-core",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-ehci",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb-ohci",
        "kmod-usb-xhci-hcd",
        "kmod-usb2",
        "kmod-usb3",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-bcp38",
        "luci-app-sqm",
        "mtd",
        "nano",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "stubby",
        "swconfig",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.285.67526-18bfcca"
}

```

azeu666, 2023-03-16:
Wrt1900acs and currently running: SNAPSHOT - r22256-14334c222e

Error building the firmware image

Server response: Unsupported package(s): collectd-mod-sensors, htop, collectd-mod-irq, collectd-mod-thermal, iftop, libnetfilter-conntrack, collectd-mod-sqm, nftables-json, libnettle, dnsmasq-full, ipset, iperf3, libnftnl, libmnl, bmon, usbutils
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r22256-14334c222e",
    "advanced_mode": "1",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "linksys,wrt1900acs",
    "target": "mvebu/cortexa9",
    "version": "SNAPSHOT",
    "packages": [
        "attendedsysupgrade-common",
        "auc",
        "banip",
        "base-files",
        "block-mount",
        "bmon",
        "busybox",
        "ca-bundle",
        "collectd-mod-irq",
        "collectd-mod-sensors",
        "collectd-mod-sqm",
        "collectd-mod-thermal",
        "dnsmasq-full",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hdparm",
        "htop",
        "iftop",
        "iperf3",
        "ipset",
        "iptables-zz-legacy",
        "irqbalance",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-acompress",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-fs-ext4",
        "kmod-gpio-button-hotplug",
        "kmod-ipt-ipset",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-storage",
        "kmod-usb2",
        "kmod-usb3",
        "kmod-usbmon",
        "libblobmsg-json",
        "libc",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "libmnl",
        "libnetfilter-conntrack",
        "libnettle",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuci-lua",
        "libuclient",
        "libucode",
        "libudev-zero",
        "libustream-openssl",
        "logd",
        "luci",
        "luci-app-advanced-reboot",
        "luci-app-attendedsysupgrade",
        "luci-app-banip",
        "luci-app-sqm",
        "luci-app-statistics",
        "luci-lib-base",
        "luci-proto-wireguard",
        "mtd",
        "mwlwifi-firmware-88w8864",
        "nano",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "tcpdump",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "urandom-seed",
        "urngd",
        "usbutils",
        "usign",
        "wpad-openssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.285.67526-18bfcca"
}

Close

khanjui, 2023-03-16:
Same here, nftables-json is missing.

illumiN8i, 2023-03-18:
This appears to be fixed now.

Stkrx, 2023-03-18:
Same here, missing curl - nftables-json

MrFoxPro, 2023-03-18:
on mi router 4c
Unsupported package(s): nftables-json, libmnl, libnftnl, openvpn-openssl

aparcar, 2023-03-18:
Im afraid this is an upstream package of broken packages and not of asu itself