openwrt/asu #472: Error while building Xiaomi AX3600 Snapshot.

starraiderx, 2023-03-22:
Server response: Unsupported package(s): procd-seccomp, wpad-basic-mbedtls, procd-ujail, procd

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r22346-e026a65af1",
    "advanced_mode": "1",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "xiaomi,ax3600",
    "target": "ipq807x/generic",
    "version": "SNAPSHOT",
    "packages": [
        "ath10k-board-qca9887",
        "ath10k-firmware-qca9887-ct",
        "ath11k-firmware-ipq8074",
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "ipq-wifi-xiaomi_ax3600",
        "irqbalance",
        "kernel",
        "kmod-ath10k-ct-smallbuffers",
        "kmod-ath11k-ahb",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-nft-offload",
        "kmod-phy-aquantia",
        "kmod-qca-nss-dp",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb3",
        "libc",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-sqm",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "uboot-envtools",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd",
        "wpad-basic-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.285.67526-18bfcca"
}

Razerwire, 2023-03-22:
**Same here on a: "Xiaomi AX3200 / Redmi-router-AX6S"**:

_Server response: Unsupported package(s): procd-ujail, procd-seccomp, procd_

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r22346-e026a65af1",
    "advanced_mode": "1",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "xiaomi,redmi-router-ax6s",
    "target": "mediatek/mt7622",
    "version": "SNAPSHOT",
    "packages": [
        "attendedsysupgrade-common",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "dmesg",
        "dnsmasq-full",
        "dropbear",
        "ethtool",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "grep",
        "hostapd-common",
        "htop",
        "iftop",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-mt76",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7615-common",
        "kmod-mt7615-firmware",
        "kmod-mt7615e",
        "kmod-mt7622-firmware",
        "kmod-mt76x2",
        "kmod-mt76x2-common",
        "kmod-mt7915-firmware",
        "kmod-mt7915e",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libopenssl-afalg",
        "libopenssl-devcrypto",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-openssl",
        "logd",
        "lscpu",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-base",
        "luci-mod-network",
        "luci-mod-status",
        "luci-ssl-openssl",
        "luci-theme-bootstrap",
        "mtd",
        "nano",
        "netifd",
        "nftables-json",
        "nmap-ssl",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "rpcd-mod-luci",
        "tcpdump",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-math",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "whois",
        "wireless-regdb",
        "wpad-openssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.285.67526-18bfcca"
}

rennewpeter, 2023-03-22:
same AX3600:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r22322-b844716158",
    "advanced_mode": "0",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "xiaomi,ax3600",
    "target": "ipq807x/generic",
    "version": "SNAPSHOT",
    "packages": [
        "ath10k-firmware-qca9887-ct",
        "ath11k-firmware-ipq8074",
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ipq-wifi-xiaomi_ax3600",
        "irqbalance",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct-smallbuffers",
        "kmod-ath11k",
        "kmod-ath11k-ahb",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-michael-mic",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-qca-nss-dp",
        "kmod-qca-ssdk",
        "kmod-qrtr",
        "kmod-qrtr-smd",
        "kmod-slhc",
        "kmod-thermal",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-xhci-hcd",
        "kmod-usb3",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-openssl",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-dawn",
        "luci-app-statistics",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-mesh-openssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.285.67526-18bfcca"
}

danpawlik, 2023-03-22:
On Xiaomi AX3200 as well

davidrapan, 2023-03-23:
The sysupgrade is still possible using the auc cli. (AX3600)

lsnsk, 2023-03-23:
Xiaomi Redmi AX6000 error building too:
Server response: Unsupported package(s): procd-seccomp, wpad-basic-mbedtls, procd-ujail, procd

davidrapan, 2023-03-23:
acu is just little bit more descriptive and says that: **installed package *package_name* cannot be found in remote list!**

jtxnn678, 2023-03-24:
Same with BPI-R3