openwrt/asu #477: Error during attempted upgrade

lopez67876, 2023-04-06:
New firmware upgrade available
Currently running: SNAPSHOT - r22497-f25abdf144

Use defaults for the safest update
Select firmware version

SNAPSHOT - r22501-0dc5fc8fa5
Cancel Request firmware image

------------------------------------------------------------------------------

`Error building the firmware image
Server response: Unsupported package(s): openssh-sftp-server, libopenssl

[Please report the error message and request](https://github.com/openwrt/asu/issues)
Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r22497-f25abdf144",
    "advanced_mode": "0",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "linksys,wrt1900acs",
    "target": "mvebu/cortexa9",
    "version": "SNAPSHOT",
    "packages": [
        "attendedsysupgrade-common",
        "banip",
        "base-files",
        "bcp38",
        "block-mount",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq-full",
        "dropbear",
        "e2fsprogs",
        "fdisk",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip-full",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-lib-chacha20",
        "kmod-crypto-lib-chacha20poly1305",
        "kmod-crypto-lib-curve25519",
        "kmod-crypto-lib-poly1305",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-fs-ext4",
        "kmod-fs-ntfs",
        "kmod-fs-vfat",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack-netlink",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-udptunnel4",
        "kmod-udptunnel6",
        "kmod-usb-core",
        "kmod-usb-ohci",
        "kmod-usb-storage",
        "kmod-usb-uhci",
        "kmod-usb2",
        "kmod-wireguard",
        "libblobmsg-json",
        "libbpf",
        "libc",
        "libelf",
        "libgmp",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libnetfilter-conntrack",
        "libnettle",
        "libnfnetlink",
        "libnftnl",
        "libnl-tiny",
        "libopenssl",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "lua",
        "luci",
        "luci-app-advanced-reboot",
        "luci-app-attendedsysupgrade",
        "luci-app-banip",
        "luci-app-bcp38",
        "luci-app-firewall",
        "luci-app-nextdns",
        "luci-app-opkg",
        "luci-app-pbr",
        "luci-base",
        "luci-compat",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-light",
        "luci-lua-runtime",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-proto-wireguard",
        "luci-theme-bootstrap",
        "mtd",
        "mwlwifi-firmware-88w8864",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssh-sftp-server",
        "openwrt-keyring",
        "opkg",
        "pbr",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "resolveip",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rpcsys",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "tcpdump-mini",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-lua",
        "ucode-mod-math",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireguard-tools",
        "wireless-regdb",
        "wpad-basic-mbedtls",
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.285.67526-18bfcca"
}`

omostan, 2023-04-06:
I had same issue:

Error building the firmware image
Server response: Unsupported profile: linksys,wrt3200acm

[Please report the error message and request](https://github.com/openwrt/asu/issues)

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19803-9a599fee93",
    "advanced_mode": "0",
    "branch": "22.03",
    "efi": null,
    "profile": "linksys,wrt3200acm",
    "target": "mvebu/cortexa9",
    "version": "22.03.3",
    "packages": [
        "base-files",
        "blkid",
        "block-mount",
        "blockd",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "cifsmount",
        "collectd-mod-openvpn",
        "collectd-mod-ping",
        "collectd-mod-wireless",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "iptables-mod-conntrack-extra",
        "iptables-mod-ipopt",
        "iptables-nft",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-btmrvl",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-gpio-button-hotplug",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-mwifiex-sdio",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-storage",
        "kmod-usb-storage-uas",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libopenssl",
        "libopenssl-conf",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-advanced-reboot",
        "luci-app-attendedsysupgrade",
        "luci-app-ddns",
        "luci-app-firewall",
        "luci-app-hd-idle",
        "luci-app-ntpc",
        "luci-app-openvpn",
        "luci-app-opkg",
        "luci-app-samba4",
        "luci-app-statistics",
        "luci-app-uhttpd",
        "luci-app-wifischedule",
        "luci-app-wol",
        "luci-base",
        "luci-compat",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-lib-px5g",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "luci-theme-material",
        "mount-utils",
        "mtd",
        "mwlwifi-firmware-88w8964",
        "nano",
        "netifd",
        "nftables-json",
        "ntfs-3g",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssh-sftp-server",
        "openssl-util",
        "openvpn-easy-rsa",
        "openvpn-openssl",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "samba4-server",
        "shadow-useradd",
        "sudo",
        "tc-mod-iptables",
        "tcpdump",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usbutils",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}

azeu666, 2023-04-06:
Same issue:
Error building the firmware image

Server response: Unsupported package(s): collectd-mod-irq, collectd-mod-sensors, collectd-mod-thermal, wpad-openssl, collectd-mod-sqm, libustream-openssl
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r22256-14334c222e",
    "advanced_mode": "1",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "linksys,wrt1900acs",
    "target": "mvebu/cortexa9",
    "version": "SNAPSHOT",
    "packages": [
        "attendedsysupgrade-common",
        "auc",
        "banip",
        "base-files",
        "block-mount",
        "bmon",
        "busybox",
        "ca-bundle",
        "collectd-mod-irq",
        "collectd-mod-sensors",
        "collectd-mod-sqm",
        "collectd-mod-thermal",
        "dnsmasq-full",
        "dropbear",
        "e2fsprogs",
        "ethtool-full",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hdparm",
        "htop",
        "iftop",
        "iperf3",
        "ipset",
        "iptables-zz-legacy",
        "irqbalance",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-crypto-acompress",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-fs-ext4",
        "kmod-gpio-button-hotplug",
        "kmod-ipt-ipset",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-storage",
        "kmod-usb2",
        "kmod-usb3",
        "kmod-usbmon",
        "libblobmsg-json",
        "libc",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "libmnl",
        "libnetfilter-conntrack",
        "libnettle",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuci-lua",
        "libuclient",
        "libucode",
        "libudev-zero",
        "libustream-openssl",
        "logd",
        "lsof",
        "luci",
        "luci-app-advanced-reboot",
        "luci-app-attendedsysupgrade",
        "luci-app-banip",
        "luci-app-sqm",
        "luci-app-statistics",
        "luci-lib-base",
        "luci-proto-wireguard",
        "mtd",
        "mwlwifi-firmware-88w8864",
        "nano",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "tcpdump",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "urandom-seed",
        "urngd",
        "usbutils",
        "usign",
        "wpad-openssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.285.67526-18bfcca"
}

lopez67876, 2023-04-07:
Fixed

trenchcoatjedi, 2023-04-07:
Still having similar issues with anything involving openssl 

Model:
Linksys E7350
Platform:
ramips/mt7621
Version:
SNAPSHOT (r22503-3b212db232)
Date:
2023-04-06 10:15:23

Unsupported package(s): wget-ssl, openssh-sftp-server, libopenssl

lopez67876, 2023-04-08:
Now working ok on my wrt1900acs
[image]

closed 2023-04-08 by lopez67876