openwrt/asu #484: Error on ASU operation on GoFlexHome

AlexeiGHub, 2023-04-17:
From:
<html><body>

Model | Seagate GoFlex Home
-- | --
Architecture | Feroceon 88FR131 rev 1 (v5l)
Target Platform | kirkwood/generic
Firmware Version | OpenWrt 22.03.3 r20028-43d71ad93e / LuCI openwrt-22.03 branch git-23.093.57104-ce20b4a

</body>
</html>

Error on ASU operation to 22.03.4:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r20028-43d71ad93e",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "seagate,goflexhome",
    "target": "kirkwood/generic",
    "version": "22.03.4",
    "packages": [
        "base-files",
        "block-mount",
        "busybox",
        "ca-bundle",
        "cgdisk",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "fwtool",
        "gdisk",
        "getrandom",
        "htop",
        "iperf3",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ata-marvell-sata",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-fs-ext4",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-ehci",
        "kmod-usb-storage",
        "kmod-usb-storage-uas",
        "kmod-usb2",
        "libblkid",
        "libblobmsg-json",
        "libc",
        "libcomerr",
        "libext2fs",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmbedtls",
        "libmnl",
        "libmount",
        "libnftnl",
        "libnl-tiny",
        "libopenssl",
        "libsmartcols",
        "libsqlite3",
        "libss",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "liburing",
        "libustream-wolfssl",
        "libuuid",
        "libwolfssl",
        "lighttpd",
        "lighttpd-mod-scgi",
        "logd",
        "lsblk",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-app-samba4",
        "luci-base",
        "luci-i18n-hd-idle-ru",
        "luci-i18n-ksmbd-ru",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mc",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "php8",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "samba4-libs",
        "samba4-server",
        "screen",
        "smartmontools",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usbutils",
        "usign"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}

STDERR:

Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/releases/22.03.4/targets/kirkwood/generic/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/build_dir/target-arm_xscale_musl_eabi/root-kirkwood/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/dl/openwrt_core
Downloading https://downloads.openwrt.org/releases/22.03.4/targets/kirkwood/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/build_dir/target-arm_xscale_musl_eabi/root-kirkwood/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/dl/openwrt_base
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/build_dir/target-arm_xscale_musl_eabi/root-kirkwood/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/dl/openwrt_luci
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/build_dir/target-arm_xscale_musl_eabi/root-kirkwood/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/dl/openwrt_packages
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/build_dir/target-arm_xscale_musl_eabi/root-kirkwood/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/dl/openwrt_routing
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/build_dir/target-arm_xscale_musl_eabi/root-kirkwood/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/build_dir/target-arm_xscale_musl_eabi/root-kirkwood/../../../../../../../../../../../home/aparcar/asu/worker1/cache/22.03.4/kirkwood/generic/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * opkg_download: Failed to download https://downloads.openwrt.org/releases/22.03.4/packages/arm_xscale/packages/php8_8.1.16-1_arm_xscale.ipk, wget returned 8.
 * opkg_install_pkg: Failed to download php8. Perhaps you need to run 'opkg update'?
 * opkg_install_cmd: Cannot install package php8.
make[2]: *** [Makefile:170: package_install] Error 255
make[1]: *** [Makefile:135: _call_manifest] Error 2
make: *** [Makefile:255: manifest] Error 2

aparcar, 2023-04-17:
Odd, I can reproduce this bug but don't know why it's specifically missing for 22.03.4...

AlexeiGHub, 2023-04-18:
Removing php8 solve problem, but "sediment left".