openwrt/asu #505: Unsupported profile: linksys,wrt1200ac (request for version 22.03.5)

bvhoesel, 2023-05-19:
This was for version 22.03.5 whic failed. Version 21.02.7 did work.

Request data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r16325-88151b8303",
    "advanced_mode": "1",
    "branch": "21.02",
    "efi": null,
    "profile": "linksys,wrt1200ac",
    "target": "mvebu/cortexa9",
    "version": "22.03.5",
    "packages": [
        "atop",
        "attr",
        "auc",
        "avahi-dbus-daemon",
... snipped ...
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-22.210.54423-3b0b325"
}

recyclebin99, 2023-05-20:
I have also had a "Server response: Unsupported profile: linksys,wrt1200ac" error.
Attempting to upgrade from 22.03.2 to 22.03.5.

My 22.03.2 setup is clean - only a few settings changed after resetting to defaults, and only this non-default software package (and dependencies) installed using LuCI System->Software:
- luci-app-attendedsysupgrade - git-23.093.42303-58b861d

Request Data:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19803-9a599fee93",
    "advanced_mode": "0",
    "branch": "22.03",
    "efi": null,
    "profile": "linksys,wrt1200ac",
    "target": "mvebu/cortexa9",
    "version": "22.03.5",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "iwinfo",
        "kernel",
        "kmod-gpio-button-hotplug",
        "kmod-mwlwifi",
        "kmod-nft-offload",
        "libc",
        "libgcc",
        "libustream-wolfssl",
        "logd",
        "luci",
        "luci-app-advanced-reboot",
        "luci-app-attendedsysupgrade",
        "mtd",
        "mwlwifi-firmware-88w8864",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "uboot-envtools",
        "uci",
        "uclient-fetch",
        "urandom-seed",
        "urngd",
        "wpad-basic-wolfssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}

recyclebin99, 2023-05-20:
It seems there have not been any standard pre-built wrt1200ac firmware images (except for SNAPSHOT) since 22.03.2.
At least when searching for 'wrt1200' on [OpenWrt Firmware Selector](https://firmware-selector.openwrt.org/), nothing is shown after 22.03.2 except for SNAPSHOT.

I don't know the relationship between the standard pre-built firmware images and the on-demand images built by the ASU service, but maybe ASU simply refuses to attempt a build if a standard pre-built image is not found for the router/version being requested?

alex7654, 2023-05-25:
Same with Linksys WRT3200ACM, which also has no standard pre-built image later than 22.03.2. Hope, they will release such pre-built images in upcoming verisons again.

aparcar, 2023-05-26:
Yes it's disabled but re-enabled in 23.05. Please be patient.

closed 2023-05-26 by aparcar

alex7654, 2023-05-26:
Thank you for the update. I'm looking forward to that.