openwrt/asu #513: build error while executing auc

ShredRum, 2023-06-05:
Building package index...
Downloading https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/dl/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/dl/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/kmods/5.15.114-1-be57503e7ec8c0657d49a10d963d9d86/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/ramips/mt76x8/kmods/5.15.114-1-be57503e7ec8c0657d49a10d963d9d86/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/dl/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/dl/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/mipsel_24kc/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/build_dir/target-mipsel_24kc_musl/root-ramips/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ramips/mt76x8/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20230523 for mtd
 * pkg_hash_fetch_best_installation_candidate: Packages for mtd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package mtd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mtd:
 *      libubox20230523
 * opkg_install_cmd: Cannot install package mtd.
make[2]: *** [Makefile:186: package_install] Error 255
make[1]: *** [Makefile:151: _call_manifest] Error 2
make: *** [Makefile:271: manifest] Error 2
Error: Impossible package selection
Bad message (77)
root@OpenWrt:~#

aparcar, 2023-06-06:
@dangowrt any ideas what happened here?

ShredRum, 2023-06-06:
I repeated this the next day when a new update came out and it successfully built.

st7105, 2023-06-06:
The problem is actual on r23288.

lopez67876, 2023-06-06:
In my case it compiles ok puting out block-mount and mtd, but I will better wait for fix ;)

lopez67876, 2023-06-07:
All ok today with OpenWrt SNAPSHOT r23293 (Linux 5.15.114)

ShredRum, 2023-06-07:
It seems to have been [fixed](https://github.com/openwrt/openwrt/commit/4747b96287aa3a0f1df5047fc0b2cc1835ff9cc8)

closed 2023-06-07 by ShredRum