openwrt/asu #515: cannot find dependency libubox20230523 for mtd

Soromeister, 2023-06-06:
Trying to generate new image using `Attended Sysupgrade` tool. Currently have `libubox20220927` and `mtd` version `26`. Same thing happens if I try to generate it online.

Request data:

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23282-a3ee2bf9a1",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "request_hash": "5a24b6166209ef0c8b20b42a2823709d",
    "profile": "xiaomi,ax9000",
    "target": "ipq807x/generic",
    "version": "SNAPSHOT",
    "packages": [
        "ath10k-firmware-qca9887-ct-full-htt",
        "ath11k-firmware-ipq8074",
        "ath11k-firmware-qcn9074",
        "base-files",
        "busybox",
        "ca-bundle",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ipq-wifi-xiaomi_ax9000",
        "iw-full",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-ath11k",
        "kmod-ath11k-ahb",
        "kmod-ath11k-pci",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-michael-mic",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-libphy",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-phy-aquantia",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-qca-nss-dp",
        "kmod-qca-ssdk",
        "kmod-qrtr",
        "kmod-qrtr-smd",
        "kmod-slhc",
        "kmod-thermal",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-xhci-hcd",
        "kmod-usb3",
        "libblobmsg-json",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-sqm",
        "luci-app-statistics",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-mbedtls"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.132.65998-fa9fb2f"
}
```

STDERR:

```
Package list missing or not up-to-date, generating it.

Building package index...
Downloading https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_core
Downloading https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_base
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/base/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/kmods/6.1.31-1-60f86a80e0a69b692f8cccde5fa6019e/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_kmods
Downloading https://downloads.openwrt.org/snapshots/targets/ipq807x/generic/kmods/6.1.31-1-60f86a80e0a69b692f8cccde5fa6019e/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_luci
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/luci/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_packages
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/packages/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_routing
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/routing/Packages.sig
Signature check passed.
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.gz
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/openwrt_telephony
Downloading https://downloads.openwrt.org/snapshots/packages/aarch64_cortex-a53/telephony/Packages.sig
Signature check passed.
Downloading file:packages/Packages
Updated list of available packages in /home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/build_dir/target-aarch64_cortex-a53_musl/root-ipq807x/../../../../../../../../../../../home/aparcar/asu/worker1/cache/SNAPSHOT/ipq807x/generic/dl/imagebuilder
Downloading file:packages/Packages.sig
Signature check failed.
Remove wrong Signature file.
Collected errors:
 * pkg_hash_check_unresolved: cannot find dependency libubox20230523 for mtd
 * pkg_hash_fetch_best_installation_candidate: Packages for mtd found, but incompatible with the architectures configured
 * opkg_install_cmd: Cannot install package mtd.
 * satisfy_dependencies_for: Cannot satisfy the following dependencies for mtd:
 * 	libubox20230523
 * opkg_install_cmd: Cannot install package mtd.
make[2]: *** [Makefile:187: package_install] Error 255
make[1]: *** [Makefile:152: _call_manifest] Error 2
make: *** [Makefile:272: manifest] Error 2
```

bzumik1, 2023-06-06:
I am having the same issue.

ShredRum, 2023-06-07:
#513

Soromeister, 2023-06-07:
Fixed by [this commit](https://github.com/openwrt/openwrt/commit/4747b96287aa3a0f1df5047fc0b2cc1835ff9cc8).

closed 2023-06-07 by Soromeister