openwrt/asu #518: Support for 23.05 release candidate

tdobes, 2023-06-11:
Now that version 23.05 has a [release candidate available](https://openwrt.org/releases/23.05/notes-23.05.0-rc1), would it be possible to add support for custom builds from that branch?

At the moment, trying to customize the packages in a 23.05.0-rc1 build results in the error `Unsupported branch: 23.05.0-rc1`

fpdevbr, 2023-06-12:
Would be nice to have support for [`23.05-SNAPSHOT`](https://downloads.openwrt.org/releases/23.05-SNAPSHOT/) as well.

fpdevbr, 2023-06-13:
Apparently it's supported now, but auc wants to downgrade almost every luci related package.

[image]

fpdevbr, 2023-06-21:
Still happening, luci packages from asu/auc have remained static since last week, while to ones bundled with the default firmware from [`23.05-SNAPSHOT`](https://downloads.openwrt.org/releases/23.05-SNAPSHOT/) were updated.

I'm holding off from using asu/auc for now.

[image]

fpdevbr, 2023-06-21:
Still happening, luci packages from asu/auc have remained static since last week, while the ones bundled with the default firmware from [`23.05-SNAPSHOT`](https://downloads.openwrt.org/releases/23.05-SNAPSHOT/) were updated.

I'm holding off from using asu/auc for now.

[image]

udanfacy22, 2023-06-29:
RC2 is now available but has the same package customization error: `Unsupported branch: 23.05.0-rc2`.

aparcar, 2023-06-30:
Thanks for the reports I’ll look into that hopefully during this weekend

udanfacy22, 2023-06-30:
> Thanks for the reports I’ll look into that hopefully during this weekend

Rc2 got added as an available branch before you replied, and I successfully customized and installed on my Ras Pi 4B 64-bit.

Based on the rc1 thread, people's understanding on the forum seemed to be that:

1. Adding available branches to the firmware selector is a manual process, done by adjusting the sysupgrade.openwrt.org configuration.
2. You, aparcar, are the only person with access to control that configuration (or, at least, you are currently the only one who actually uses their access to that).
3. You are traveling, and thus a bit slower to update the that configuration than you might usually be (and obviously OpenWrt is run by volunteers).

I take it at least some of those points are false?