openwrt/asu #531: linksys,wrt1200 failed to buid image

hongdat1106, 2023-07-05:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19803-9a599fee93",
    "advanced_mode": "1",
    "branch": "22.03",
    "efi": null,
    "profile": "linksys,wrt1200ac",
    "target": "mvebu/cortexa9",
    "version": "22.03.2",
    "packages": [
        "auc",
        "base-files",
        "bind-host",
        "bind-libs",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "coreutils",
        "coreutils-sort",
        "curl",
        "ddns-scripts",
        "ddns-scripts-noip",
        "ddns-scripts-services",
        "dnsmasq",
        "dropbear",
        "firewall",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "igmpproxy",
        "ip6tables-nft",
        "iptables-nft",
        "iw",
        "iwinfo",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-gpio-button-hotplug",
        "kmod-ip6tables",
        "kmod-ipt-conntrack",
        "kmod-ipt-core",
        "kmod-ipt-nat",
        "kmod-ipt-offload",
        "kmod-lib-crc-ccitt",
        "kmod-mac80211",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-ipt",
        "kmod-nf-ipt6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-tun",
        "libatomic",
        "libblobmsg-json",
        "libc",
        "libcurl",
        "libgcc",
        "libip4tc",
        "libip6tc",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "liblzo",
        "libnghttp2",
        "libnl-tiny",
        "libopenssl",
        "libopenssl-conf",
        "libpthread",
        "libqrencode",
        "librt",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libustream-wolfssl",
        "libuv",
        "libwolfssl",
        "libxtables",
        "logd",
        "lua",
        "luci",
        "luci-app-advanced-reboot",
        "luci-app-attendedsysupgrade",
        "luci-app-ddns",
        "luci-app-firewall",
        "luci-app-openvpn",
        "luci-app-opkg",
        "luci-app-p910nd",
        "luci-app-udpxy",
        "luci-base",
        "luci-compat",
        "luci-i18n-attendedsysupgrade-en",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-ipkg",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-theme-bootstrap",
        "mtd",
        "mwlwifi-firmware-88w8864",
        "netifd",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssl-util",
        "openvpn-easy-rsa",
        "openvpn-openssl",
        "openwrt-keyring",
        "opkg",
        "p910nd",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "qrencode",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "udpxy",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl",
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.093.42303-58b861d"
}

danpawlik, 2023-07-05:
same issue for mediatek and ramips.

Dugas, 2023-07-05:
same issue but on x86/64

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r19777-2853b6d652",
    "advanced_mode": "0",
    "branch": "22.03",
    "efi": null,
    "profile": "qemu-standard-pc-i440fx-piix-1996",
    "target": "x86/64",
    "version": "22.03.5",
    "packages": [
        "auc",
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "grub2",
        "grub2-bios-setup",
        "grub2-efi",
        "hostapd",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-amazon-ena",
        "kmod-amd-xgbe",
        "kmod-ath10k",
        "kmod-ath9k",
        "kmod-bnx2",
        "kmod-button-hotplug",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-e1000",
        "kmod-e1000e",
        "kmod-forcedeth",
        "kmod-fs-vfat",
        "kmod-igb",
        "kmod-igc",
        "kmod-input-core",
        "kmod-ixgbe",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-nls-cp437",
        "kmod-nls-iso8859-1",
        "kmod-nls-utf8",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-r8169",
        "kmod-slhc",
        "kmod-tg3",
        "libblkid",
        "libblobmsg-json",
        "libcomerr",
        "libext2fs",
        "libf2fs",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libsmartcols",
        "libss",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libuuid",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mkf2fs",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "partx-utils",
        "pciutils",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "px5g-wolfssl",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usbutils",
        "usign",
        "wpa-supplicant"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/xxx"
}

danpawlik, 2023-07-09:
Should be fine now. Could you close issue?

hongdat1106, 2023-07-09:
Not yet, I don't know how to solve this issue :)

danpawlik, 2023-07-09:
> Not yet, I don't know how to solve this issue :)

Just go on the firmware selector site or update image via auc.