openwrt/asu #538: Attended sysupgrade failed on BPI-R3

go2car, 2023-07-07:
Error building the firmware image

Server response: The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.
[Please report the error message and request](https://github.com/openwrt/asu/issues)

Request Data:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23488-11588c52b4",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "bananapi,bpi-r3",
    "target": "mediatek/filogic",
    "version": "SNAPSHOT",
    "packages": [
        "base-files",
        "block-mount",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "e2fsprogs",
        "eip197-mini-firmware",
        "f2fsck",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "glib2",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-authenc",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-des",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-hw-safexcel",
        "kmod-crypto-manager",
        "kmod-crypto-md5",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha1",
        "kmod-crypto-sha256",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-hwmon-pwmfan",
        "kmod-i2c-algo-bit",
        "kmod-i2c-core",
        "kmod-i2c-gpio",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-libphy",
        "kmod-mac80211",
        "kmod-mt76-connac",
        "kmod-mt76-core",
        "kmod-mt7915e",
        "kmod-mt7986-firmware",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-phylink",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-sfp",
        "kmod-slhc",
        "kmod-thermal",
        "kmod-usb-core",
        "kmod-usb-xhci-hcd",
        "kmod-usb-xhci-mtk",
        "kmod-usb3",
        "libattr",
        "libblkid",
        "libblobmsg-json",
        "libc",
        "libcomerr",
        "libext2fs",
        "libf2fs",
        "libffi",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "liblucihttp",
        "liblucihttp-ucode",
        "libmbedtls",
        "libmnl",
        "libmount",
        "libncurses",
        "libnftnl",
        "libnl-tiny",
        "libopenssl",
        "libpcre2",
        "libss",
        "libssh2",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "libuuid",
        "logd",
        "luci",
        "luci-app-adblock",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-opkg",
        "luci-base",
        "luci-light",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-theme-bootstrap",
        "mc",
        "mc-skins",
        "mkf2fs",
        "mt7986-wo-firmware",
        "mtd",
        "nano-full",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openssh-keygen",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "rpcd-mod-ucode",
        "terminfo",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-html",
        "ucode-mod-math",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-basic-mbedtls",
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.132.65998-fa9fb2f"
}

danpawlik, 2023-07-09:
Should be fine now. Could you close issue?

go2car, 2023-07-09:
works fine for me

closed 2023-07-09 by go2car