openwrt/asu #553: auc: Invalid argument (22)

bam80, 2023-08-08:
```
# auc
auc/0.3.1-1
Server:    https://sysupgrade.openwrt.org
Running:   22.03.0-rc6 r19590-042d558536 on ramips/mt76x8 (kroks,kndrt31r19)
Invalid argument (22)
#

basrieter, 2023-11-22:
Save here:

```
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   23.05.2 r23630-842932a63d on ipq806x/generic (linksys,ea7500-v1)
Invalid argument (22)

```

bam80, 2024-02-21:
Please reopen if it's still an issue for someone, I suppose it has been addressed already.

AntonioMeireles, 2024-02-26:
```
root@AP_S:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r25252-63f7ced2f0 on qualcommax/ipq807x (dynalink,dl-wrx36)
WARNING: cannot determing currently running branch.
Invalid argument (22)
```

fwiw, back since this weekend here

humaita-github, 2024-03-20:
Also, please note the typo in the warning: it should be "cannot determine..." instead of "cannot determing..."

aparcar, 2024-04-01:
@Delphius7 did you restart `rpcd` service?

GoetzGoerisch, 2024-06-28:
Please retry with [owut](https://github.com/efahl/owut/), the replacement of `auc`