openwrt/asu #564: Error in attended system upgrade to r23769-324673914d

andrelmoliveira, 2023-08-23:
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23769-324673914d",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "linksys,whw03v2",
    "target": "ipq40xx/generic",
    "version": "SNAPSHOT",
    "packages": [
        "announce",
        "ath10k-board-qca4019",
        "ath10k-firmware-qca4019-ct",
        "ath10k-firmware-qca9888-ct",
        "avahi-utils",
        "base-files",
        "busybox",
        "ca-bundle",
        "dbus",
        "dbus-utils",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "ip-full",
        "iw-full",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-ath",
        "kmod-ath10k-ct",
        "kmod-bluetooth",
        "kmod-cfg80211",
        "kmod-crypto-acompress",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha512",
        "kmod-gpio-button-hotplug",
        "kmod-hwmon-core",
        "kmod-leds-gpio",
        "kmod-leds-pca963x",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mac80211",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-spi-dev",
        "kmod-usb-core",
        "kmod-usb-dwc3",
        "kmod-usb-dwc3-qcom",
        "kmod-usb-xhci-hcd",
        "kmod-usb3",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "libmbedtls",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-mbedtls",
        "logd",
        "luci",
        "luci-app-advanced-reboot",
        "luci-app-attendedsysupgrade",
        "luci-app-commands",
        "luci-app-nextdns",
        "luci-app-nft-qos",
        "luci-app-openvpn",
        "luci-app-pbr",
        "luci-app-qos",
        "luci-app-uhttpd",
        "luci-app-watchcat",
        "luci-compat",
        "luci-i18n-advanced-reboot-pt-br",
        "luci-i18n-attendedsysupgrade-pt-br",
        "luci-i18n-base-pt-br",
        "luci-i18n-commands-pt-br",
        "luci-i18n-firewall-pt-br",
        "luci-i18n-nextdns-pt-br",
        "luci-i18n-nft-qos-pt-br",
        "luci-i18n-openvpn-pt-br",
        "luci-i18n-opkg-pt-br",
        "luci-i18n-pbr-pt-br",
        "luci-i18n-qos-pt-br",
        "luci-i18n-uhttpd-pt-br",
        "luci-i18n-watchcat-pt-br",
        "luci-proto-batman-adv",
        "luci-ssl",
        "mdns-repeater",
        "mdns-utils",
        "mdnsresponder",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openvpn-openssl",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "umdns",
        "urandom-seed",
        "urngd",
        "usign",
        "wireless-regdb",
        "wpad-mesh-wolfssl",
        "wsdd2"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.223.40904-987ae1a"
}

lewinson, 2023-08-24:
I think I'm also getting the same error:

{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23703-c764a8c791",
    "advanced_mode": "1",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "netgear,wac104",
    "target": "ramips/mt7621",
    "version": "SNAPSHOT",
    "packages": [
        "auc",
        "avahi-daemon-service-http",
        "avahi-daemon-service-ssh",
        "avahi-dbus-daemon",
        "base-files",
        "batctl-full",
        "busybox",
        "ca-bundle",
        "diffutils",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iperf3",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-batman-adv",
        "kmod-crypto-acompress",
        "kmod-crypto-crc32c",
        "kmod-crypto-hash",
        "kmod-gpio-button-hotplug",
        "kmod-leds-gpio",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc32c",
        "kmod-lib-lzo",
        "kmod-mt7603",
        "kmod-mt76x2",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-slhc",
        "kmod-usb-ledtrig-usbport",
        "kmod-usb3",
        "libblobmsg-json",
        "libc",
        "libiwinfo",
        "libiwinfo-data",
        "libjson-c",
        "libjson-script",
        "libmnl",
        "libnftnl",
        "libnl-tiny",
        "libopenssl",
        "libubox",
        "libubus",
        "libuci",
        "libuclient",
        "libucode",
        "libustream-wolfssl",
        "libwolfssl",
        "logd",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-proto-batman-adv",
        "mtd",
        "netifd",
        "nftables-json",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "snmpd",
        "ubi-utils",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "urandom-seed",
        "urngd",
        "usign",
        "wpad-mesh-openssl"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.132.65998-fa9fb2f"
}

alienatedsec, 2023-08-24:
same here
```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23727-e6f8b69918",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "linksys,wrt32x",
    "target": "mvebu/cortexa9",
    "version": "SNAPSHOT",
    "packages": [
        "base-files",
        "busybox",
        "ca-bundle",
        "cgi-io",
        "dnsmasq",
        "dropbear",
        "firewall4",
        "fstools",
        "fwtool",
        "getrandom",
        "hostapd-common",
        "iw",
        "iwinfo",
        "jansson",
        "jshn",
        "jsonfilter",
        "kernel",
        "kmod-bluetooth",
        "kmod-btmrvl",
        "kmod-cfg80211",
        "kmod-crypto-aead",
        "kmod-crypto-ccm",
        "kmod-crypto-cmac",
        "kmod-crypto-crc32c",
        "kmod-crypto-ctr",
        "kmod-crypto-ecb",
        "kmod-crypto-ecdh",
        "kmod-crypto-gcm",
        "kmod-crypto-gf128",
        "kmod-crypto-ghash",
        "kmod-crypto-hash",
        "kmod-crypto-hmac",
        "kmod-crypto-kpp",
        "kmod-crypto-manager",
        "kmod-crypto-null",
        "kmod-crypto-rng",
        "kmod-crypto-seqiv",
        "kmod-crypto-sha256",
        "kmod-gpio-button-hotplug",
        "kmod-hid",
        "kmod-input-core",
        "kmod-input-evdev",
        "kmod-lib-crc-ccitt",
        "kmod-lib-crc16",
        "kmod-lib-crc32c",
        "kmod-mac80211",
        "kmod-mmc",
        "kmod-mwifiex-sdio",
        "kmod-mwlwifi",
        "kmod-nf-conntrack",
        "kmod-nf-conntrack6",
        "kmod-nf-flow",
        "kmod-nf-log",
        "kmod-nf-log6",
        "kmod-nf-nat",
        "kmod-nf-reject",
        "kmod-nf-reject6",
        "kmod-nfnetlink",
        "kmod-nft-core",
        "kmod-nft-fib",
        "kmod-nft-nat",
        "kmod-nft-offload",
        "kmod-nls-base",
        "kmod-ppp",
        "kmod-pppoe",
        "kmod-pppox",
        "kmod-regmap-core",
        "kmod-scsi-core",
        "kmod-slhc",
        "kmod-usb-core",
        "kmod-usb-ehci",
        "kmod-usb-ohci",
        "kmod-usb-ohci-pci",
        "kmod-usb-storage",
        "kmod-usb-storage-uas",
        "kmod-usb-uhci",
        "kmod-usb-xhci-hcd",
        "kmod-usb-xhci-pci-renesas",
        "kmod-usb2",
        "kmod-usb2-pci",
        "kmod-usb3",
        "libblobmsg-json",
        "libc",
        "libevdev",
        "libexpat",
        "libgd",
        "libiwinfo",
        "libiwinfo-data",
        "libiwinfo-lua",
        "libjpeg-turbo",
        "libjson-c",
        "libjson-script",
        "liblua",
        "liblucihttp",
        "liblucihttp-lua",
        "libmnl",
        "libncurses",
        "libneon",
        "libnftnl",
        "libnl-tiny",
        "libopenssl",
        "libpng",
        "libpthread",
        "librt",
        "libubox",
        "libubus",
        "libubus-lua",
        "libuci",
        "libuclient",
        "libucode",
        "libudev-zero",
        "libusb-1.0",
        "libusb-compat",
        "libwebp",
        "libwolfssl",
        "logd",
        "lua",
        "luci",
        "luci-app-attendedsysupgrade",
        "luci-app-firewall",
        "luci-app-nut",
        "luci-app-opkg",
        "luci-base",
        "luci-compat",
        "luci-lib-base",
        "luci-lib-ip",
        "luci-lib-json",
        "luci-lib-jsonc",
        "luci-lib-nixio",
        "luci-mod-admin-full",
        "luci-mod-network",
        "luci-mod-rpc",
        "luci-mod-status",
        "luci-mod-system",
        "luci-proto-ipv6",
        "luci-proto-ppp",
        "luci-ssl",
        "luci-theme-bootstrap",
        "mtd",
        "mwifiex-sdio-firmware",
        "mwlwifi-firmware-88w8964",
        "nano",
        "netifd",
        "nftables-json",
        "nut",
        "nut-common",
        "nut-driver-usbhid-ups",
        "nut-server",
        "nut-upsc",
        "nut-upsmon",
        "nut-web-cgi",
        "odhcp6c",
        "odhcpd-ipv6only",
        "openwrt-keyring",
        "opkg",
        "ppp",
        "ppp-mod-pppoe",
        "procd",
        "procd-seccomp",
        "procd-ujail",
        "rpcd",
        "rpcd-mod-file",
        "rpcd-mod-iwinfo",
        "rpcd-mod-luci",
        "rpcd-mod-rrdns",
        "terminfo",
        "ubi-utils",
        "uboot-envtools",
        "ubox",
        "ubus",
        "ubusd",
        "uci",
        "uclient-fetch",
        "ucode",
        "ucode-mod-fs",
        "ucode-mod-ubus",
        "ucode-mod-uci",
        "uhttpd",
        "uhttpd-mod-ubus",
        "urandom-seed",
        "urngd",
        "usbutils",
        "usign",
        "wireless-regdb",
        "wpad-basic-wolfssl",
        "zlib"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.223.40904-987ae1a"
}
```

alienatedsec, 2023-08-24:
I also get those errors sometimes

```
{
    "url": "https://sysupgrade.openwrt.org",
    "revision": "r23727-e6f8b69918",
    "advanced_mode": "0",
    "sha256_unsigned": "",
    "branch": "SNAPSHOT",
    "efi": null,
    "profile": "linksys,wrt32x",
    "target": "mvebu/cortexa9",
    "version": "SNAPSHOT",
    "packages": [
        "0",
        "1",
        "10",
        "100",
        "101",
        "102",
        "103",
        "104",
        "105",
        "106",
        "107",
        "108",
        "109",
        "11",
        "110",
        "111",
        "112",
        "113",
        "114",
        "115",
        "116",
        "117",
        "118",
        "119",
        "12",
        "120",
        "121",
        "122",
        "123",
        "124",
        "125",
        "126",
        "127",
        "128",
        "129",
        "13",
        "130",
        "131",
        "132",
        "133",
        "134",
        "135",
        "136",
        "137",
        "138",
        "139",
        "14",
        "140",
        "141",
        "142",
        "143",
        "144",
        "145",
        "146",
        "147",
        "148",
        "149",
        "15",
        "150",
        "151",
        "152",
        "153",
        "154",
        "155",
        "156",
        "157",
        "158",
        "159",
        "16",
        "160",
        "161",
        "162",
        "163",
        "164",
        "165",
        "166",
        "167",
        "168",
        "169",
        "17",
        "170",
        "171",
        "172",
        "173",
        "174",
        "175",
        "176",
        "177",
        "178",
        "179",
        "18",
        "180",
        "181",
        "182",
        "183",
        "184",
        "185",
        "19",
        "2",
        "20",
        "21",
        "22",
        "23",
        "24",
        "25",
        "26",
        "27",
        "28",
        "29",
        "3",
        "30",
        "31",
        "32",
        "33",
        "34",
        "35",
        "36",
        "37",
        "38",
        "39",
        "4",
        "40",
        "41",
        "42",
        "43",
        "44",
        "45",
        "46",
        "47",
        "48",
        "49",
        "5",
        "50",
        "51",
        "52",
        "53",
        "54",
        "55",
        "56",
        "57",
        "58",
        "59",
        "6",
        "60",
        "61",
        "62",
        "63",
        "64",
        "65",
        "66",
        "67",
        "68",
        "69",
        "7",
        "70",
        "71",
        "72",
        "73",
        "74",
        "75",
        "76",
        "77",
        "78",
        "79",
        "8",
        "80",
        "81",
        "82",
        "83",
        "84",
        "85",
        "86",
        "87",
        "88",
        "89",
        "9",
        "90",
        "91",
        "92",
        "93",
        "94",
        "95",
        "96",
        "97",
        "98",
        "99"
    ],
    "diff_packages": true,
    "filesystem": "squashfs",
    "client": "luci/git-23.223.40904-987ae1a"
}
```

alienatedsec, 2023-08-24:
when ran manually using `auc` its comes with the following error

```
root@WRT-Router:~# auc
auc/0.3.2-1
Server:    https://sysupgrade.openwrt.org
Running:   SNAPSHOT r23727-e6f8b69918 on mvebu/cortexa9 (linksys,wrt32x)
Available: SNAPSHOT r23789-ce6ad123e7
Requesting package lists...
 kmod-usb-storage: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-gf128: 5.15.126-1 -> 5.15.127-1
 kmod-usb-core: 5.15.126-1 -> 5.15.127-1
 kmod-usb-storage-uas: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-manager: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-ctr: 5.15.126-1 -> 5.15.127-1
 kmod-btmrvl: 5.15.126-1 -> 5.15.127-1
 kmod-nft-fib: 5.15.126-1 -> 5.15.127-1
 kmod-nfnetlink: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-hash: 5.15.126-1 -> 5.15.127-1
 kmod-nf-reject6: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-aead: 5.15.126-1 -> 5.15.127-1
 kmod-nf-flow: 5.15.126-1 -> 5.15.127-1
 kmod-lib-crc-ccitt: 5.15.126-1 -> 5.15.127-1
 kmod-usb2-pci: 5.15.126-1 -> 5.15.127-1
 kmod-pppoe: 5.15.126-1 -> 5.15.127-1
 luci-lua-runtime: git-23.119.80867-2a9c3a3 -> git-23.233.49476-18b637f
 kmod-pppox: 5.15.126-1 -> 5.15.127-1
 kmod-nf-reject: 5.15.126-1 -> 5.15.127-1
 kmod-lib-crc16: 5.15.126-1 -> 5.15.127-1
 base-files: 1534-r23727-e6f8b69918 -> 1534-r23789-ce6ad123e7
 kmod-nf-nat: 5.15.126-1 -> 5.15.127-1
 kmod-input-core: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-crc32c: 5.15.126-1 -> 5.15.127-1
 netifd: 2023-07-17-0ff22a6a-1 -> 2023-08-17-4711f744-1
 kmod-usb-xhci-pci-renesas: 5.15.126-1 -> 5.15.127-1
 uboot-envtools: 2023.04-1 -> 2023.07.02-1
 kmod-usb-ohci: 5.15.126-1 -> 5.15.127-1
 kmod-usb-xhci-hcd: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-hmac: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-sha256: 5.15.126-1 -> 5.15.127-1
 kmod-lib-crc32c: 5.15.126-1 -> 5.15.127-1
 kmod-usb-ehci: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-sha512: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-seqiv: 5.15.126-1 -> 5.15.127-1
 kmod-nft-nat: 5.15.126-1 -> 5.15.127-1
 kmod-usb2: 5.15.126-1 -> 5.15.127-1
 kmod-usb3: 5.15.126-1 -> 5.15.127-1
 kmod-mmc: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-ghash: 5.15.126-1 -> 5.15.127-1
 kmod-input-evdev: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-kpp: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-ecdh: 5.15.126-1 -> 5.15.127-1
 kmod-mwifiex-sdio: 5.15.126+6.1.24-4 -> 5.15.127+6.1.24-4
 kmod-crypto-ccm: 5.15.126-1 -> 5.15.127-1
 kmod-nf-log: 5.15.126-1 -> 5.15.127-1
 luci-base: git-23.223.85458-f7583b6 -> git-23.233.52631-b4594b3
 kmod-gpio-button-hotplug: 5.15.126-3 -> 5.15.127-3
 kmod-bluetooth: 5.15.126-1 -> 5.15.127-1
 kmod-nf-log6: 5.15.126-1 -> 5.15.127-1
 kmod-mac80211: 5.15.126+6.1.24-4 -> 5.15.127+6.1.24-4
 kmod-crypto-ecb: 5.15.126-1 -> 5.15.127-1
 kmod-mwlwifi: 5.15.126+2023-04-29-6a436714-2 -> 5.15.127+2023-04-29-6a436714-2
 kmod-ppp: 5.15.126-1 -> 5.15.127-1
 kmod-nft-offload: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-null: 5.15.126-1 -> 5.15.127-1
 kmod-nf-conntrack: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-gcm: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-cmac: 5.15.126-1 -> 5.15.127-1
 kmod-usb-ohci-pci: 5.15.126-1 -> 5.15.127-1
 kmod-crypto-rng: 5.15.126-1 -> 5.15.127-1
 kmod-nf-conntrack6: 5.15.126-1 -> 5.15.127-1
 kmod-usb-uhci: 5.15.126-1 -> 5.15.127-1
 luci-mod-network: git-23.223.85272-4c3584b -> git-23.233.29374-a6a428f
 kmod-nft-core: 5.15.126-1 -> 5.15.127-1
 kmod-regmap-core: 5.15.126-1 -> 5.15.127-1
 kmod-nls-base: 5.15.126-1 -> 5.15.127-1
 kmod-hid: 5.15.126-1 -> 5.15.127-1
 kmod-scsi-core: 5.15.126-1 -> 5.15.127-1
 kmod-slhc: 5.15.126-1 -> 5.15.127-1
 kmod-cfg80211: 5.15.126+6.1.24-4 -> 5.15.127+6.1.24-4
Are you sure you want to continue the upgrade process? [N/y] y
The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.
Bad message (74)
```

andrelmoliveira, 2023-08-25:
I didn't know about this "auc" utility. I decided to install it here now to test it out. Funnily enough it worked... It did the update from r23769-324673914d to r23793-12f53724c6. Perhaps the original bug is related to the update feature via Luci. Honestly I don't know, I'm just guessing.

alienatedsec, 2023-08-25:
@andrelmoliveira it had to be server related issue as both methods worked yesterday evening

andrelmoliveira, 2023-08-25:
> @andrelmoliveira it had to be server related issue as both methods worked yesterday evening

Oh ok got it